OpenVnmrJ/OpenVnmrJ #15: Build fails on OS X (Yosemite 10.10.5) using scons 2.5

timburrow, 2016-07-15:
- Building on a Mac (case-sensitive file system) fails with the error (at end). Investigation found that header files in the SConstruct file are not copied
```
buildMethods.symLinkHeader(vnmrBgCEnv, unmrComFileList, cwd, unmrComPath, unmrComHdrList)
```

- The build fail exactly the same for when sconsJoption = 5 or sconsJoption = 1

- The build succeeds when using scons 2.2.0.

 - I have not tested other versions of scons or whether this affects builds on CentOS with scons 2.5.

Error reported in log:
```
scons: *** Found dependency cycle(s):
  src/vnmrbg/specfreq.c -> src/vnmrbg/variables.h -> src/vnmrbg/aipVnmrFuncs.C -> src/vnmrbg/aipStructs.h -> src/vnmrbg/AspTraceList.C -> src/vnmrbg/AspDisInteg.h -> src/vnmrbg/aipCommands.C -> src/vnmrbg/aipGraphicsWin.h -> src/vnmrbg/AspPolyline.C -> src/vnmrbg/AspFrameMgr.h -> src/vnmrbg/aipInterface.C -> src/vnmrbg/aipCInterface.h -> src/vnmrbg/integ.c -> src/vnmrbg/iplan.h -> src/vnmrbg/aipSpecViewList.C -> src/vnmrbg/aipSpecStruct.h -> src/vnmrbg/csiFuncs.c -> src/vnmrbg/group.h -> src/vnmrbg/aipGframe.C -> src/vnmrbg/aipCStructs.h -> src/vnmrbg/annotate.c -> src/vnmrbg/pvars.h -> src/vnmrbg/ecc_setup.c -> src/vnmrbg/oopc.h -> src/vnmrbg/specfreq.c
  src/vnmrbg/AspDisInteg.h -> src/vnmrbg/aipCommands.C -> src/vnmrbg/aipStructs.h -> src/vnmrbg/AspTraceList.C -> src/vnmrbg/AspDisInteg.h
  Internal Error: no cycle found for node src/vnmrbg/acquisition.h (<SCons.Node.FS.File object at 0x7fe430f820e0>) in state pending
  Internal Error: no cycle found for node src/vnmrbg/AspIntegList.o (<SCons.Node.FS.File object at 0x7fe430f940f0>) in state pending
  src/vnmrbg/magic.gram.h -> src/vnmrbg/io.c -> src/vnmrbg/variables.h -> src/vnmrbg/aipVnmrFuncs.C -> src/vnmrbg/aipStructs.h -> src/vnmrbg/AspTraceList.C -> src/vnmrbg/AspDisInteg.h -> src/vnmrbg/aipCommands.C -> src/vnmrbg/aipGraphicsWin.h -> src/vnmrbg/AspPolyline.C -> src/vnmrbg/AspFrameMgr.h -> src/vnmrbg/aipInterface.C -> src/vnmrbg/aipCInterface.h -> src/vnmrbg/integ.c -> src/vnmrbg/iplan.h -> src/vnmrbg/aipSpecViewList.C -> src/vnmrbg/aipSpecStruct.h -> src/vnmrbg/csiFuncs.c -> src/vnmrbg/group.h -> src/vnmrbg/aipGframe.C -> src/vnmrbg/aipCStructs.h -> src/vnmrbg/annotate.c -> src/vnmrbg/pvars.h -> src/vnmrbg/ecc_setup.c -> src/vnmrbg/oopc.h -> src/vnmrbg/specfreq.c -> src/vnmrbg/acodes.h -> src/vnmrbg/ap_device.c -> src/vnmrbg/REV_NUMS.h -> src/vnmrbg/go.c -> src/vnmrbg/params.h -> src/vnmrbg/freqfuncs.c -> src/vnmrbg/Pbox_HT.h -> src/vnmrbg/ht.c -> src/vnmrbg/stack.h -> src/vnmrbg/magic.gram.c -> src/vnmrbg/magic.gram.h
  Internal Error: no cycle found for node src/vnmrbg/AspCellList.o (<SCons.Node.FS.File object at 0x7fe430dea140>) in state pending
  Internal Error: no cycle found for node src/vnmrbg/sky.h (<SCons.Node.FS.File object at 0x7fe4330d9550>) in state pending
  Internal Error: no cycle found for node src/vnmrbg/aipInterpolation.h (<SCons.Node.FS.File object at 0x7fe430eb0190>) in state pending
  src/vnmrbg/set3dproc.c -> src/vnmrbg/variables.h -> src/vnmrbg/aipVnmrFuncs.C -> src/vnmrbg/aipStructs.h -> src/vnmrbg/AspTraceList.C -> src/vnmrbg/AspDisInteg.h -> src/vnmrbg/aipCommands.C -> src/vnmrbg/aipGraphicsWin.h -> src/vnmrbg/AspPolyline.C -> src/vnmrbg/AspFrameMgr.h -> src/vnmrbg/aipInterface.C -> src/vnmrbg/aipCInterface.h -> src/vnmrbg/integ.c -> src/vnmrbg/iplan.h -> src/vnmrbg/aipSpecViewList.C -> src/vnmrbg/aipSpecStruct.h -> src/vnmrbg/csiFuncs.c -> src/vnmrbg/group.h -> src/vnmrbg/aipGframe.C -> src/vnmrbg/aipCStructs.h -> src/vnmrbg/annotate.c -> src/vnmrbg/pvars.h -> src/vnmrbg/ecc_setup.c -> src/vnmrbg/oopc.h -> src/vnmrbg/specfreq.c -> src/vnmrbg/acodes.h -> src/vnmrbg/ap_device.c -> src/vnmrbg/REV_NUMS.h -> src/vnmrbg/go.c -> src/vnmrbg/params.h -> src/vnmrbg/freqfuncs.c -> src/vnmrbg/Pbox_HT.h -> src/vnmrbg/ht.c -> src/vnmrbg/stack.h -> src/vnmrbg/magic.gram.c -> src/vnmrbg/magic.gram.h -> src/vnmrbg/io.c -> src/vnmrbg/STAT_DEFS.h -> src/vnmrbg/acqfuncs.c -> src/vnmrbg/ACQ_SUN.h -> src/vnmrbg/acqhwcmd.c -> src/vnmrbg/ftpar.h -> src/vnmrbg/set3dproc.c
```

timburrow, 2016-07-19:
I just built with icons 2.4.1 on macOS 10.11.5, Xcode 7.2.1 with no problem

tesch1, 2016-10-18:
MacPorts installs 2.4.1.  I dont know what version Brew ships with, but it builds with whatever version is installed on the travis-ci build machines with "brew install scons".  Also, all of the following install versions of scons that work and dont require any changes: CentOS{6,7} Ubuntu{trusty,xenial}.

So, I'd say this is probably a close-able issue?

timburrow, 2016-10-18:
I don't know when scons 2.5 will become standard; I installed it manually. 

We can close this until the time comes to re-open.

closed 2016-10-18 by timburrow