OpenVnmrJ/OpenVnmrJ #17: Virtual Scanner

tesch1, 2016-07-18:
I think a critical component for development would be a scanner emulator.  This would allow automated testing of changes to the whole system by providing a way to run continuous testing (ie via travisci) on any changes - providing a way to check that the entire scan pipeline from click to reconstruction is functioning correctly, for all scanners "supported" by the software, and for all sequences.

I did some work a while ago on simulating timing from ACODES, and that part is not impossible.  However, there's a lot of unknowns to me, mainly in the communication from software to scanner.  I wanted to open this ticket to lay out the idea, and gather some ideas as to whether this would be a feasible project, and what would need to be done?

Features:
- Opens up the same communication points as the firmware does, and responds identically (or very closely) in terms of communication.  Basically a state-machine that acts like hardware.
- The actual signal data returned when running experiments can either be simulated, or recorded examples.
- Emulate different scanners and firmware revisions. (which?)
- Check that "uploaded" firmware is correct, ie checksum or something similar (?).
- Error handling / catching in abundance.
- Check that the experimental acode program is correct, or at least not blatantly wrong.
- Interface for test harness.

Unknowns:
- Communication layer between OpenVnmrJ and firmware - is there an open source implementation that's compatible with existing firmware?

timburrow, 2016-07-18:
Thanks for the very good suggestion! 

Here are a few of my thoughts:
- the command go('acqi') leaves the ACODES in /vnmr/acqqueue 
  - We can have a "reference" system to produce ACODES and save those, then compare to any builds
  - The ACODES files are binary, so it would be helpful to have documentation on the format of the ACODES file
  - Some of the VxWorks code is in the repository, so that could be useful 
- One issue is that the pulse programs to produce ACODES can be compiled locally, so a local gcc could generate different ACODES. We distributed the standard pulse programs as code and compiled form

- Having a state machine to emulate hardware would be useful for testing
  - The code is old, and I heard that different compiler options can cause communication problems (Qtune on Mercury console was one)
  - There are multiple models of consoles (scanners) that this software supports and it would be difficult to test all, especially if different compilers/OSes are included

tesch1, 2016-07-18:
The file format for ACODES isn't secret, it's all written by the psglib, but not really documented anywhere (publicly afaik).  Changing the compiler shouldn't change the ACODES (unless there's a bug in the psglib).  Basically it's a hodgepodge of assembly-like instructions for experiment execution; I suspect they're executed directly on an FPGA.

Would be a cool undergrad engineering project to build a compatible spectrometer.

That part is easy enough to figure out, will be even easier the second time around.  IIRC, there are a few system-dependent codes though -- that only work, ie, on DDR2, so it's not a single language.  Which flavor the psglib spits out is dependent on the system configuration (again, IIRC).

A stronger statement is that the ACODES for an experiment should be exactly the same on any system, given that the experimental parameters and configured spectrometer are the same.

timburrow, 2016-07-19:
src/xracq/acodes.h has some information on ACODES.
also checkout src/vwacq that runs on the console.

You are right: the ACODES for an experiment should be exactly the same on any system, given that the experimental parameters and configured spectrometer are the same. There is a set of ACODES; I think one binary file per controller in the console (each RF channel, gradient controller, Lock transceiver, Receiver(s), Master (that controls shim, spin, temperature etc)).

A TODO: write a simple parser that converts the binary ACODES to ASCII and set up a unit test to generate ACODES