OpenVnmrJ/OpenVnmrJ #18: Packaged releases and flexible installation location

tesch1, 2016-07-18:
I know this is a pie-in-the-sky ticket!

It would be great to have versioned .rpm/.deb/.dmg/MacPorts(?)/(and whatever windows does) packages  so that versioned updates/downgrades can be managed properly.  Moving away from the "patch-level" approach to upgrades.

And it would be nice to be able to install multiple versions of OpenVnmrJ side-by-side.

Let this ticket be a place to collect tasks toward this dream ;o)

1. dont rely on /vnmr as the install location anywhere.  Maybe/probably have an environment variable ($vnmrsystem ..?) when looking for the install root of whatever's currently executing.

timburrow, 2016-07-19:
It is an interesting (and I think a good) idea. It would be easier to install an new version beside an old version rather than patching a system. The packaged versions are actually quite small as the data heavy bits were not released by Agilent.
I think the patches were a response to:
1. systems where the user made changes within the /vnmr hierarchy such as macros or UX and didn't want to replicate these changes in a new system. Now, with appdirs, this is no longer a worry. But there should be some mechanism to use local appdirs. 
2. the VnmrJ 4.2 installs were distributed on DVD and weighed several GB. The OpenVnmrJ distribution is not large and easily downloaded.

Your second point:
There is no reason $vnmrsystem could be used to have to path the install location but /vnmr is hard coded throughout the software. I think it is better not to rely on /vnmr, but it would be a big task (herculean) to free us from that link. 
Major users of /vnmr are:
1. dynamic libraries
There are several shared libraries that are in /vnmr/lib and having them move would cause problems with the binaries that call the nmr shared libraries.
I am setting up the macOS version to have the entire /vnmr to be inside the application bundle but I use a helper to link /vnmr to the location within the bundle when the binary launches. All dynamic libraries must be within the application bundle on macOS Sierra, otherwise a malicious external library could subvert an internal library.
 Of course, a shell variable  could be used so dynamic linked libraries could be found: LD_LIBRARY_PATH can be set on Linux and DYLD_FALLBACK_FRAMEWORK_PATH on macOS
Perhaps some changes to Java to set the library path might need to be done (I'm not sure)

2.  There are many hard coded "/vnmr" throughout. A quick look shows all the system "appdirs" have /vnmr in the path, many shell scripts have /vnmr, the path in all the tcl scripts is /vnmr/tcl/bin.vnmrWish and so on.

I am afraid to see how difficult it would be if /vnmr is assumed not to exist and $vnmrsys be used going forward.

tesch1, 2016-07-19:
Yeah, it's everywhere.  At the very least it should be /opt/vnmr/

Some background for RedHat: https://fedoraproject.org/wiki/Packaging:Guidelines#Filesystem_Layout
And on Macos, /opt is where MacPorts and X11 go, so that's maybe ok there too.

Totally agree that the system-specific stuff should be isolated from the upgradeable software "package".

If there were proper test coverage, it wouldn't be that herculean, rather just pedantic. Should be mainly a matter of figuring out the right way to replace the hard-coded '/vnmr" in all the contexts in which it appears (code/scripts/magical/...), and doing that and then checking  (via aforementioned testing) that they all have been caught.

tesch1, 2016-12-20:
I've started working on this.  My plan is to do it in stages:
1. rework the installation procedure so that the current install can be scripted within a .deb/.rpm/.pkg
2. create .deb/.rpm/.pkg packages for releases.
3. automate installation testing on the various "supported" targets- rh6,7, debian, ubuntu14,16
4. much further down the road, figure out (how to / whether to) allow multiple versions run at the same time / activate a particular version (change symlink, restart servers..) -- it's a use-case that makes sense for datastations, but not really for spectrometer control.  Maybe there should be a more explicit "datastation" installation for linux?

Any thoughts on this plan?  It probably means replacing the existing java installation program that appears to mostly just be a wrapper around install scripts anyhow, although I'm not sure yet what will crop up.

timburrow, 2016-12-22:
That's a great goal. Are you thinking of rewriting the java app or writing new scripts? 

I wrote an alternative installer for OS X, but is was in Objective-C. I'll take the holiday break to clean up and push to GitHub. As you said, it just called scripts (and checked for successful completion).

tesch1, 2016-12-23:
Haven't worked out all the details, so I'm not sure yet which approach is best... so I'm open for discussion...  I want it to be clean and lower the barrier of entry; the current installation is somewhat arcane and doesn't match common expectations about modern software installation (ie, that I can just type `yum install OpenVnmrJ` and then be able to run vnmrj...).

Step one will be understanding the intricacies of java program, and abstract the option-selection from the install scripts.. sounds like you've already done that?  It would be really helpful to see your ObjC installer.

If the java is just mostly deciding what files get unpacked, one possible approach would be to have separate packages (.rpms/.debs/checkboxes in macos installer) for each "feature set" that has those files:
OpenVnnmrJ-base (basics for pure datastation)
OpenVnnmrJ-spectrometer (additions to control spectrometer)
OpenVnnmrJ-docs 
OpenVnnmrJ-solids
OpenVnnmrJ-imaging
...