OpenVnmrJ/OpenVnmrJ #19: Style guide

tesch1, 2016-07-19:
Is there a style guide?  I dont see anything under "contributing" or anywhere I've looked in the docs, maybe I'm not looking in the right place?

If there's not, maybe adopt an existing modern style guide, something like google's, or one for c++ and one for java.

For example:
https://google.github.io/styleguide/cppguide.html
https://google.github.io/styleguide/javaguide.html

timburrow, 2016-07-19:
Another great idea. 👍  

I thought a little about this. Some of the companies around here enforce style by running a lint or [uncrustify](https://github.com/uncrustify/uncrustify) on code on check in. But that has to be new code only, I don't want to touch older code just yet...

It's just you and me changing code so far, so I'll read over the style guides and let's choose one.

tesch1, 2016-07-19:
I think anything that a bigger organization / open-source project is using is good.  People have worked through this issue and have put a lot of effort into figuring out reasonable things, just have to pick one and stick with it. And document it.

should also cover python for all the scons code.
and tcl.
and magical.