OpenVnmrJ/OpenVnmrJ #20: Remove dependence on 32-bit compilers / packages

tesch1, 2016-07-19:
More on the engineering wish-list!

There shouldn't be any dependence on 32-bit compiler flags and libraries. (?)

At least this involves:
- cleaning up all the c/c++ types that are size-nonspecific but matter for communication. (there's some of this in psglib at least -- a few years ago I cleaned these up and had a fully-working psglib without -m32.)
- ?

timburrow, 2016-07-19:
@DanIverson did update the some of the code to 64-bit, and I believe it compiles OK. If there weren't thousands of warnings (e.g. loss of precision from implicit casting) this won't be so hard to do.

Communication does require byte-size specify types.  It would be nice to have hardware to test on. I worry that m68k consoles might be 32-bit whereas ppc might be 64-bit. Maybe @DanIverson knows.

Do you have a console to test on? If not, UO has been testing on older hardware.

tesch1, 2016-07-19:
No, I dont have a console.

Warning cleanup is another big project, probably best done as an ongoing side skirmish.  But I'm hesitant to make any sweeping changes until there's some sort of testing process (other than "hope someone tests it on their console" -- thus the virtual scanner idea #17 .)

timburrow, 2016-07-19:
I agree. Unit tests + ACODE tests are a first order of business. In the last Agilent release, some testing was included in vjqa. It appears to have some unit tests for java and a "QA" routine to test on hardware.