OpenVnmrJ/OpenVnmrJ #22: Package lesstif2 required for Ubuntu install, but not available on Trusty nor Xenial

timburrow, 2016-07-29:
openmotif replaces the open source clone, lesstif2, on Ubuntu Trusty an newer, see https://wiki.debian.org/lesstif2motifTransition

Furthermore, installation on Ubuntu Trusty (14.04) fails with a message to install packages from /tmp/agilent_preinstall/installpkgs. The directory /tmp/agilent_preinstall/ is created but empty (installpkgs missing).

tesch1, 2016-09-08:
I ran into this too, I think the installer shouldn't necessarily fail if it doesn't find the packages in the list, since it doesn't actually have access to the packages that were used to build the binaries.

My stop-gap was to add an environment var check to the installer QUIET_INSTALL=y that prevents the installer from dying if it doesn't find all the packages.

tesch1, 2016-10-18:
Can this be closed?

osparking, 2016-12-02:
Hello tesch1,

I'm trying to install OpenVnmrJ on my Ubuntu 14.04.5 (VirtualBox client).
However, I'm having 
"https://github.com/timburrow/openvnmrj-source/blob/master/src/scripts/rpmInstruction.txt"
error.
Would you eloborate on "add an environment var check to the installer QUIET_INSTALL=y"?
I have not been using Ubuntu almost a year. So, I'm kind of novice. Please help me.

tesch1, 2016-12-02:
That repo is marked as "to be deleted", you probably dont want to use it.

I'd suggest trying the VM-based builds, which should "just work", but so far only available in the 'development' branch (scroll down in the README for instructions) :
https://github.com/OpenVnmrJ/ovjTools/tree/development
If you'd rather try to do everything piecemeal, there is a full list of required packages for ubuntu in the Vagrant box_defs files under here:
https://github.com/OpenVnmrJ/ovjTools/tree/development/vms/box_defs

osparking, 2016-12-02:
Actually, the repo (that is to be deleted) was not that I was using.
I was using development branch already.
How may I use/run Vagrantfile under "box_defs"?

tesch1, 2016-12-02:
(scroll down in the README for instructions) :
https://github.com/OpenVnmrJ/ovjTools/tree/development

osparking, 2016-12-03:
Hello,

I just uploaded two issues on OpenVnmrJ. Please, check those out when you
have time.
Talk to you later,

JB

DanIverson, 2019-07-30:
This should be fixed with PR #277

closed 2019-07-30 by DanIverson