OpenVnmrJ/OpenVnmrJ #23: Package ia32-libs required for install on Ubuntu (64-bit) but not available on Trusty or Xenial

timburrow, 2016-07-29:
Installation fails when installing on Ubuntu Trusty Tahr (64-bit) with a requirement for the package ia32-libs. No such package exists.  

I am not sure it is needed, multi arch allows one to specify the i386 version of libraries, and libc6 has a 32 bit version installed:  
```
apt list --installed | grep 386

WARNING: apt does not have a stable CLI interface yet. Use with caution in scripts.

libc6-dev-i386/unknown,unknown,now 2.19-0ubuntu6.9 amd64 [installed]
libc6-i386/unknown,unknown,now 2.19-0ubuntu6.9 amd64 [installed,automatic]
libmotif-dev/unknown,now 2.3.4-5ubuntu0.1 i386 [installed]
```
Perhaps more 32-bit libraries are needed.

timburrow, 2016-07-29:
See also issue #20—remove dependences on 32-bit libraries

bjohnmr, 2016-07-29:
Tim,
I am not sure if this package is necessary. 

However, I found this useful link-
http://stackoverflow.com/questions/23182765/how-to-install-ia32-libs-in-ubuntu-14-04-lts-trusty-tahr

-Boban

tesch1, 2016-08-24:
Using Boban's link, it gets pretty far on trusty (albeit- haven't finished the build yet), you can probably close this issue.  ia32-libs just doesn't exist on trusty.

tesch1, 2016-09-08:
possible fix for this and #22 like this: 216e6c52b9ad6b12145b5d489e74ea2ca342f717

tesch1, 2016-10-18:
Can this be closed?

bjohnmr, 2016-10-18:
Yes.Boban

PR #55 (2016-09-27): update package check for trusty 14.04
this fixes #23 and #22
closed 2019-07-30 by DanIverson