OpenVnmrJ/OpenVnmrJ #26: Failed to build on Ubuntu 14.04.2 LTS

DanIverson, 2016-08-09:
Boban reported that buildovj ran into problems with the sconsPostAction script in vnmr/manuals. It seemed to go into a loop trying to make a soft link from AC1 to AC9.

timburrow, 2016-08-09:
I built on Ubuntu 14.04.4 with no problem with the soft links. 

I did run into problems with 
- Postgresql not installed 
- User name (vnmr1) not in /vnmr/userlist
- VnmrJ admin not starting (due to above)

closed 2016-08-10 by DanIverson

neuroglobin, 2016-10-18:
Hello, was this issue really resolved? I built on Ubuntu 14.04 with the most recent version (https://github.com/OpenVnmrJ/ovjTools) and met all three problems timburrow described on Aug 9.

tesch1, 2016-10-18:
Which branch are you building?

neuroglobin, 2016-10-18:
I checked out ovjTools by `git clone https://github.com/OpenVnmrJ/ovjTools.git` and it was the master branch. 
The most recent commit shown by `git log` is 

```
commit c54e6936519adb068e266aa0640d3c9e981a0a28
Merge: d21594a 3ad8880
Author: Timothy Burrow <timburrow@users.noreply.github.com>
Date:   Mon Aug 29 17:13:21 2016 -0400

    Merge pull request #12 from timburrow/master

    Check of case insensitive experiments; git pull doesn't pull only master
```

tesch1, 2016-10-19:
Hm, maybe a more detailed bug report would be helpful.  Are you running into the link problem, or just the three issues that Tim ran into?

For postgres, you just have to install it(?)  There is a small change to the install script that fixes a dir permission in order to run it, but I think that change is still in the 'developement' branch somewhere.

For username not in vnmrlist, I'm not sure. Maybe open a new issue with more details if the 'development' branch doesn't work.  I'd suggest using one of the Vagrant builds as a reference, since then it'll be an exactly reproducible environment.

neuroglobin, 2016-10-19:
Thanks for the quick response. 
It seems that I did not have the link problem, at least I did not pay attention to it.  Although _the postal not installed problem_ happened, the installation was still successfully done. 
For me the fatal problem is the userlist problem (Error message attached). In my `/vnmr/adm/users` folder there was no `userlist` file, but only a `userlist.xml` file, which does not contain any username (in my case, I used the default `vnmr1`) . 
[image]
Running `vnmrj adm` did not help. I will try the `development` branch. Thanks.

tesch1, 2016-10-19:
Yeah, my response was quick but I was being a bit hasty :)  Yes, unfortunately there's not (yet) always a clear failure immediately when things fail.

The fix for postgres is here:
https://github.com/OpenVnmrJ/OpenVnmrJ/commit/cd2274700f0a421fadf2c788acd68d499a3bac0f

You could try installing postgres, do the chmod needed to run postgres as a user (chmod a+rw /var/run/postgresql) and then re-installing.

DanIverson, 2016-10-19:
The userlist file is created by dbsetup called during the install process. There is a log file at /vnmr/adm/log/vnmrj<date>. Does it show any errors? This might be related to issue #70

neuroglobin, 2016-10-19:
Thanks tesch1 and  Danlverson. 

It looks that the `cd22747` is already fixed in my scripts. I installed postgres and did `chmod a+rw /var/run/postgresql` and then tried re-installing. 
However this time the installation wouldn't be done after the following error happened: 

```
Success. You can now start the database server using:
WARNING: enabling "trust" authentication for local connections

You can change this by editing pg_hba.conf or using the option -A, or
    /usr/lib/postgresql/9.3/bin/postgres -D /home/openvnmrj_1.1_A/pgsql/data
--auth-local and --auth-host, the next time you run initdb.
or
    /usr/lib/postgresql/9.3/bin/pg_ctl -D /home/openvnmrj_1.1_A/pgsql/data -l logfile start

server starting

Creating /home/openvnmrj_1.1_A/templates/vnmrj/choicefiles/pis
java.sql.SQLException: FATAL:  database "vnmr" does not exist

Recreating 1 user(s)...
ERROR Oct 19, 2016 16:42:31 makeDBConnection failed on ubuntu port 5432,
     Adding vnmr1
   be sure database exists and that vnmr1 has access to it.
Loading data for all users (this may take a moment)...
   Try having the system administrator execute the command 'createuser vnmr1'.
   Also check that if a special port number  is used for postgres,
   that this user has the environment variable 'PGPORT' set  correctly.
   Is it possible that a specific error can be found in the log file:
      /vnmr/pgsql/pgsql.log
java.sql.SQLException: FATAL:  database "vnmr" does not exist

ERROR Oct 19, 2016 16:42:31 makeDBConnection failed on ubuntu port 5432,
```
The `/vnmr/pgsql/pgsql.log shows 
```
LOG:  database system was shut down at 2016-10-19 16:24:50 CDT
LOG:  MultiXact member wraparound protections are now enabled
LOG:  database system is ready to accept connections
LOG:  autovacuum launcher started
FATAL:  database "vnmr" does not exist
FATAL:  database "vnmr" does not exist
FATAL:  database "vnmr" does not exist
FATAL:  database "vnmr" does not exist
FATAL:  database "vnmr" does not exist
FATAL:  database "vnmr" does not exist
FATAL:  database "vnmr" does not exist
FATAL:  database "vnmr" does not exist
FATAL:  database "vnmr" does not exist
FATAL:  database "vnmr" does not exist
FATAL:  database "vnmr" does not exist
FATAL:  database "vnmr" does not exist
FATAL:  database "vnmr" does not exist
FATAL:  database "vnmr" does not exist
```

I've attached the log files mentioned above.

[log.zip](https://github.com/OpenVnmrJ/OpenVnmrJ/files/540439/log.zip)

Btw, `net stat -nlp|grep "5433"` shows
```
 will not be shown, you would have to be root to see it all.)
tcp        0      0 0.0.0.0:5432            0.0.0.0:*               LISTEN      -
tcp6       0      0 :::5432                 :::*                    LISTEN      -
unix  2      [ ACC ]     STREAM     LISTENING     49442    -                   /var/run/postgresql/.s.PGSQL.5432
```
Is this normal? 

Thanks!

DanIverson, 2016-10-20:
Most of the errors in the log file can be attributed to loading a new OVJ on top of an old installation. At some point, OVJ should probably be fixed to disallow that. One error is related to installing rsh. Does this Ubuntu version not have a /etc/xinetd.d directory? Where would be the proper place for the rsh script?

There seems to be a problem with the Ubuntu postgres program. See issue #70 for some details. 

Could you try an updated dbsetup? First, remove the /vnmr link and the /home/openvnmrj_1.1_A directory. With your existing dvdimageOVJ directory, replace the code/dbsetup script with the one in the attached zip file. Then try the ./load.nmr again.
[dbsetup.zip](https://github.com/OpenVnmrJ/OpenVnmrJ/files/540542/dbsetup.zip)

neuroglobin, 2016-10-21:
Thanks Danlverson for your suggestions.  
Yes my Ubuntu version did not have a `/etc/xinetd.d` directory. Therefore I ran the command `sudo apt-get install xinetd`  
I removed `/vnmr`/, `/home/openvnmrj_1.1_A` directories, update `dbsetup` that you attached (I had to modify line 1021 from `chmod a+rw /var/run/postgresql` to `sudo chmod a+rw /var/run/postgresql`). 
Then try the `./load.nmr`. The installation went on well except an error message warning `Database postmaster daemon failed to start!`  However, several minutes later the OpenVnmrJ GUI launched. 
This time I don't see any log file named as /vnmr/adm/log/vnmrj{date}. The VnmrjInstallationScript.log looks fine. There is only 1 line in the pgsql.log
```
FATAL:  unrecognized configuration parameter "unix_socket_directory"
```
I was able to see such GUI, however not sure whether the postmaster daemon failure would cause any problems. I would be happy if it is able to visualize and process Agilent/Varian spectra. 
[image]
Thanks!

tesch1, 2016-10-21:
@neuroglobin as a follow up for the postgresql issue (#70), could you check the version of postgresql on your system?  by runing `dpkg -s postgresql postgresql-9.3`  the version installed on my machine (which apparently set the perms of /var/run/postgresql correctly) is 9.13.14-0ubuntu0.14.04 for amd64.

Thanks!

neuroglobin, 2016-10-26:
@tesch1 Sorry I missed this post. Here is the log. Thanks. 
```
vnmr1@ubuntu:~$ dpkg -s postgresql postgresql-9.3
Package: postgresql
Status: install ok installed
Priority: optional
Section: database
Installed-Size: 71
Maintainer: Ubuntu Developers <ubuntu-devel-discuss@lists.ubuntu.com>
Architecture: all
Source: postgresql-common (154ubuntu1)
Version: 9.3+154ubuntu1
Depends: postgresql-9.3
Description: object-relational SQL database (supported version)
 This metapackage always depends on the currently supported PostgreSQL
 database server version.
 .
 PostgreSQL is a fully featured object-relational database management
 system.  It supports a large part of the SQL standard and is designed
 to be extensible by users in many aspects.  Some of the features are:
 ACID transactions, foreign keys, views, sequences, subqueries,
 triggers, user-defined types and functions, outer joins, multiversion
 concurrency control.  Graphical user interfaces and bindings for many
 programming languages are available as well.
Original-Maintainer: Debian PostgreSQL Maintainers <pkg-postgresql-public@lists.alioth.debian.org>

Package: postgresql-9.3
Status: install ok installed
Priority: optional
Section: database
Installed-Size: 11384
Maintainer: Ubuntu Developers <ubuntu-devel-discuss@lists.ubuntu.com>
Architecture: amd64
Version: 9.3.14-0ubuntu0.14.04
Depends: libc6 (>= 2.16), libcomerr2 (>= 1.01), libgssapi-krb5-2 (>= 1.8+dfsg), libkrb5-3 (>= 1.6.dfsg.2), libldap-2.4-2 (>= 2.4.7), libpam0g (>= 0.99.7.1), libpq5 (>= 9.1~), libssl1.0.0 (>= 1.0.0), libxml2 (>= 2.7.4), postgresql-client-9.3, postgresql-common (>= 142~), tzdata, ssl-cert, locales
Suggests: oidentd | ident-server, locales-all
Description: object-relational SQL database, version 9.3 server
 PostgreSQL is a fully featured object-relational database management
 system.  It supports a large part of the SQL standard and is designed
 to be extensible by users in many aspects.  Some of the features are:
 ACID transactions, foreign keys, views, sequences, subqueries,
 triggers, user-defined types and functions, outer joins, multiversion
 concurrency control.  Graphical user interfaces and bindings for many
 programming languages are available as well.
 .
 This package provides the database server for PostgreSQL 9.3. Servers
 for other major release versions can be installed simultaneously and
 are coordinated by the postgresql-common package. A package providing
 ident-server is needed if you want to authenticate remote connections
 with identd.
Homepage: http://www.postgresql.org/
Original-Maintainer: Debian PostgreSQL Maintainers <pkg-postgresql-public@lists.alioth.debian.org>
```

FredericPerras, 2016-12-08:
My installation gets stuck at 95% with the following text in the log:

WARNING: enabling "trust" authentication for local connections
Success. You can now start the database server using:
You can change this by editing pg_hba.conf or using the option -A, or

--auth-local and --auth-host, the next time you run initdb.
    /usr/lib/postgresql/9.3/bin/postgres -D /home/openvnmrj_1.1_A/pgsql/data
or
    /usr/lib/postgresql/9.3/bin/pg_ctl -D /home/openvnmrj_1.1_A/pgsql/data -l logfile start

server starting
 
Database postmaster daemon failed to start!

I have all the required packages and postgres installed.