OpenVnmrJ/OpenVnmrJ #41: Build fails after last pull request: missing craft

timburrow, 2016-08-29:
On OS X, get error:
Copy("/Volumes/Tim/Users/timburrow/Documents/OpenVnmrJ/github-timburrow/OpenVnmrJ/src/vnmrj/../../../vnmr/java/simplemovie.jar", "/Volumes/Tim/Users/timburrow/Documents/OpenVnmrJ/github-timburrow/ovjTools/JMF-2.1.1e/src/simplemovie.jar")
gsl library not found. Skipping compiles requiring that library
OSError: [Errno 2] No such file or directory: '/Volumes/Tim/Users/timburrow/Documents/OpenVnmrJ/github-timburrow/OpenVnmrJ/../vnmr/craft':
  File "/Volumes/Tim/Users/timburrow/Documents/OpenVnmrJ/github-timburrow/OpenVnmrJ/SConstruct", line 279:
    runSconsPostAction(os.path.join(vnmrPath, 'craft'))
  File "/Volumes/Tim/Users/timburrow/Documents/OpenVnmrJ/github-timburrow/OpenVnmrJ/SConstruct", line 270:
    dirList = os.listdir(dir)
scons returned: 2
===>>> scons Build error  <<<===

DanIverson, 2016-08-29:
I just re-built the development branch of OVJ on my Mac without any errors. What happens if you remove or rename github-timburrow/vnmr and then go to OpenVnmrJ/src/common and run scons? Does that build a github-timburrow/vnmr/craft directory?

timburrow, 2016-08-29:
It's seems common/SConstruct is missing on the repo (timburrow/OpenVnmrJ, development branch) when it is pulled. It is the development branch on GitHub, but not master, so perhaps a bug in how the development branch is checkout out?

timburrow, 2016-08-29:
This is what I have in OpenVnmrJ as pulled by running makeovj:
git status
On branch development
Your branch is ahead of 'origin/development' by 22 commits.
  (use "git push" to publish your local commits)
Changes not staged for commit:
  (use "git add/rm <file>..." to update what will be committed)
  (use "git checkout -- <file>..." to discard changes in working directory)

	deleted:    SConstruct
	deleted:    ../macos/.gitignore
	deleted:    ../vjqa/vjtest/maclib.futureTests/strstrTest2

no changes added to commit (use "git add" and/or "git commit -a")

gitSSH='ssh://git@github.com/timburrow/OpenVnmrJ.git'
gitBranch='development'

timburrow, 2016-08-29:
🔨 doGitClone=pull ignores the branch in makeovj:
  59  if [ x$doGitClone = "xpull" ]
 60   then
 61     cd $buildDir/OpenVnmrJ
 62     rm -rf src/* # src is mixed with copied files and build artifacts
 63 #   git checkout src
 64     echo "git fetch origin master"
 65 #   git pull src
 66     git fetch origin master
 67     git checkout origin/master -- src
 68     git pull origin master # on CentOS's older git above doesn't work
 69   fi

DanIverson, 2016-08-29:
The 3 deleted files in your repo were all added to OpenVnmrJ/OpenVnmrJ master and development branches 6 days ago (commit #32). The missing common/SConstruct was added 19 days ago (commit #29) I still do not understand github's pull requests. I made a PR 4 days ago (go('check') was failing on MacOS) and it was merged into OpenVnmrJ. I just made another PR for the MacOS PSG files and the go('check') commit re-appears in this new PR.

For makeovj, the pull option does not look correct if one is on the development branch. Could it not just be
rm -rf src
git checkout src
git pull
That would maintain the current branch one is on.

timburrow, 2016-08-31:
Updated makeovj, works OK now

closed 2016-08-31 by timburrow