OpenVnmrJ/OpenVnmrJ #54: detect if VnmrJ4.2 is running and offer to switch automatically

tesch1, 2016-09-22:
apropos https://spinsights.chem.agilent.com/docs/DOC-12771

timburrow, 2016-09-22:
💯 
The script is Apache licensed now, so it can be included and used in the install script.

It'll be part of the openvnmrj-utils appdir

tesch1, 2016-09-22:
Ah, cool!

(somewhat off topic..)
I'm a little confused as to what all the git repos are supposed to do?

It's the first project I've seen that's organized this way, so it's taking me a bit to wrap my head around....

Am I understanding right that this the intended directory layout:

ovjbuild/
ovjbuild/ovjTools
ovjbuild/OpenVnmrJ
ovjbuild/(other buid staging directories)
ovjbuild/appdirs...

with environment variables:
ovjBuildDir=../path/to/ovjbuild
OVJ_TOOLS=$ovjBuildDir/ovjTools

?

tesch1, 2018-08-09:
covered by #18

closed 2018-08-09 by tesch1