OpenVnmrJ/OpenVnmrJ #67: Install on Ubuntu Xenial gives error, no script made

timburrow, 2016-09-28:
When trying to install on Ubuntu Xenial (16.04.1) I got the following error message:
```
 sudo ./load.nmr 
[sudo] password for timburrow: 
VnmrJ required Ubuntu package "openjdk-6-jre" NOT installed
VnmrJ required Ubuntu package "csh" NOT installed
VnmrJ required Ubuntu package "libmotif-dev" NOT installed
VnmrJ required Ubuntu package "expect" NOT installed
VnmrJ required Ubuntu package "tcl8.4-dev" NOT installed
VnmrJ required Ubuntu package "tk8.4-dev" NOT installed
VnmrJ required Ubuntu package "rarpd" NOT installed
VnmrJ required Ubuntu package "rsh-server" NOT installed
VnmrJ required Ubuntu package "tftpd" NOT installed
VnmrJ required Ubuntu package "openssh-server" NOT installed
VnmrJ required Ubuntu package "mutt" NOT installed
VnmrJ required Ubuntu package "sharutils" NOT installed
VnmrJ required Ubuntu package "sendmail-cf" NOT installed
VnmrJ required Ubuntu package "k3b" NOT installed
VnmrJ required Ubuntu package "kdiff3" NOT installed
 Standby........ 
chmod: cannot access '/tmp/agilent_preinstall/*': No such file or directory
```
A directory /tmp/agilent_preinstall is made, but empty.

Since Ubuntu is a data station only, tftpd, rarpd, rsh-server are not necessary as those are required for console communication.

tesch1, 2016-09-28:
Every supported platform should have a list of packages needed for

- building
- installing datastation
- installing console

There's a start in the Vagrant scripts, and there's more lists scattered about (John's kickstarter script, the current install scripts).  Should be cleaned up and unified, with one script to install/check any combination of (at least) ...  (any others?)

| OS | Build | Datastation | Console |
| --- | --- | --- | --- |
| el6 | x | x | x |
| el7 | x | x | ? |
| ubuntu12 | x | ? | ? |
| ubuntu14 | x | x | ? |
| ubuntu16 | x | x | ? |

DanIverson, 2016-09-28:
If you have a VJ42 DVD, there is a directory code/linux. This contains the
files that are copied to /tmp/agilent_preinstall to install packages. I
never added those to OVJ. I am not sure which are required nor compatible.
For example, I know that John Ryan recommends against installing nvidia
drivers. The files in that directory may well be out of date.

DanIverson, 2016-09-28:
A subset of those files are copied to /vnmr/adm/linux in VJ42. Those are
probably the important ones.

closed 2019-07-30 by DanIverson