OpenVnmrJ/OpenVnmrJ #68: Need a better way to indicate updates (and some ideas for updates)

timburrow, 2016-09-28:
We could have updates on a regular basis, but the current naming "OpenVnmrJ 1.1A" reflects the longer update cycles when the application was distributed from Agilent on DVDs. Now we can have monthly updates. 58 people downloaded the previous version on OpenVnmrJ, which was marked as "prerelease". Maybe they'll come back to check for updates? 

I can think of some options, but any other ideas are welcome
 1. Use OpenVnmrJ 1.1A, OpenVnmrJ 1.1B... for each minor update
 2. Use OpenVnmrJ 1.1A.1, OpenVnmrJ 1.1A.2... for each minor update
 3. Use OpenVnmrJ 1.1.1A, OpenVnmrJ 1.1.2A... for each minor update

Perhaps there can be a roadmap of updates? Some ideas:
 1. Add covariance processing
 2. Add IST style NUS processing so NMRPipe is not necessary
 3. Add translations for other languages (I started a translation setup on https://poeditor.com)
 4. Add a Bruker/JEOL to Agilent pulse sequence converter
 5. Add a check to GitHub to see if a new release is available, and if so, alert the user
 6. Check for Chempack updates

If these are of interest they could be made into separate issues.

Also, we should have a "human readable" list of what was updated for each version, not just a git log.

tesch1, 2016-10-20:
Regarding versioning, I'd tend toward just keeping it simple (#1 from the above list... A,B,C...).  It seems like there are only three people actively looking at code, so the less overhead the better.

Along with the philosophy "Release often and make upgrades painless."

Roadmap idea:
0. (Comprehensive) automated regression testing

re: 4, it's probably easier to just compile other pulse sequence languages directly into acodes

tesch1, 2016-10-26:
- Another killer-app for OpenVnmrJ would be the ability to browse/load/process Bruker/JEOL datasets.

DanIverson, 2016-10-26:
The ability to read and process Bruker data (convertbru, src/bin/convertbru.c) has been a part of VJ for a long time.  I was able to add something for JEOL data (rtjeol, src/common/maclib/rtjeol) in the 109 - 110 patch