OpenVnmrJ/OpenVnmrJ #70: dbsetup.sh fails on trusty tahr14

tesch1, 2016-09-29:
/home/openvnmrj_1.1A/pgsql/pgsql.log says:

FATAL:  could not create lock file "/var/run/postgresql/.s.PGSQL.5432.lock": Permission denied

tesch1, 2016-09-29:
needs a "chmod a+rw /var/run/postgresql" somewhere

DanIverson, 2016-10-19:
This link ( http://askubuntu.com/questions/442295/postgresql-server-installation-keeps-failing-due-to-postgresql-common-error ) suggests a fix. However, since the postgresql.conf file is in /vnmr/pgsql/data and that directory is removed by dbsetup, we will need to fix dbsetup and /vnmr/bin/S99pgsql. On line 766 of dbsetup that has the call to pg_ctl, the -o option should append one more argument so it goes from
   -o "-N 10 -B 45 -i -c sort_mem=100000"
 to
   -o "-N 10 -B 45 -i -c sort_mem=100000 --unix_socket_directory=/tmp"
Could someone test this.

DanIverson, 2016-10-19:
Probably all calls to pg_ctl that use the sort_mem option should also be appended with the unix_socket_directory option

tesch1, 2016-10-20:
Even though the (new) chmod seems to be erroring, I dont see this
particular error anymore,
on Ubuntu 14.04.5.  The exact setup is reproducible with a virtual machine:

git clone --branch vagrant-builds git@github.com:tesch1/ovjTools.git
export OVJ_TOOLS=`pwd`/ovjTools
${OVJ_TOOLS}/bin/build_vm.sh ubuntu14 --branch development
cd ${OVJ_TOOLS}/vms/ubuntu14
VMGUI=y vagrant reload                 # this will reboot the VM in
graphical mode

(login)

xhost +
... ./load.nmr

tesch1, 2016-10-20:
fixed, at least in 14.04.5

closed 2016-10-20 by tesch1