OpenVnmrJ/OpenVnmrJ #78: Concise Instruction for a Beginner needed.

osparking, 2016-12-03:
Well, I'm new to VnmrJ completely, as a matter of fact. I heard the word itself only a couple of days ago.
Still I want to get at this kind of status (see figure below.)

[image]

These are my current position.
- Desktop computer running Windows 7
- Installed VirtualBox on Win7, and Ubuntu 14.04 machine in VBox.

What I don't know:
1. The relationship of Vagrant and buildovj.

What I need:
1. Rough (maybe 5 to 10) (1 line) steps to arrive at the above image(I don't know what to call it, maybe desktoop?)

End.

tesch1, 2016-12-05:
What are you trying to accomplish with OpenVnmrJ?  Do you have some NMR data you want to process, or do you have a magnet?  By to the image above, it looks like you have already successfully installed OpenVnmrJ.

Vagrant is a tool to manage the creation and setup of virtual machines.  You can find more information with google.

The build script for building OpenVnmrJ in a Vagrantfile-described VM is build_vm.sh, which calls build_box.sh to setup the VM.  buildovj is not used by these scripts.

Have you installed Vagrant?  Have you tried the steps from the 'development' branch README for building "In a Virtual Machine"?  What happened?

osparking, 2016-12-05:
Last week a physics professor contacted me asking if I could customize
OpenVnmrJ.
And now I'm studying on it to make sure if I can handle the task.

The image was sent from the professor to me as a reference.
Now I am trying to compile and load OpenVnmrJ not using Vagrant.

If buildobj were createing a log of 10M byte, is it normal?

Thanks,
JB

tesch1, 2016-12-05:
It would be nice to improve the build process.  I've tried to make some steps in that direction, for example by abstracting the build platform from the build via the Vagrant layer.  I'm sure any contributions you can make to improving the situation would be appreciated.

The source code is chock full of warning messages, due to a number of factors, one being that the original code is >30 years old, and C was a somewhat different language back then.  Two, compiler warnings have changed continuously over the last 30 years.

osparking, 2016-12-05:
All right, but I'm not sure if I would take this task. I'll decide on it
before the end of this year.
So, "C" is the main language I need to recall to be able to customize
OpenVnmrJ (its GUI, etc.) right?

DanIverson, 2016-12-05:
OpenVnmrJ (OVJ) can be customized at several different levels. The GUI part uses java and xml. If you just want to customize the interface with modified panels or menus, you can do that without modifying the java code. There is a built-in panel editor. Perhaps if we had a better idea of the changes your physics professor wants to implement we could direct you to the correct tools.

osparking, 2016-12-05:
Very nice, let me take a note on it.

DanIverson, 2016-12-05:
Also, if you just want to get it running in virtualbox, follow the installation instructions here https://github.com/OpenVnmrJ/OpenVnmrJ/releases

osparking, 2016-12-06:
> OpenVnmrJ/releases

I followed steps in https://github.com/OpenVnmrJ/OpenVnmrJ/releases.
Especially, the below part

- *tar jxf dvdImage_OpenVnmrJ_1.1A_DD2.tbz*
- *cd dvdimageOVJ*
- *./load.nmr*
However, when I clicked Install Button at the bottom, I had the 'postgress'
error show in the image here:
(I also attached the log file to this mail. Could you please give me some
advice?

[image: Inline image 1]

DanIverson, 2016-12-06:
Neither the image nor log  file seem to be attached to your last message

osparking, 2016-12-06:
It's strange. I sent both of them in my above mail (I have a proof.)
However, git may doesn't work that way. So, let me try that again, this
time on the github site directly.

JB

DanIverson, 2016-12-06:
You can probably ignore the postgres error. Login in as the vnmr1 account. There should be a OpenVnmrJ or VnmrJ icon on the desktop that you can double-click. Or you can open a terminal and enter vnmrj on the command line.

osparking, 2016-12-07:
I don't have any icon on the desktop.  
(A user 'vnmr1' was created during installation I guess. I had to change it's password though.)
So, I logged in as 'vnmr1' and tried the command vnmrj as follows: 
```
vnmr1@JB-VBOX-Ubt:~/orun$ vnmrj
No protocol specified
xrdb: Permission denied
xrdb: Can't open display ':0'
vnmr1@JB-VBOX-Ubt:~/orun$ No protocol specified
Exception in thread "main" java.lang.InternalError: Can't connect to X11 window server using ':0' as the value of the DISPLAY variable.
	at sun.awt.X11GraphicsEnvironment.initDisplay(Native Method)
	at sun.awt.X11GraphicsEnvironment.access$100(X11GraphicsEnvironment.java:52)
	at sun.awt.X11GraphicsEnvironment$1.run(X11GraphicsEnvironment.java:155)
	at java.security.AccessController.doPrivileged(Native Method)
	at sun.awt.X11GraphicsEnvironment.<clinit>(X11GraphicsEnvironment.java:131)
	at java.lang.Class.forName0(Native Method)
	at java.lang.Class.forName(Class.java:171)
	at java.awt.GraphicsEnvironment.getLocalGraphicsEnvironment(GraphicsEnvironment.java:68)
	at sun.awt.X11.XToolkit.<clinit>(XToolkit.java:89)
	at java.lang.Class.forName0(Native Method)
	at java.lang.Class.forName(Class.java:171)
	at java.awt.Toolkit$2.run(Toolkit.java:834)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.awt.Toolkit.getDefaultToolkit(Toolkit.java:826)
	at vnmr.images.VnmrImageUtil.<init>(VnmrImageUtil.java:23)
	at vnmr.util.Util.getImageIcon(Util.java:591)
	at vnmr.ui.VNMRFrame.main(VNMRFrame.java:80)
 
vnmr1@JB-VBOX-Ubt:~/orun$ 
```
Did I miss any step?

DanIverson, 2016-12-07:
You are very close. The error is related to X-windows. Did to su vnmr1 or actually login as vnmr1. If you did "su", use "su - vnmr1". Also, try running "xhost +" prior to running vnmrj. I have not actually run OVJ under Ubuntu. Perhaps @tesch1 has run into this problem and knows the solution.

osparking, 2016-12-07:
I'm back and tried
`
$ ./load.nmr
`
"Postgres Not Installed" error, 
(waited 5 to 10 minutes => percentage became 100% and "Done" button appeared.

// clicked [Done] button.

I had this terminal output

[image]

// I didn't think my system is spectrometer and 

$ su vnmr1
pw: ...

$ /vnmr/bin/makeuser

I had "Updating for vnmr1 complete" on the terminal.

Now, my question is "What is the VnmrJ interface?", what do I do after that? Do you my  log?

DanIverson, 2016-12-07:
That all looks good. You are not a spectrometer so you don't need to run the **/vnmr/bin/setacq** command. Use **su - vnmr1** rather than **su vnmr1** when you switch to vnmr1 account. The **vnmrj** command should start the "VnmrJ interface". If you get the "Can't connect to X11 window" error, enter **xhost +** and then try **vnmrj** again.

osparking, 2016-12-08:
As "xhost +" didn't work. I followed next steps refering [No protocol specified](http://unix.stackexchange.com/questions/209746/how-to-resolve-no-protocol-specified-for-su-user).
```
$ export LOGIN_USER="vnmr1" 
$ su - $LOGIN_USER 
$ /vnmr/bin/makeuser 
$ sudo xhost local:$LOGIN_USER &>/dev/null 
$ vnmrj adm 
```
And the result is this,

[image]

How would I be able to define user 'vnmr1' in the userlist file?

DanIverson, 2016-12-08:
Could you do the following:
cd /vnmr/adm
zip -r adm.zip groups users
Attach the adm.zip file to an email and send it to me at daniverson2016@gmail.com

tesch1, 2016-12-08:
I strongly suspect this is something that's already fixed in the development branch.  There were some small changes to the install scripts to deal with this.  The install and startup does work on the Vagrant-described Ubuntu installs.

osparking, 2016-12-09:
Dear Iverson, let me get back to you later.

Now I'm using Vagrant and Vagrantfile that you provide for Ubuntu-14.
When I followed https://github.com/OpenVnmrJ/OpenVnmrJ/releases 
and ran
```
$ ./load.nmr
```
I had following error list. Do I need to install each package one by one?
```
vagrant@vagrant-ubuntu-trusty-64:/vagrant/project/dvdimageOVJ$ ./load.nmr
VnmrJ required Ubuntu package "openjdk-6-jre" NOT installed
VnmrJ required Ubuntu package "csh" NOT installed
VnmrJ required Ubuntu package "libmotif-dev" NOT installed
VnmrJ required Ubuntu package "expect" NOT installed
VnmrJ required Ubuntu package "tcl8.4-dev" NOT installed
VnmrJ required Ubuntu package "tk8.4-dev" NOT installed
VnmrJ required Ubuntu package "rarpd" NOT installed
VnmrJ required Ubuntu package "rsh-server" NOT installed
VnmrJ required Ubuntu package "tftpd" NOT installed
VnmrJ required Ubuntu package "mutt" NOT installed
VnmrJ required Ubuntu package "sharutils" NOT installed
VnmrJ required Ubuntu package "sendmail-cf" NOT installed
VnmrJ required Ubuntu package "gnome-power-manager" NOT installed
VnmrJ required Ubuntu package "k3b" NOT installed
VnmrJ required Ubuntu package "kdiff3" NOT installed
VnmrJ required Ubuntu package "ghostscript" NOT installed
VnmrJ required Ubuntu package "imagemagick" NOT installed
 Standby........
chmod: cannot access ‘/tmp/agilent_preinstall/*’: No such file or directory

==================================================================

There are RHEL packages required by OpenVnmrJ that are not installed

OpenVnmrJ Installation can not proceed until the required Linux packages are Installed.

Please following the instructions below to install the Linux packages:

Eject the OpenVnmrJ Installation CD and go to the directory /tmp/agilent_preinstall
Insert your systems RHEL 5 Installation DVD and run the installpkgs script
e.g.:

cd /tmp/agilent_preinstall
./installpkgs

After completion Eject RHEL DVD, insert the OpenVnmrJ CD and start the OpenVnmrJ install.

==================================================================
```

tesch1, 2016-12-09:
Ok, looks like we need to figure out a process for providing development branch install packages.

You're trying to install a version that doesn't have the "fixes" (I think this stuff is fixed, but only after the 1.1A release) onto a machine that is tested with the latest development branch with the fixes.

You can try building the development branch and then installing it, using the instructions in the README for the development branch... (it should only take an hour or so)... or wait until we have a way to provide intermediate packages... or work through the failures...

tesch1, 2016-12-19:
@osparking could you try the REL_TEST7_linux package at https://github.com/tesch1/OpenVnmrJ/releases ?
You'll probably still need to do `xhost +` before `./load.nmr` but I think the other issues should be resolved -- just tried it here successfully on the ubuntu14 Vagrant-defined VM.

osparking, 2016-12-19:
Thank you Mr. Michael. However, as the professor who asked me if I could
customize OpenVnmrJ for him found a different person than me. So, let me
refer him this issue. And I found Vagrant which I learnt for OpenVnmrJ is
fancy. -- osparking(=JB)

tesch1, 2016-12-19:
Sounds like this issue can be closed?

osparking, 2016-12-20:
Yes, you may. As I understand that issue could be reopen when you need.
Good luck.

closed 2020-06-05 by DanIverson