OpenVnmrJ/OpenVnmrJ #81: Newbie question regarding ./load.nmr failure (Ubuntu 14.04LTS in virtualbox)

goopeelchung, 2016-12-29:
Hi,
I am new to openvnmrj, and stuck at the following error (see below). Any pointers?
Thanks a lot.
Goopeel

.......
server starting
 
Creating /home/openvnmrj_1.1_A/templates/vnmrj/choicefiles/pis

Recreating 1 user(s)...
     Adding vnmr1
Loading data for all users (this may take a moment)...
.........................
.........................
.........................
..... 
Loading protocols...
,pg_ctl (PostgreSQL) 9.6.1
End of Database Setup.

Starting Post Installation Actions
Running VnmrJ Admin, Please Update User(s)
 
No protocol specified

tesch1, 2016-12-29:
Goopeel,
What version of OpenVnmrJ are you installing?  It looks like 1.1A, which has some known problems installing on Ubuntu.  Could you try the REL_TEST9 from https://github.com/tesch1/OpenVnmrJ/releases ?  If there are any errors, could you tar-up and post the files in /vnmr/adm/log/?  Thanks!

On Ubuntu, you may still need to run `xhost +` before running `./load.nmr`.

goopeelchung, 2016-12-29:
Yes, it's 1.1A. I will try that and let you know.
Thanks so much!

goopeelchung, 2016-12-29:
Hi, Michael,
I was able to get the UI running. Thanks a lot!
Goopeel

tesch1, 2016-12-29:
Great!  Could you close this issue?  I can't do that :-)

goopeelchung, 2016-12-29:
Sure. Another newbie issue. :)

closed 2016-12-29 by goopeelchung