OpenVnmrJ/OpenVnmrJ #92: Installation in MACOS error

jsaltod, 2017-01-21:
Hello,

I found an error while I was trying to install openvnmrj in mac OS

Here, there is the last lines of .log file:

gcc -o src/3D/lpqld.o -c -O4 -m32 -DHAVE_LARGEFILE_SUPPORT -D_FILE_OFFSET_BITS=64 -D_LARGEFILE_SOURCE -DLINUX -DFT3DIO -DNOASYNC -DMACOS -DFT3D -Isrc/3D src/3D/lpqld.c
src/3D/fmap.c:10:19: fatal error: stdio.h: No such file or directory
src/3D/errorlog.c:10:20: fatal error: unistd.h: No such file or directory
src/3D/fileinfo.c:10:19: fatal error: stdio.h: No such file or directory
src/3D/compressfid.c:10:20: fatal error: unistd.h: No such file or directory
compilation terminated.
compilation terminated.
compilation terminated.
src/3D/errLogLib.c:75:19: fatal error: stdio.h: No such file or directory
compilation terminated.
src/3D/fft3d.c:10:19: fatal error: stdio.h: No such file or directory
src/3D/lpqld.c:10:19: fatal error: stdio.h: No such file or directory
compilation terminated.
compilation terminated.
compilation terminated.
src/3D/comline.c:10:20: fatal error: unistd.h: No such file or directory
compilation terminated.
scons: *** [src/3D/fft3d.o] Error 1
scons: *** [src/3D/compressfid.o] Error 1
scons: *** [src/3D/lpqld.o] Error 1
scons: *** [src/3D/comline.o] Error 1
scons: *** [src/3D/fileinfo.o] Error 1
scons: *** [src/3D/errorlog.o] Error 1
scons: *** [src/3D/fmap.o] Error 1
scons: *** [src/3D/errLogLib.o] Error 1
scons: building terminated because of errors.
scons returned: 2
===>>> scons Build error  <<<===

I've also tried to install by ./load.nmr but in macos fails (I was able to install it in centos as well as ubuntu). Has someone any idea how to install openvnmrj in mac os? 

Thanks you for the inconvenience.

DanIverson, 2017-01-21:
I assume you have installed XCode on the Mac. You also need to install XCode's "command line tools". The procedure to install those tools depends on the version of XCode you are using. If you just want to install a pre-compiled version of OpenVnmrJ on a Mac, I have posted a beta version here ( https://github.com/DanIverson/OpenVnmrJ/releases ). Download the OpenVnmrJ11A.pkg

jsaltod, 2017-01-21:
At the moment It works fine!!! Very thanks Daniverson!!!!!

DanIverson, 2017-01-21:
Great. I am going to close this issue.

closed 2017-01-21 by DanIverson