OpenVnmrJ/OpenVnmrJ #106: Error printing NOESY

jsaltod, 2017-07-13:
Hello everyone!

I was trying to plot a noesy but I found this error and I was not able to plot it. 
[image]

Can someone help me?

Thanks!!

DanIverson, 2017-07-13:
The macro looks correct but the error messages mean something is wrong. From the OpenVJ command line, could you enter the commands 
  jFunc(55,'/vnmr/tmp/plbug')
  debug('c3')
and then do whatever you did to generate the error. After the error appears, enter
  debug('C')
and then attach the /vnmr/tmp/plbug file.

jsaltod, 2017-07-14:
Hello!! 

Thank you!

Here you have plbug file:

terminal /vnmr/tmp/plbug opened. 
macro xmaction('click','Sqn0004') started
 command exists('copyid','parameter','global') started
 command exists:"1" returned
 $activestudy=''
 $ret=1
 $dst='null'
 macro xmactionw('click','Sqn0004','null') started
  command exists('xmhaha','maclib') started
  command exists:"2" returned
  $activestudy=''
 macro xmactionw:1 returned
 copyid='Sqn0004','null'
macro xmaction returned
macro xmaction('doubleclick','Sqn0004') started
 command exists('copyid','parameter','global') started
 command exists:"1" returned
 $activestudy=''
 $ret=1
 $dst='null'
 macro xmactionw('doubleclick','Sqn0004','null') started
  command exists('xmhaha','maclib') started
  command exists:"2" returned
  $ret=0
  macro xmhaha('doubleclick','Sqn0004','null') started
   $tailX='tail'
   macro whichtail started
   macro whichtail:'tail' returned
   $sqattr='id','type','status','lock','title','exp','time','macro','data','CPnode','tooltext'
   $sqval='Sqn0001','Sqn0002','Sqn0003','Sqn0004','PROTON_01','PROTON','','return','/Users/joan/Google Drive/TESIS/NMR/217-f2-caract_20170713_01/PROTON_01','PROTON_001','PROTON'
   $ret=0
   macro xmhaha_select('Sqn0004') started
    $node='Sqn0004'
    $sqattr=''
    $sqval=''
    $prop='/Users/joan/vnmrsys/studies/exp35/tmpstudy/info/Sqn0004/prop'
    command exists('/Users/joan/vnmrsys/studies/exp35/tmpstudy/info/Sqn0004/prop','file') started
    command exists:"1" returned
    $sqattr=''
    $sqval=''
    command readfile('/Users/joan/vnmrsys/studies/exp35/tmpstudy/info/Sqn0004/prop','$sqattr','$sqval','','local') started
    command readfile returned
    $type=''
    $status=''
    $title=''
    $data=''
    $lock=''
    $macro=''
    command teststr('$sqattr','type','local') started
    command teststr:"1" returned
    $type='LIB'
    command teststr('$sqattr','status','local') started
    command teststr:"2" returned
    $status='Completed'
    command teststr('$sqattr','data','local') started
    command teststr:"8" returned
    $data='/Users/joan/Google Drive/TESIS/NMR/217-f2-caract_20170713_01/NOESY_01'
    command teststr('$sqattr','lock','local') started
    command teststr:"3" returned
    $lock='on'
    command teststr('$sqattr','title','local') started
    command teststr:"4" returned
    $title='NOESY_01'
    command teststr('$sqattr','macro','local') started
[... 3526 lines cut ...]
   mstring='dconi(`toggle`)','mfaction('mfzoom',0)','mfaction('fullzoom')','mfaction('mfzoom',1)','mfaction('mfzoom',-1)','setButtonMode(2) menu(`dconi`)','setButtonMode(3) menu(`dconi`)','init_dis(`trace`) dconi(`trace`)','if(mfShowAxis=1) then mfShowAxis=0 else mfShowAxis=1 endif repaint','if(aspMode<>0) then aspSetState(0) endif newmenu(`dconi_proj`) dconi(`restart`)','dconi(`again`)','if trace=`f2` then trace=`f1` else trace=`f2` endif dconi(`again`) if(overlayMode=3) then vpLayout(`overlayMode`,`align`) endif','vs2d=vs2d*1.2 dconi(`again`)','vs2d=vs2d/1.2 dconi(`again`)','newmenu(`dconi_phase`) dconi(`trace`)','newmenu(`ll2d`) dconi(`restart`)','menu(`display_2D`)','','aspAnno(`load`)','aspSetState(1)','aspSetState(18)','aspSetState(28)','aspSetState(20)','aspSetState(22)','aspSetState(5)','aspSetState(24)'
   micon='2D2cur.gif','2Dfull.gif','FullScale.gif','1Dexpand.gif','1Dzoomout.gif','ZoomMode.gif','1Dspwp.gif','2Dtrace.gif','1Dscale.gif:on','2Dvhproj.gif','recycle.gif','2Drotate.gif','2Dvs+20.gif','2Dvs-20.gif','1Dphase.gif','2Dpeakmainmenu.gif','return.gif','blackHorBumps.gif','Annotations_up.png','Select_up.png','Peaks_up.png','Text_up.png','Line_up.png','Arrow_up.png','Rectangle_up.png','Circle_up.png'
   $vjm=27
   mlabel='Box','Show Full Spectrum','Reset to Full Display','Zoom in','Zoom out','Zoom mode','Pan & Stretch Mode','Trace','Show/Hide Axis','Projections','Redraw','Rotate','Scale +20%','Scale -20%','Phase2D','Peak Picking','Return','','show/hide annotations','select','Peak Annotation','Text','Line','Arrow','Box','Oval','Polygon'
   mstring='dconi(`toggle`)','mfaction('mfzoom',0)','mfaction('fullzoom')','mfaction('mfzoom',1)','mfaction('mfzoom',-1)','setButtonMode(2) menu(`dconi`)','setButtonMode(3) menu(`dconi`)','init_dis(`trace`) dconi(`trace`)','if(mfShowAxis=1) then mfShowAxis=0 else mfShowAxis=1 endif repaint','if(aspMode<>0) then aspSetState(0) endif newmenu(`dconi_proj`) dconi(`restart`)','dconi(`again`)','if trace=`f2` then trace=`f1` else trace=`f2` endif dconi(`again`) if(overlayMode=3) then vpLayout(`overlayMode`,`align`) endif','vs2d=vs2d*1.2 dconi(`again`)','vs2d=vs2d/1.2 dconi(`again`)','newmenu(`dconi_phase`) dconi(`trace`)','newmenu(`ll2d`) dconi(`restart`)','menu(`display_2D`)','','aspAnno(`load`)','aspSetState(1)','aspSetState(18)','aspSetState(28)','aspSetState(20)','aspSetState(22)','aspSetState(5)','aspSetState(24)','aspSetState(26)'
   micon='2D2cur.gif','2Dfull.gif','FullScale.gif','1Dexpand.gif','1Dzoomout.gif','ZoomMode.gif','1Dspwp.gif','2Dtrace.gif','1Dscale.gif:on','2Dvhproj.gif','recycle.gif','2Drotate.gif','2Dvs+20.gif','2Dvs-20.gif','1Dphase.gif','2Dpeakmainmenu.gif','return.gif','blackHorBumps.gif','Annotations_up.png','Select_up.png','Peaks_up.png','Text_up.png','Line_up.png','Arrow_up.png','Rectangle_up.png','Circle_up.png','Polygon_up.png'
   $vjm=28
   mlabel='Box','Show Full Spectrum','Reset to Full Display','Zoom in','Zoom out','Zoom mode','Pan & Stretch Mode','Trace','Show/Hide Axis','Projections','Redraw','Rotate','Scale +20%','Scale -20%','Phase2D','Peak [... cut ...]
   micon='2D2cur.gif','2Dfull.gif','FullScale.gif','1Dexpand.gif','1Dzoomout.gif','ZoomMode.gif','1Dspwp.gif','2Dtrace.gif','1Dscale.gif:on','2Dvhproj.gif','recycle.gif','2Drotate.gif','2Dvs+20.gif','2Dvs-20.gif','1Dphase.gif','2Dpeakmainmenu.gif','return.gif','blackHorBumps.gif','Annotations_up.png','Select_up.png','Peaks_up.png','Text_up.png','Line_up.png','Arrow_up.png','Rectangle_up.png','Circle_up.png','Polygon_up.png','Open_Poly_up.png','X_bar_up.png','Y_bar_up.png'
  macro dconi_menu returned
 command menu returned
command dconi returned
$S=1
macro PersonaManager('showInPanel','process','basic') started
 command bgmode_is started
 command

tesch1, 2017-07-14:
Looks like it fails on directories with spaces in their name, can you
rename 'Google Drive' to GoogleDrive ?

On Fri, Jul 14, 2017 at 9:55 AM, jst <notifications@github.com> wrote:

> Hello!!
>
> Thank you!
>
> Here you have plbug file:
>
> terminal /vnmr/tmp/plbug opened.
> macro xmaction('click','Sqn0004') started
> command exists('copyid','parameter','global') started
> command exists:"1" returned
> $activestudy=''
> $ret=1
> $dst='null'
> macro xmactionw('click','Sqn0004','null') started
> command exists('xmhaha','maclib') started
> command exists:"2" returned
> $activestudy=''
> macro xmactionw:1 returned
> copyid='Sqn0004','null'
> macro xmaction returned
> macro xmaction('doubleclick','Sqn0004') started
> command exists('copyid','parameter','global') started
> command exists:"1" returned
> $activestudy=''
> $ret=1
> $dst='null'
> macro xmactionw('doubleclick','Sqn0004','null') started
> command exists('xmhaha','maclib') started
> command exists:"2" returned
> $ret=0
> macro xmhaha('doubleclick','Sqn0004','null') started
> $tailX='tail'
> macro whichtail started
> macro whichtail:'tail' returned
> $sqattr='id','type','status','lock','title','exp','time','
> macro','data','CPnode','tooltext'
> $sqval='Sqn0001','Sqn0002','Sqn0003','Sqn0004','PROTON_01'
> ,'PROTON','','return','/Users/joan/Google Drive/TESIS/NMR/217-f2-caract_
> 20170713_01/PROTON_01','PROTON_001','PROTON'
> $ret=0
> macro xmhaha_select('Sqn0004') started
> $node='Sqn0004'
> $sqattr=''
> $sqval=''
> $prop='/Users/joan/vnmrsys/studies/exp35/tmpstudy/info/Sqn0004/prop'
> command exists('/Users/joan/vnmrsys/studies/exp35/tmpstudy/info/Sqn0004/prop','file')
> started
> command exists:"1" returned
> $sqattr=''
> $sqval=''
> command readfile('/Users/joan/vnmrsys/studies/exp35/tmpstudy/info/
> Sqn0004/prop','$sqattr','$sqval','','local') started
> command readfile returned
> $type=''
> $status=''
> $title=''
> $data=''
> $lock=''
> $macro=''
> command teststr('$sqattr','type','local') started
> command teststr:"1" returned
> $type='LIB'
> command teststr('$sqattr','status','local') started
> command teststr:"2" returned
> $status='Completed'
[... 2044 lines cut ...]
> command exists('ni2','parameter','current') started
> command exists:"0" returned
> command exists('ni3','parameter','current') started
> command exists:"0" returned
> macro getdim:2 returned
> $e=-1
> command exists('/Users/joan/vnmrsys/exp35/datadir3d/extr','file') started
> command exists:"0" returned
> $ei2=-1
> command exists('index2','parameter') started
> command exists:"0" returned
> mlabel='Box'
> micon='2D2cur.gif'
> mstring='dconi(toggle)'
> $vjm=2
> mlabel='Box','Show Full Spectrum'
> micon='2D2cur.gif','2Dfull.gif'
> mstring='dconi(toggle)','mfaction('mfzoom',0)'
> $vjm=3
> mlabel='Box','Show Full Spectrum','Reset to Full Display'
> micon='2D2cur.gif','2Dfull.gif','FullScale.gif'
> mstring='dconi(toggle)','mfaction('mfzoom',0)','mfaction('fullzoom')'
> $vjm=4
> mlabel='Box','Show Full Spectrum','Reset to Full Display','Zoom in'
> micon='2D2cur.gif','2Dfull.gif','FullScale.gif','1D

jsaltod, 2017-07-14:
Completly solved! 

Thanks you!!!

DanIverson, 2017-07-28:
I fixed the specific instance of this problem for plotting spectra. However, the larger problem of spaces in path names still exists. I am closing this issue but opened a new issue #109 describing the larger issue.

closed 2017-07-28 by DanIverson