OpenVnmrJ/OpenVnmrJ #129: website claims OS X binaries exist but there doesn't seem to be any present

jwhowarth, 2017-09-09:
The main web page at https://github.com/OpenVnmrJ/OpenVnmrJ claims under the Downloads section that "OS X: Built on 10.11 (El Capitan) and tested on 10.11 and 10.10 (Yosemite)" exist for use on OS X. However the only files available are dvdImage_OpenVnmrJ_1.1A_DD2.tbz and dvdImage_OpenVnmrJ_1.1A_MI.tbz both of which, when uncompressed and load.nmr is executed, fail on macOS with the error...

grep: /etc/selinux/config: No such file or directory

Disabling SELinux, System Reboot Required.
You must reboot the system prior to continuing the install.

cat: /etc/selinux/config: No such file or directory
/Users/howarth/Downloads/dvdimageOVJ/load.nmr: line 250: /etc/selinux/config.mod: No such file or directory
mv: rename /etc/selinux/config.mod to /etc/selinux/config: No such file or directory
chmod: /etc/selinux/config: No such file or directory

Where are the actual binaries for OS X located?

DanIverson, 2017-09-09:
I used to have a version on my personal github account. I must have accidentally removed it. I just uploaded it to the OpenVnmrJ releases page at https://github.com/OpenVnmrJ/OpenVnmrJ/releases

jwhowarth, 2017-09-10:
Dan,
     Thanks. The OpenVnmrJ11A.pkg installed and worked fine under High
Sierra Developer Beta 9 with the exception of one glitch. The application
produced an error alert and quit unless I change the line in /etc/hosts
from...

127.0.0.1       localhost

to

127.0.0.1       localhost Mac-Pro.local

to match the output of 'hostname'. I would be nice if OpenVnmr could be
made to accept a simple 'localhost' as a fallback hostname so users didn't
have to edit the network files manually.
          Jack
ps I also noticed that the
binary /Applications/OpenVnmrJ_1.1.app/Contents/MacOS/VJ is reported by
file to be a fat binary...

$ file VJ
VJ: Mach-O universal binary with 2 architectures: [i386:Mach-O executable
i386] [x86_64:Mach-O 64-bit executable x86_64]
VJ (for architecture i386): Mach-O executable i386
VJ (for architecture x86_64): Mach-O 64-bit executable x86_64

Is the entirety of OpenVnmrJ compiled as x86_64 on macOS? I ask because the
next release of macOS past High Sierra might eliminate the i386 code
support in the OS so it would be a very good idea to convert the  OpenVnmrJ
build to be 64-bit throughout.

jwhowarth, 2017-09-12:
Dan,
     What is the status of being able to execute 'seqgen' to compile
sequences on non-console workstations? It is useful to be able to display
the pulse sequence used for an experiment. Has anyone managed to edit the
existing /vnmr/psg/seqgenmake to allow it to work on under x86_64 Linux
with the i386 legacy support packages?
              Jack

DanIverson, 2017-09-12:
I will look into using localhost if the hostname does not exist in /etc/hosts. Compiling for 64-bit is a bigger job, especially if data acquisition is involved (i.e., for Linux systems). For MacOS, it may be more manageable and it may be a good place to start.
The seqgen command should work on all systems. On MacOS systems from a terminal, you may need to do "source /vnmr/user_templates/.vnmrenvsh" first, in order to set the PATH variable and some other parameters.

tesch1, 2017-09-12:
I worked through converting psglib to 64 bit at one point (see #20), it's
not that much work, mostly just getting all the structure alignments
correct, iirc.  I was linking the pulse sequence against some 64bit
libraries, so all the seqgenmake output (up to and including the pulse
sequence binary) was 64 bit.

Dan, has there been any progress on clarifying the contributions issue?

closed 2020-06-05 by DanIverson