OpenVnmrJ/OpenVnmrJ #131: Clarify path to accepting contributions to OpenVnmrJ

tesch1, 2017-09-12:
Just opening an Issue to have a place to discuss and keep track of clarifying what the path is for the community to contribute to OpenVnmrJ.

tesch1, 2017-10-04:
Last discussion here: https://github.com/OpenVnmrJ/ovjTools/pull/21

DanIverson, 2017-10-14:
I re-read the links about contributing to OpenVnmrJ. I am pretty sure the "Contributor License Agreement" (CLA) was modeled on another open-source project, although I do not remember which one. From my point of view, my interest is to allow people to make changes and add code with a minimum amount of trouble. To that end, the original concept was that the copyright for all contributions would be assigned to the Univ. of Oregon (UO).  That is certainly what I am doing with my contributions. Attribution to the developer as comments within the code is certainly acceptable.

If there were multiple copyrights, it becomes fuzzy in my mind what is permissible. Do I have to contact the copyright owner in order to make a change. If I were to make a small change to fix a bug, what would happen to the copyright?  We had hoped to avoid these questions by simply stating that the copyright of everything contributed by any individual would be assigned to UO.

The current CLA states:

Copyright License. You hereby grant, and agree to grant, to UO a non-exclusive, perpetual, irrevocable, worldwide, fully-paid, royalty-free, transferable copyright license to reproduce, prepare derivative works of, publicly display, publicly perform, and distribute your Contributions and such derivative works, with the right to sublicense the foregoing rights through multiple tiers of sublicensees.

As I interpret this, any code submitted should have either no copyright statements, as in the case of macros, or comments such as included in many of the scripts and source code files. An example is

\# Copyright (C) 2017  University of Oregon
\#
\# You may distribute under the terms of either the GNU General Public
\# License or the Apache License, as specified in the LICENSE file.
\#
\# For more information, see the LICENSE file.
\#

For licensing, the VOSSG decided on the Apache license for all code except code that which requires a GPL license, due to its use of GPL code or libraries.

In terms of contributing code, there are at least three categories.
1. A simple bug fix. This should be non-controversial and inclusion in the master branch should be fairly straight-forward.
2. Brand new functionality. As long as it is completely new and does not break existing functionality, inclusion in the master branch should be straight-forward.
3. Modification of existing functionality. This is where things can get a little tricky. Some might consider certain functionality a "bug" while others consider it the correct behavior. Preferences over UI features are seldom universal. Arbitration would ideally involve a team of people to decide. At this point in time, that team is Tim Burrow and me. In practice, it is just me. However, I generally reach out to others for their opinion when it comes to these questions.

I agree that the github site could be updated so that these things are clearer. If you are happy with these ideas, I could try to formalize them a bit for review and then put them on github.

tesch1, 2017-10-14:
I'm completely happy with the ideas - I think that everyone basically has the same interests, but the legalities aren't quite clear, of course it's better to clarify ahead of time, I agree.  I'll leave the questions about deciding project direction aside, since that's separate from the legalities of how to contribute, and is already up to you guys.

Have there been any lawyers involved in drafting the strategy?  From my understanding of copyright, the excerpt of the license:

> Copyright License. You hereby grant, and agree to grant, to UO a non-exclusive, perpetual, irrevocable, worldwide, fully-paid, royalty-free, transferable copyright license to reproduce, prepare [...]

^ is a broad license from the holder of the copyright to UO, which means that the author still owns copyright on their contribution, but UO can do whatever UO wants with it... which in effect means that UO can re-license the entire work under whatever they want in the future (for whatever reason.. can't imagine a scenario that this would help UO.. but whatever..).  On the other hand, if a contributor were to license their code directly under the Apache2/GPL license to the world, to be combined with the rest of the OU+other-contributor Apache2/GPL-licensed body of work, anybody can still edit/use/distribute the work under Apache2/GPL, but UO wouldn't have the rights to change the license away from the Apache2/GPL license, as a whole, licensed.  I suppose it doens't make too much of a difference to me, since it's always possible to fork into a new project off the UO Apache2 license, (and if someone wants to try to make a commercial offshoot- more power to them).

It's creating this, with the associated bottleneck of any contributor needing to sign up with UO in order to contribute:

[image]

I'm a little confused about having this extra license to UO beyond the Apache2/GPL, why not just have all contributors agree to contribute under Apache2/GPL and leave it at that?  Seems like the Apache2/GPL is much simpler to wrap-up than a very broad full-rights assignment (maybe have contributors sign a form confirming that any and all contributions are under Apache2/GPL?)  Like:

[image]

It wouldn't prevent anyone from fixing bugs in or modifying the combined work, since that's clearly allowed in Apache2/GPL.

And I'm not sure about this:

> As I interpret this, any code submitted should have either no copyright statements, as in the case of macros, or comments such as included in many of the scripts and source code files. An example is

The license snippet above isn't an assignment of the author's copyright, it just creates an indefinite essentially unlimited license to the work.  So the work's copyright is still owned by the author, with or without a copyright notice (lack of copyright notice doesn't change whether the work's copyright is owned by the author - in current US law).  I always put a copyright notice on things I write out of habit, unless I'm sure I've assigned the copyright, since being clear upfront is better than creating a mess to pay lawyers to figure out.

Am I understanding something incorrectly?  I'm really not trying to be pedantic, just have been around enough IP issues to want everything clear and correct, and have seen lawyers do goofy stuff around open source trying to extend their employer's ownership as much as possible, pointlessly and often in the most complicated paperwork-inducing way possible.

tesch1, 2017-10-14:
Did a little more googlin, here's why FSF wants copyright assignments: https://www.gnu.org/licenses/why-assign.en.html but when would UO ever want to go after anyone for using Apache2-licensed code?  It already gives you rights to do pretty much whatever you want with it.

tesch1, 2017-10-16:
Ah, and another problem with UO asking for a copyright assignment is that for such an assignment there usually has to be some sort of exchange to complete the transaction, otherwise the door remains open for contributors or their employers to later on come back and say the assignment isn't based on a valid contract.  Having everyone contributing on equal grounds (under Apache2/GPL) significantly diminishes any such sorts of arguments, because all the licensing is symmetric between parties.

DanIverson, 2017-10-27:
I think we are agreeing, but let me get your take on the following.
1. Person A contributes some code to OpenVnmrJ (OVJ) with their copyright. As a developer for OVJ, I can make changes to this code as I see fit. I do not need to contact the copyright holder in order to do this. Is this correct?
2. Person A contributes some code to OVJ with their copyright. Over a period of time, multiple OVJ contributors make changes to the code so that it bears little resemblance to the original. What does the copyright mean in this case?

tesch1, 2017-10-28:
1. Absolutely correct.  Any Contribution (as defined by section 1 of Apache 2) is subject to the grants under section 2 and 3.  By section 5, simply submitting a pull-request is enough to create a license ("perpetual, worldwide, non-exclusive, no-charge, royalty-free, irrevocable copyright license..." etc...) to the Contribution.  Anyone who has contributed to OpenVnmrJ has already granted you this license.

2. The Copyright (different from "copyright notice") is associated with having "created" a work, a Contribution (under Apache2) is a grant for others to do anything they want with that work, including creating derivative works (continuing the quote above: "...to reproduce, prepare Derivative Works of, publicly display, publicly perform, sublicense, and distribute the Work and such Derivative Works in Source or Object form").

The license is designed to effectively group all the derived works under a single license that's so permissive that it's not really important who exactly owns the copyright on any particular snippet, since the rights a Licensee (using the definition from Apache2) receives are so broad that you have very few obligations to the Licensee.

If I create a file and type some stuff into it, I've created a work, which I own the Copyright to (whether or not I put a copyright notice on it).  If I simply add that file to a pull-request to OpenVnmrJ, I'm offering a license to that work under Apache2 (sections 2 and 3 describe the scope of the rights I'm granting), to OpenVnmrJ.  For clarity, before accepting new files, OpenVnmrJ should require the apache2/GPL license notice to be added to Contributions.

If someone modifies a file (or files) they received from OpenVnmrJ sufficiently to create a new "work" and then submit that work back to OpenVnmrJ, ie via a pull-request, even without adding their own copyright notice they still own the copyright, AND they have created a license to that work permitting all the things permitted by Apache2.  In summary, it's not necessary to put each and every copyright notice on each file that any particular author thinks they've created some copyrighted "work" in, since that information is tracked by git, should it ever be needed down the road.  It is good to have all the files contain the Apache2/GPL notice, for clarity about the licensing.

Here's a really good article on the details, it also discusses the issues around managing copyright notices on individual files in large projects (such as OpenVnmrJ) : http://www.softwarefreedom.org/resources/2012/ManagingCopyrightInformation.html

A copyright "assignment" is a different beast, and is far more onerous on both parties.  It also means that anything anyone assigns to OU they no longer hold the original rights to... which is a bit odd for open source software, especially under Apache2.

tesch1, 2018-08-09:
resolved

closed 2018-08-09 by tesch1