OpenVnmrJ/OpenVnmrJ #135: OpenVnmrJ11A on macOS hangs when new Java 9 is installed

jwhowarth, 2017-09-30:
While the existing OpenVnmrJ11A.pkg installer completes without errors when run on macOS 10.13 with the new Java 9 GM release is installed, the resulting OpenVnmrJ_1.1.app hangs at the OpenVNMRJ splash screen when launched.

DanIverson, 2017-09-30:
Are there any messages in ~/vnmrsys/VnmrjMsgLog?

jwhowarth, 2017-10-03:
Nothing particularly informative. The diff between vnmrsys/VnmrjMsgLog under jdk9 (which hangs at the OpenVNMJ splash screen) and jdk8 is below...

`--- /Users/howarth/vnmrsys/VnmrjMsgLog.jdk9	2017-10-02 20:42:32.000000000 -0400
+++ /Users/howarth/vnmrsys/VnmrjMsgLog	2017-10-02 20:52:27.000000000 -0400
@@ -1,17 +1,17 @@
-## INFO  Oct 2, 2017 20:42:30 **********************************************
-## INFO  Oct 2, 2017 20:42:30 ********** Starting OpenVnmrJ as howarth **********
-## INFO  Oct 2, 2017 20:42:30 ** OpenVnmrJ VERSION 1.1 REVISION A **
-## INFO  Oct 2, 2017 20:42:30 ** January 20, 2017 **
-## INFO  Oct 2, 2017 20:42:30 ** vnmrs **
-## INFO  Oct 2, 2017 20:42:30 **********************************************
-ERROR Oct 2, 2017 20:42:32 makeDBConnection failed on Mac-Pro.local port 5432,
+## INFO  Oct 2, 2017 20:51:40 **********************************************
+## INFO  Oct 2, 2017 20:51:40 ********** Starting OpenVnmrJ as howarth **********
+## INFO  Oct 2, 2017 20:51:40 ** OpenVnmrJ VERSION 1.1 REVISION A **
+## INFO  Oct 2, 2017 20:51:40 ** January 20, 2017 **
+## INFO  Oct 2, 2017 20:51:40 ** vnmrs **
+## INFO  Oct 2, 2017 20:51:40 **********************************************
+ERROR Oct 2, 2017 20:51:42 makeDBConnection failed on Mac-Pro.local port 5432,
    be sure database exists and that howarth has access to it.
    Try having the system administrator execute the command 'createuser howarth'.
    Also check that if a special port number  is used for postgres,
    that this user has the environment variable 'PGPORT' set  correctly.
    Is it possible that a specific error can be found in the log file:
       /vnmr/pgsql/pgsql.log
-## ERROR Oct 2, 2017 20:42:32 makeDBConnection failed on Mac-Pro.local port 5432,
+## ERROR Oct 2, 2017 20:51:42 makeDBConnection failed on Mac-Pro.local port 5432,
    be sure database exists and that howarth has access to it.
    Try having the system administrator execute the command 'createuser howarth'.
    Also check that if a special port number  is used for postgres,
@@ -21,8 +21,8 @@
 Connection refused. Check that the hostname and port is correct, and that the postmaster is running with the -i flag, which enables TCP/IP networking.
 	at postgresql.Connection.openConnection(Connection.java:119)
 	at postgresql.Driver.connect(Driver.java:103)
-	at java.sql/java.sql.DriverManager.getConnection(DriverManager.java:678)
-	at java.sql/java.sql.DriverManager.getConnection(DriverManager.java:229)
+	at java.sql.DriverManager.getConnection(DriverManager.java:664)
+	at java.sql.DriverManager.getConnection(DriverManager.java:247)
 	at vnmr.ui.shuf.FillDBManager.makeDBConnection(FillDBManager.java:4326)
 	at vnmr.ui.shuf.FillDBManager.checkDBConnection(FillDBManager.java:4265)
 	at vnmr.ui.shuf.FillDBManager.executeQuery(FillDBManager.java:7140)
@@ -42,17 +42,32 @@
 	at vnmr.ui.VNMRFrame.buildWindows(VNMRFrame.java:251)
 	at vnmr.ui.VNMRFrame.access$000(VNMRFrame.java:31)
 	at vnmr.ui.VNMRFrame$1.run(VNMRFrame.java:89)
-	at java.desktop/java.awt.event.InvocationEvent.dispatch(InvocationEvent.java:313)
-	at java.desktop/java.awt.EventQueue.dispatchEventImpl(EventQueue.java:764)
-	at java.desktop/java.awt.EventQueue.access$500(EventQueue.java:97)
-	at java.desktop/java.awt.EventQueue$3.run(EventQueue.java:717)
-	at java.desktop/java.awt.EventQueue$3.run(EventQueue.java:711)
-	at java.base/java.security.AccessController.doPrivileged(Native Method)
-	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:89)
-	at java.desktop/java.awt.EventQueue.dispatchEvent(EventQueue.java:734)
-	at java.desktop/java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:199)
-	at java.desktop/java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:124)
-	at java.desktop/java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:113)
-	at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:109)
-	at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
-	at java.desktop/java.awt.EventDispatchThread.run(EventDispatchThread.java:90)
+	at java.awt.event.InvocationEvent.dispatch(InvocationEvent.java:311)
+	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:756)
+	at java.awt.EventQueue.access$500(EventQueue.java:97)
+	at java.awt.EventQueue$3.run(EventQueue.java:709)
+	at java.awt.EventQueue$3.run(EventQueue.java:703)
+	at java.security.AccessController.doPrivileged(Native Method)
+	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:80)
+	at java.awt.EventQueue.dispatchEvent(EventQueue.java:726)
+	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:201)
+	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
+	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
+	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
+	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
+	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)
+ERROR Oct 2, 2017 20:52:02 makeDBConnection failed on Mac-Pro.local port 5432,
+   be sure database exists and that howarth has access to it.
+   Try having the system administrator execute the command 'createuser howarth'.
+   Also check that if a special port number  is used for postgres,
[... cut ...]
+      /vnmr/pgsql/pgsql.log
+## DEBUG Oct 2, 2017 20:52:08 Using default profile (AllLiquids.xml) for howarth.  To assign
+    a different profile, use the Configure->Operators->Edit operators... panel
+    in the vnmrj adm tool.
+## DEBUG Oct 2, 2017 20:52:25 ### appDir 0 /Users/howarth/vnmrsys
+## DEBUG Oct 2, 2017 20:52:25 ### appDir 1 /vnmr
+## DEBUG Oct 2, 2017 20:52:25 ### appDir 2 /vnmr/openvnmrj-utils
+## DEBUG Oct 2, 2017 20:52:25 ### rebuildUI
+## DEBUG Oct 2, 2017 20:52:27 start UI`

DanIverson, 2017-10-03:
The VnmrjMsgLog errors are about the Locator. They don't indicate why OVJ hangs at the splash screen. I don't have a Mac to use to investigate this, so I did the following. I installed CentOS 7 in a VirtualBox VM. I then installed OpenVJ and Java 9 (from the rpm). I removed the /vnmr/jre directory. I changed line 176 in the /vnmr/bin/vnmrj startup script for OVJ from
   javabin="$vnmrsystem/jre/bin/java"
to
   javabin="java"
When I ran vnmrj, I got a blank splash screen. I got different messages in the VnmrjMsgLog file.  I edited the /vnmr/bin/vnmrj script again, this time changing line 213 from
    -classpath $vjclasspath $dargs   \
to 
   -classpath $vjclasspath $dargs  --add-modules ALL-SYSTEM \

When I ran vnmrj again, the windows came up and I could process /vnmr/fidlib/fid1d. I have not tried other things.

DanIverson, 2017-10-20:
Did adding the option  --add-modules ALL-SYSTEM as described in the previous post help?

jwhowarth, 2017-10-21:
Yes, that change seems to work under JRE/JDK 9.0.1 on High Sierra with OpenVnmrJ11A.

JCFNMR, 2017-11-24:
Hello, I fixed the issue following this thread, but now when the OpenVnmrJ opens, the following error appears and the application closes.
[image]
The VnnmrjMsgLog says:
Connection refused. Check that the hostname and port is correct, and that the postmaster is running with the -i flag, which enables TCP/IP networking.
	at postgresql.Connection.openConnection(Connection.java:119)
	at postgresql.Driver.connect(Driver.java:103)
	at java.sql/java.sql.DriverManager.getConnection(DriverManager.java:678)
	at java.sql/java.sql.DriverManager.getConnection(DriverManager.java:229)
	at vnmr.ui.shuf.FillDBManager.makeDBConnection(FillDBManager.java:4326)
	at vnmr.ui.shuf.FillDBManager.checkDBConnection(FillDBManager.java:4265)
	at vnmr.ui.shuf.FillDBManager.executeQuery(FillDBManager.java:7140)
	at vnmr.ui.shuf.SymLinkMap.getSymLinksFromDB(SymLinkMap.java:465)
	at vnmr.ui.shuf.SymLinkMap.<init>(SymLinkMap.java:54)
	at vnmr.ui.shuf.FillDBManager.<init>(FillDBManager.java:232)
	at vnmr.ui.shuf.ShufDBManager.<init>(ShufDBManager.java:55)
	at vnmr.ui.shuf.ShufDBManager.<clinit>(ShufDBManager.java:29)
	at vnmr.ui.shuf.LocatorHistory.<init>(LocatorHistory.java:62)
	at vnmr.ui.shuf.LocatorHistoryList.<init>(LocatorHistoryList.java:126)
	at vnmr.ui.SessionShare.FillLocatorHistoryList(SessionShare.java:98)
	at vnmr.ui.SessionShare.<init>(SessionShare.java:84)
	at vnmr.ui.VnmrjUI.<init>(VnmrjUI.java:185)
	at vnmr.ui.VnmrjUI.<init>(VnmrjUI.java:328)
	at vnmr.ui.VNMRFrame.installApp(VNMRFrame.java:480)
	at vnmr.ui.VNMRFrame.<init>(VNMRFrame.java:411)
	at vnmr.ui.VNMRFrame.buildWindows(VNMRFrame.java:251)
	at vnmr.ui.VNMRFrame.access$000(VNMRFrame.java:31)
	at vnmr.ui.VNMRFrame$1.run(VNMRFrame.java:89)
	at java.desktop/java.awt.event.InvocationEvent.dispatch(InvocationEvent.java:313)
	at java.desktop/java.awt.EventQueue.dispatchEventImpl(EventQueue.java:764)
	at java.desktop/java.awt.EventQueue.access$500(EventQueue.java:97)
	at java.desktop/java.awt.EventQueue$3.run(EventQueue.java:717)
	at java.desktop/java.awt.EventQueue$3.run(EventQueue.java:711)
	at java.base/java.security.AccessController.doPrivileged(Native Method)
	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:89)
	at java.desktop/java.awt.EventQueue.dispatchEvent(EventQueue.java:734)
	at java.desktop/java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:199)
	at java.desktop/java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:124)
	at java.desktop/java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:113)
	at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:109)
	at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.desktop/java.awt.EventDispatchThread.run(EventDispatchThread.java:90)
## DEBUG lis 24, 2017 03:00:54 WARNING: label property not found: May Edit Password
## DEBUG lis 24, 2017 03:00:54 WARNING: label property not found: rights
## DEBUG lis 24, 2017 03:00:54 WARNING: label property not found: Enable Locator Update
## DEBUG lis 24, 2017 03:00:54 WARNING: label property not found: rights
and this goes on and on and on...
Any suggestions on how to fix this? A step-by-step guide would be very appreciated. Working on High Sierra with JDK 9.0.1.

DanIverson, 2017-11-27:
The hostname needs to be added to the /etc/hosts file. In the terminal, enter
  hostname
to display the hostname. Then, enter
  sudo  vi  /etc/hosts
Depending on your system security, you may need to provide a password. If your system uses DHCP or is not networked, append the hostname to the
  127.0.0.1   localhost
line. If your system has a fixed IP address, add the IP address and hostname to a new line.

jwhowarth, 2017-11-27:
Why not fix OpenVNMRJ so that 'localhost' is honored without resorting to appending the actual hostname of the machine?

closed 2017-11-27 by jwhowarth