OpenVnmrJ/OpenVnmrJ #140: How create new command for VnmrJ command line?

trigchen, 2017-11-03:
I want to do some specific fid and spectra manipulation in current experiment. 
How can I write a new command which can be run in VnmrJ command line?
I found there were a user guide for VNMR 6.1C source code (seen attachment).  Is there anyone can share this document?

tesch1, 2017-11-03:
All the manuals are available at SpinSights (free login required) here: https://spinsights.chem.agilent.com/community/resources/manuals/content

You'll probably want the "VnmrJ 4 User Programming Reference Guide" : https://spinsights.chem.agilent.com/docs/DOC-2127

NeutralKaon, 2017-11-03:
To write a macro, just create a file with the name you desire in ~/vnmrsys/maclib and write in VnmrJ's simple scripting language, MAGICAL ("MAGnetics Instrumentation Control and Analysis Language", I think). You can also turn on macro debugging with 'debug('c3')' in the command line, and VnmrJ debugging with 'JFunc(55,'/a/path/to/a/TTY/or/file')'. These are things I wish people had told me when I started. 

All the best, 

-- Jack

trigchen, 2017-11-05:
Thanks for the replies. 
What I want to create is a built-in VNMR command which is the same as ft, aph, but not a MAGICAL macro.

tesch1, 2017-11-05:
Probably the easiest way to do this is to write an external program that does your transformation, and write a macro that saves the current data, runs your program on the saved data, and then loads the result.

trigchen, 2017-11-05:
external Unix program can't access the spectrum buffer in experiments.

tesch1, 2017-11-05:
the 'flush' command can be used to write it to disk

trigchen, 2017-11-06:
How about the processed FID before FFT to spectrum?

DanIverson, 2017-11-06:
Check the fidproc command. /Vnmr/manual/fidproc. It has many FID processing tools. One can also add built-in commands. See the file table.c in src/vnmrbg I can give more details when I return from my vacation on Nov. 17

trigchen, 2017-11-21:
Can anyone share the document "VNMR Source Code User Guide" (pdf format file) in the CD("VNMR 6.1 C Source Code") released with VNMR 6.1C ?