OpenVnmrJ/OpenVnmrJ #149: Vnmrbg could not be started on Ubuntu 16.04 on Oracle VirtualBox

Kas-scientist, 2018-03-29:
Hello,
first of all I want to thank you for all the work that all of you have done to make this running.

I had Open VnmrJ on a laptop half a year ago, working without problems in Ubuntu 16.04 (not in a Virtual machine) but now I have a new laptop and I don't want to install Ubuntu in a dedicated partition, I want to run it in a Virtual Machine.

I had instaled 2 times Ubuntu 16.04 and VnmrJ (dvdimageOVJMI_REL_TEST9) on the Oracle VirtualBox but I have this error when I start vnmrj.
[image]

My hostname is on etc/hosts file
[image]
[image]

So I don't know what might be happening and how can I solve this error.

Thanks.

DanIverson, 2018-03-29:
Could you edit the /etc/hosts file. Delete the line with
 127.0.1.1  kas
and add kas to the line with localhost so it reads
  127.0.0.1    localhost  kas

Kas-scientist, 2018-03-29:
Still the same error.

I have restarted the VirtualBox after changing etc/hosts because I want to be sure that Ubuntu reads the changes in the file.

[image]

DanIverson, 2018-03-29:
Does "ping kas" work?  Also, what does "ldd /vnmr/bin/Vnmrbg" display?

Kas-scientist, 2018-03-29:
"ping kas" work
[image]

and "ldd /vnmr/bin/Vnmrbg" says "it's not a dinamic executable" in Spanish (my system language)
[image]

Kas-scientist, 2018-03-29:
I'm searching even more using your tips as orientation.
I have instaled gcc-multilib with the usual "sudo apt-get install gcc-multilib" and now "ldd /vnmr/bin/Vnmrbg" says:
[image]

Kas-scientist, 2018-03-29:
[image]
And it seems that with that it works now, thanks.

The solution was installing "gcc-multilib", I find it strange because I think that I had instaled the 32 libraries before installing open VnmrJ, but anyway, that solves the problem.

Thanks.

Kas-scientist, 2018-03-29:
Hmm... it's still giving some errors and I can't see the experiment selector tree and fids can't load :(

[image]

DanIverson, 2018-03-29:
It looks like progress. The errors in your last post are because OVJ is in Exp: 0, as seen in the blue line just below the command line. Enter the following into the OVJ command line and things should look better.
    jexp1   rt('/vnmr/fidlib/fid1d') wft

Kas-scientist, 2018-03-29:
[image]
It works now and I can see experiments on it, thanks.

closed 2018-03-29 by DanIverson