OpenVnmrJ/OpenVnmrJ #196: /bin/sh -> /bin/dash on ubuntu

tesch1, 2018-08-07:
Another issue with Ubuntu - they use /bin/dash as the system shell, which breaks some things, in particular: non-posix redirect syntax: https://wiki.ubuntu.com/DashAsBinSh#A.26.3E

One obvious possibility would be to change all the calls to 'shell()' to use the correct posix redirect syntax, but that doesn't help users who have local scripts that depend on the "normal" syntax.

Another possibility would be to bake /bin/bash into Vnmr* (popen_call) instead of /bin/sh.  It's probably a safe assumption for all Linuxen.  That still wouldn't fix the places where system() is used with the assumption of bashy-like features, since system() has /bin/sh hardwired .. however, I'm not sure how big of an issue that is.

Or we could force/check that /bin/sh -> /bin/bash .. that's pretty reasonable too.

```
Vnmr -mback "shell('ls >& /dev/null')"
centos6 -> ok
ubuntu18 -> sh: 1: Syntax error: Bad fd number
```

on ubuntu18:

```
/bin/dash -c 'echo x >& /dev/null'
/bin/dash: 1: Syntax error: Bad fd number
/bin/bash -c 'echo x >& /dev/null'
ok
```

tesch1, 2018-09-22:
I think the right thing to do is check that /bin/sh IS /bin/bash-compatible during the install.  There are very many `system()` and `shell` calls that assume the bash redirect style, would be a lot of work to change and test them all.

DanIverson, 2019-10-25:
As part of the OVJ installation, we do

echo "dash dash/sh boolean false" | debconf-set-selections
dpkg-reconfigure -u dash

I think that should fix it.

tesch1, 2019-10-26:
Agreed.

closed 2019-10-26 by tesch1