OpenVnmrJ/OpenVnmrJ #208: Support or at least detect Java9/10 on macos, and/or fail deterministically

tesch1, 2018-08-30:
Running on Mac uses the system java - if the user has java 10 installed, startup hangs without any useful message to the user.

tesch1, 2018-08-30:
From this thread on spinsights: https://spinsights.chem.agilent.com/message/20671

DanIverson, 2019-02-20:
I have been looking for a robust way to get the java version in a bash script. This seems to work:
vers=$(java -fullversion 2>&1 | awk 'BEGIN {FS="\""} {print $2}' | awk 'BEGIN {FS="."} {print $1}')

DanIverson, 2019-02-20:
Should be fixed with PR #241

tesch1, 2019-02-20:
should the second one be `print $2`?

DanIverson, 2019-02-20:
No, it is the first number. They changed the java version scheme. Up until java 8, it was 1.6.x, 1.7.x, and 1.8.x. With java 9 and 10, it is now 9.x.y and 10.x.y. I just get the first number and test whether it is greater than 8.

tesch1, 2019-02-20:
Ah yes, what good is a scheme if you dont change it once in a while :)  I suppose I'll close this too.

closed 2019-02-20 by tesch1