OpenVnmrJ/OpenVnmrJ #209: how to install teh soft?

uebmaster, 2018-09-06:
i have doubts about the installation process.

Actually the computer where i have the varian soft has failed, and i must to use a different computer and i am having troubles to make a new install.

Searching i found this soft that i understand could remplace the original soft, but i read this: "OpenVnmrJ does not contain console software"

What means?

I am newbie in the topic but we need that the computer works, someone could help me?

Thanks in advance.

DanIverson, 2018-09-06:
The comment that "OpenVnmrJ does not contain console software" means that if you want to connect your new computer to a Varian/Agilent spectrometer, you first need to install VnmrJ 4.2 and the "110" patch. After that, you can install OpenVnmrJ. If your new computer is only going to be used for data processing, then you do not need to install VnmrJ 4.2 first.

uebmaster, 2018-09-06:
where can i download VnmrJ 4.2? we just have Vnmrj 3.2 Linux Version.

DanIverson, 2018-09-06:
If you are on SpinSights, all of the information is here ( https://spinsights.chem.agilent.com/docs/DOC-12723 ). Basically, you need to order it from your Agilent service person.

uebmaster, 2018-09-06:
ok, thanks.

DanIverson, 2020-06-05:
With the release of OpenVnmrJ Version 2.1, one no longer needs to first install VnmrJ 4.2 and the 110 patch. OpenVnmrJ Version 2.1 includes the console software. Also, there are installation instructions at https://github.com/OpenVnmrJ/OpenVnmrJ/blob/master/Install.md

closed 2020-06-05 by DanIverson