OpenVnmrJ/OpenVnmrJ #235: Is CRAFT functional?

kolmank, 2019-02-12:
Hello,

First I would like to thank for your amazing work regarding OpenVnmrJ. I have installed the software on my Linux Mint 19 and it has been working very nicely so far (only for data analysis). However, yesterday I wanted to test CRAFT to process my FID data and every time I tried I was getting a lot of errors. I am sure that it starts to analyse the data but it just creates a lot of errors. My NMR data was obtained using a Varian NMR spectrometer.
I will be gratefully for any help. Thanks in advance!

Best regards,
Krzysztof

tesch1, 2019-02-12:
Hi,
Since Mint is a newer debian-based system, maybe not everything is tested.
To start, you probably want to make sure that the system sh is actually bash instead of dash, as root:
```
echo "dash dash/sh boolean false" | debconf-set-selections
dpkg-reconfigure dash
```
What errors are you getting specifically?  Can you copy/paste, or attach a debugging file?

(to get a log, see this answer: _Originally posted by @DanIverson in https://github.com/OpenVnmrJ/OpenVnmrJ/issues/106#issuecomment-315169808_ )

Cheers,
Michael

DanIverson, 2019-02-13:
The ~/vnmrsys/VnmrjMsgLog file may also contain hints as to what might have gone wrong. Out of curiousity, are you using the pre-made OVJ 1.1 release from github or are you rebuilding the release from the current source code on github. The OVJ 1.1 release includes CRAFT version 1, which was made while Agilent was still doing NMR. That version does CRAFT on basic 1D spectra. I brought /vnmr/fidlib/fid1d into an experiment, adjusted the two cursors to enclose the peaks from 1.9 - 2.1 ppm, selected the CRAFT->Basic CRAFT Tools. In the popup, I gave the workspace a name and then clicked the "Auto-define ROIs between cursors Icon in the Setup section. I then clicked Execute in the Analyze section. Everything seemed to work.

Since the version 1 release of CRAFT, Krish has been updating CRAFT to do 2D CRAFT, water-suppressed CRAFT and many other improvements. I have been updating OVJ to support these new CRAFT features.  If you are interested in some of these new features, you might contact Krish at chempacker@gmail.com

kolmank, 2019-02-13:
Dear Michael and Dan,

First thank you for your answer and willingness to help. Sorry for late answer but I had no access to the computer with OpenVnmrJ installed.  The version I have installed is the precompiled one that can be downloaded from github
Michael
My system had dash as default shell system but I changed it to bash according to your suggestions, however, the problem persist.
I used exactly the same FID file as Dan, however I get such errors:
Information window:
8	retrieving and processing: /home/openvnmrj_1.1_A/fidlib/fid1d.fid
9	Preparing craftspace for fidlib/fid1d.  Please wait..
10	Starting background CRAFT analysis (1 FID; 1 ROIs)
11	CRAFT Analysis errored for /home/openvnmrj_1.1_A/fidlib/fid1d.fid
12	CRAFT background Analysis errored in background process 1
13	Review spectra generation Failed

CraftLog
exp1: 	Preparing CRAFT workspace (Wed Feb 13 21:01:57 CET 2019)
exp1: 	Total prep time 1 sec
exp10: 	Fid 1 of 1: fidlib/fid1d Started
exp10: 	Downsampling 2.14ppm to 1.84ppm region
exp10: 	Bayesian Analysis of 2.14ppm to 1.84ppm region
exp10: 	Downsampling 2.14ppm to 1.84ppm region
exp10: 	Bayesian Analysis of 2.14ppm to 1.84ppm region
exp10: 	Downsampling 2.14ppm to 1.84ppm region
exp10: 	Bayesian Analysis of 2.14ppm to 1.84ppm region
exp10: 	CRAFT background Analysis errored
exp10: 	Total analysis time 4 sec
exp10: 	Review spectra generation Failed

ErrorLog
*** Analysis Parameters for the errored ROI ***
MdlDir /home/vnmr1/vnmrsys/cnmrmodels/space0006/ModelsDir/m00001
data  /home/openvnmrj_1.1_A/fidlib/fid1d
sample  fidlib
fidname  fid1d
band  2.14ppm to 1.84ppm region
---------------------------------------------------------------

~/vnmrsys/VnmrjMsgLog file contains such errors:

## ERROR Feb 13, 2019 21:02:03 CRAFT Analysis errored for /home/openvnmrj_1.1_A/fidlib/fid1d.fid
## ERROR Feb 13, 2019 21:02:03 CRAFT background Analysis errored in background process 1
## ERROR Feb 13, 2019 21:02:03 Review spectra generation Failed
java.sql.SQLException: FATAL:  unsupported frontend protocol 1.0: server supports 2.0 to 3.0

java.sql.SQLException: FATAL:  unsupported frontend protocol 1.0: server supports 2.0 to 3.0

java.sql.SQLException: FATAL:  unsupported frontend protocol 1.0: server supports 2.0 to 3.0

java.sql.SQLException: FATAL:  unsupported frontend protocol 1.0: server supports 2.0 to 3.0

## DEBUG Feb 13, 2019 21:10:13 Liquids  Experiment Selector

Thank you again for help!

Best regards,
Krzysztof

DanIverson, 2019-02-14:
There are potentially several things wrong. As Michael pointed out, bash needs to be the default rather than dash.  OpenVJ version 1.1 does not include certain third party packages. If VnmrJ 4.2 plus the 110 patch are installed, then when OpenVJ is installed it will copy these third party packages from VnmrJ 4.2 into the proper places for OpenVJ. My guess is that you installed OpenVJ without VnmrJ being present.  Therefore, the "Bayes" and "NMRPipe" packages are missing. To correct this, go to your Varian spectrometer and, as vnmr1, do the following
cd /vnmr/craft
zip -ry bayes.zip Bayes3
cd /vnmr
zip -ry nmrpipe.zip nmrpipe

Move the two zip files to your Mint19 system and, as the admin (vnmr1?), do the following
mv bayes.zip /vnmr/craft
mv nmrpipe.zip /vnmr
cd /vnmr
unzip nmrpipe.zip
rm nmrpipe.zip
cd /vnmr/craft
unzip bayes.zip
rm bayes.zip

Finally, I found that csh behaves differently on CentOS and Mint19. The $# variable in not available on the Mint19 system. One of the important CRAFT scripts is failing for this reason. I re-wrote the script as a bash script. The attached zip file has a vnmrbayes script that should be copied to /vnmr/craft/bin
The CRAFT application should then work.

[vnmrbayes.zip](https://github.com/OpenVnmrJ/OpenVnmrJ/files/2866489/vnmrbayes.zip)

kolmank, 2019-02-14:
Thank you so much for help! Unfortunately, I do not have access to vnmr1 account of Varian NMR spectrometer at my institution, I am only able to log in on walkup account.
You are right, I installed OpenVJ at my home computer to analyse my NMR data. Is it possible to avoid this problem by compiling OpenVJ from scratch (do not use the precompiled version)?

Best regards,
Krzysztof

DanIverson, 2019-02-14:
Compiling OpenVJ on a CentOS system would work. I tried it on a Mint19 system and it is not completely straightforward. However, we can still win. Log into your Varian NMR spectrometer as walkup. Then issue the commands
cd /vnmr/craft
zip -ry /home/walkup/bayes.zip Bayes3
cd /vnmr
zip -ry /home/walkup/nmrpipe.zip nmrpipe
Now the zip files will be in walkup's home directory. Move them to your Mint19 system and proceed as above.

kolmank, 2019-02-15:
I checked at the spectrometer computer and there is no folder named craft at /vnmr directory which most likely means that the version of VnmrJ is to old (we have VnmrJ 4.2A with patch 4.2_LNX_DDR_110). I could only find nmrpipe folder.

DanIverson, 2019-02-15:
I think that when VnmrJ 4.2 was released, CRAFT was a passworded option. We can still get what we need. On your Mint19 PC, make sure git is installed (sudo apt install git) and then do the following
cd
mkdir ovjbuild
cd ovjbuild
git clone https://github.com/OpenVnmrJ/OpenVnmrJ.git
cd OpenVnmrJ/src/common/craft
cp -r Bayes3 /vnmr/craft

If you did not get the nmrpipe.zip from the NMR instrument, you can also use an OpenVnmrJ script to download and install it. 
cd ~/ovjbuild/OpenVnmrJ/src/scripts
make ovjGetpipe
./ovjGetpipe

kolmank, 2019-02-15:
Thank you so much! Now it is working like a charm! Few post earlier you have mentioned that there are few updates of CRAFT including 2D CRAFT. You also mentioned that you have included the changes in OVJ. What I have to do to update the existing craft?
After all these problems I see that my Mint is not the best linux distribution for OVJ. What would you suggest as the best way to install it? Vagrant? VirtualBox and Kickstarter image of Centos that can be downloaded from spinsight? Or maybe something else?

DanIverson, 2019-02-15:
That is great news. 

To update CRAFT, you need to contact Krish at chempacker@gmail.com . He and I have worked on CRAFT for 2D, water suppression, reaction monitoring, fingerprinting, and other things are in the works. Some of these developments have not yet made their way onto github.

It sounds like Mint is fine for installing and running OVJ but not for building OVJ. We have also installed VirtualBox on Linux, MacOS, and Windows systems and loaded a guest CentOS and OVJ in the virtual machine. It works fine.

In terms of building OVJ on a Mint system, I believe there will be some changes required to OVJ before it will work. Michael has developed vagrant tools to build OVJ on a number of different OSes. He is the vagrant expert.

kolmank, 2019-02-18:
Good to know. Thank you very much for all the help. I think we can close and mark the issue as solved!.

closed 2019-02-18 by kolmank