OpenVnmrJ/OpenVnmrJ #271: Command "vnmrj" in CentOs 7 does not exists after installation.

kloudpaper, 2019-06-25:
I have just installed OpenVnmrJ in a CentOS 7 virtual box in Windows 10. Everything looks fine for the installation in the releases of OpenVnmrJ. I got for the vnmr/bin the installation log, and this is the cat from the terminal:

vnmrj20190625-1638  vnmrj20190625-1645
localhost.localdomain:vnmr1 22>cat VnmrjInstallationScript.log 
========================================================
 
 ins_vnmr script log 
Tue 25 Jun 16:44:46 CDT 2019
 
========================================================
 
 
OS Version: rht
console type: vnmrs
Dest Directory: /home/openvnmrj_1.1_A
Admin: vnmr1
Group: nmr
Admin Home DIr: /home
Install List: VNMR Backprojection Biosolidspack DOSY_for_VnmrJ Imaging_or_Triax STARS
Install Options List:

=================================================

Distro: rhat
Distro Version: 7.6.1810
User already present, be sure user is configured.
chmod 755 /home/vnmr1
symlink /vnmr/tcl/tklibrary/vnmr /usr/share/tk8.4/vnmr
symlink /usr/bin/convert /vnmr/bin/convert
Unpacking cups filters
gzip: /usr/share/cups/model/*.gz: No such file or directory
chmod 666 /dev/console
cp -rf /vnmr/sudo.lnx/usr/local /usr
cp /usr/bin/sudo /usr/local/bin/sudo
Running sudoers script
/home/nube/Documents/dvdimageOVJ/code/ins_vnmr: line 2545: /home/openvnmrj_1.1_A/bin/sudoins: No such file or directory
sudoers script complete
Software Installation Completed.
 
========================================================
 
Installation script complete Tue 25 Jun 16:45:26 CDT 2019 
 
========================================================

It looks to me I got the installation completed including a new vnmr1 user. But if I typed vnmrj to get into the GUI I got the message that the command does not exist. Is there any error I did ? Any advice is welcome.

Cheers,

Aaron

DanIverson, 2019-06-25:
I think it is just that the Linux PATH variable is not set. I would first try running
   /vnmr/bin/makeuser
and let it automatically update the account. Logout and then login again and see if it works. If not, what does 
  echo $SHELL
and
  ls -a ~
show.

kloudpaper, 2019-06-26:
I did the /vnmr/bin/makeuser after su - vnmr1 and pass. But it looks that the command does not exists either. I type the whole addres to seee if I have a bin directory in vnmr but it looks that there isn't one.

DanIverson, 2019-06-26:
If there is no /vnmr/bin, then something went wrong. OpenVnmrJ will need to be reinstalled. Since OVJ was never qualified with CentOS 7, I would recommend making a new virtual machine with CentOS 6.10. A kickstart developed by Agilent is here ( https://drive.google.com/open?id=0BzjYq4Cic9NLajk1aVN3a1dFMEk )

kloudpaper, 2019-06-26:
I managed to run a new vm but how do I get into the vnmrj GUI? or it is just the pure system to start a new install? Almost there I guess.

DanIverson, 2019-06-26:
The kickstart just installs CentOS 6.10. We still need to install OVJ as vnmr1 with  ./load.nmr 
Note, the kickstart gives agilent1 as the password for both root and vnmr1

kloudpaper, 2019-06-26:
Sorry to be a pain again, but there are two things: The first is that there is not a CentOS 6.10 in the drive but I used the 6.9. After doing so, I try to reinstall from the dvdimage for OpenVnmrJ_1.1-A_DD2 and I got a new error for the dbsetup file persmission. I attached a snapshot from the screen. I tried already umask if I can change the permissions in that particular file and still it doesn´t work.  

And again Sorry to be a pain.

Cheers,

Aaron

DanIverson, 2019-06-26:
CentOS 6.9 is okay. No snapshot is attached.

kloudpaper, 2019-06-26:
Sorry for being a pain. But there are two thing: the first thing is that there is not CentOS 6.10 in the drive link, but I installed 6.9. The second is that I got another error, where the dbsetup permission is denied. I attached the snapshot of screen, I tried umask to change permissions but it didn't work.

And again sorry for being a pain.

Cheers,

Aaron
[image]

DanIverson, 2019-06-27:
For reasons I do not understand, the installation does not work if the dvdimageOVJ directory is in /root.
Do the following as root.
rm /vnmr
rm -rf /home/openvnmrj_1.1_A
unzip the OVJ zip file in /tmp
cd /tmp/dvdimageOVJ
./load.nmr

kloudpaper, 2019-06-27:
It worked! I think we can close it now. 
[image]

closed 2019-06-27 by DanIverson