OpenVnmrJ/OpenVnmrJ #287: OpenVNMRJ on CentOS 7?

DavidGeeraerts, 2019-08-13:
Is it possible to get OpenVNMRJ (Mercury console distribution) to work with CentOS 7?
If I understand the documentation correctly for OpenVNMRJ, VNMRJ 4.2 needs to be already installed (?).
Where could I get VNMRJ 4.2 if I don't have it?
We have an old Varian\Agilent AS400 NMR, currently running on RHEL 5.x with VNMRJ 2.2D.
Our goal is to replace the computer, running CentOS 7 ideally, and if possible, OpenVNMRJ.

Thank you for your time.
David Geeraerts
Scientific Computing Manager | Evergreen State College

DanIverson, 2019-08-13:
Yes, it is possible. I have it working with CentOS 7.4, 7.5, and 7.6. I have not yet committed the required changes to github. I want to do a little beta testing first. I think it will be ready in a month or so. Version 1 of OpenVnmrJ does require that VnmrJ 4.2 and the 110 patch are installed. With this new Version 2, that will not be required.

DavidGeeraerts, 2019-08-15:
That is great news on all fronts! So with OpenVnmrJ v2.0, it will be a stand-alone installer, not requiring any previous VnmrJ installations (?).
We here in the Sciences are grateful for the work you do to continue support for older NMR's.

DanIverson, 2020-01-23:
OpenVnmrJ Version 2 is now released and it supports CentOS 7. It does not require any previous VnmrJ installation.

closed 2020-01-23 by DanIverson