OpenVnmrJ/OpenVnmrJ #334: Menus not appearing on some virtual machines

Gerardwx, 2019-12-06:
**Describe the bug**
On some, but not all, NMRbox virtual machines (VMs) the OpenVMNRJ menus do not appear and the program hangs.

**Version of OpenVnmrJ you are using.**
OpenVmrj VERSION 2.0 REVISION A
October 28, 2019
propulse
 
OS:
Ubuntu 18.04.3 LTS

**It only occurs on some of our VMs, and we can't determine the different between the working and non-working installations. We've done file checks all the files, except those with user names, in /vnmr are the same (based on the linux _sum_ utility)

**Messages and Logs**
~/vnmrsys/VnmrjMsgLog attached
[VnmrjMsgLog.txt](https://github.com/OpenVnmrJ/OpenVnmrJ/files/3933078/VnmrjMsgLog.txt)

DanIverson, 2019-12-09:
This looks like a problem with the postgres Locator. Ubuntu 18 uses postgresql 10. OVJ has a problem with it's new client server model. I removed the Ubuntu postgres and used an older OVJ postgres 7 by doing
  cd /vnmr/pgsql
  mv bin_ver7 bin
Alternatively, you can completely turn off the Locator.

Gerardwx, 2019-12-11:
Thanks for the help. How would we turn Locator off?

DanIverson, 2019-12-11:
You can turn off the Locator by entering
touch /vnmr/pgsql/persistence/LocatorOff
or by running
vnmrj admin
and then from the Configure menu, select
   Locator Config...
There is a "Locator Off" checkbox
However, reviewing the traceback suggests that turning the Locator off may not fix it. Could you run
  vnmrj -debug com
and after it breaks, send the VnmrjMsgLog

Gerardwx, 2019-12-12:
Log attached. This was after _touch /vnmr/pgsql/persistence/LocatorOff
[VnmrjMsgLog.txt](https://github.com/OpenVnmrJ/OpenVnmrJ/files/3956730/VnmrjMsgLog.txt)

_

DanIverson, 2019-12-12:
The traceback made me think it was related to postgres but it turns out to somehow be related to the user id. The line just before the traceback
  com SVF  workspace "exp9" "/home/nmrbox/gweatherby/vnmrsys/exp9"
should have been
  com SVF  <operator> workspace "exp9" "/home/nmrbox/gweatherby/vnmrsys/exp9"
The operator name is obtained from the password file using the users uid. The function 
   getpwuid( getuid() )
is used. Apparently, that function call fails, resulting in the value of operator being set to ''. That could be the cause of the missing menus. The question is, why is getpwuid() failing?

Gerardwx, 2019-12-12:
The menu appearing / not appearing appears independent of the NullPointerException; the example log 
[VnmrjMsgLogMenusAppear.txt](https://github.com/OpenVnmrJ/OpenVnmrJ/files/3957527/VnmrjMsgLogMenusAppear.txt) is from a VM where the menus work.

**Com issue:**
Our id information is coming from an LDAP server. The small test program 
[usernametest.cpp.txt](https://github.com/OpenVnmrJ/OpenVnmrJ/files/3957534/usernametest.cpp.txt) properly fetches the correct information:
./openvnmrsupport 
id = 16342
pwentry = gweatherby

Running openvnmrj from an account defined in the password file does produce the com line expected:
# DEBUG Dec 12, 2019 14:42:36 com SVF localacct workspace "exp6" "/home/localacct/vnmrsys/exp6"
, not sure what the discrepancy is.

DanIverson, 2019-12-12:
Can you copy the login macro from the attached zip to /vnmr/maclib for one system that works and one that doesn't'. Run vnmrj and then send the generated /vnmr/tmp/debugXXX files
Remove the login file when you are done.

[login.zip](https://github.com/OpenVnmrJ/OpenVnmrJ/files/3958258/login.zip)

Gerardwx, 2019-12-13:
Results follow. 

[debug881067works.txt](https://github.com/OpenVnmrJ/OpenVnmrJ/files/3960979/debug881067works.txt)
[debug3414996fail.txt](https://github.com/OpenVnmrJ/OpenVnmrJ/files/3960980/debug3414996fail.txt)

DanIverson, 2020-03-02:
It it not clear why it is failing. Perhaps if someone is coming to ENC, they could bring the failing VM and I can try to figure out what is going wrong.

Gerardwx, 2020-04-04:
At the least minute most of the team elected not to go to ENC due to COVID.

We have our next release coming out this month, we'll have to see if the problem reoccurs. (We never see it on a VM that has been booted recently.)

DanIverson, 2020-04-04:
Okay. Note that I put an official OpenVnmrJ Version 2.1 on the github releases page in Jan. 2020

Gerardwx, 2020-04-04:
So we should package Version 2.1  with NMRbox Version 7 rather than the current Master?

DanIverson, 2020-04-04:
It is up to you but that would be my preference.

DanIverson, 2020-08-31:
Since the VnmrjMsgLog and debug output match what I was seeing on a system with seven digit PID values, I am going to assume this is fixed. If it happens again and the PID is less than seven digits, we can re-open it.

closed 2020-08-31 by DanIverson