OpenVnmrJ/OpenVnmrJ #357: Failure at “console download” step

ctsubasaddr, 2020-02-10:
**Describe the bug**
We have a VNMRS that is using VNMRJ 2.1C, and we wanted to load openvnmrj on a new hard drive. (The computer would stay the same, just a different hard drive). We successfully installed CentOS 7.7 via the everything ISO (tried both  KDE and Gnome, all options installed). Due to lack of internet access, an offline install of OVJ was attempted. Everything seemed to go to perfectly well until the console download part. An error pops up stating that libutilsxx.so is missing, and the download fails. Running OVJ after this results in “inactive” console mode even when ‘su acqproc’ confirms that the communication has started.

**To Reproduce**
Steps to reproduce the behavior:
1. Clean install of CentOS 7.7 using everything ISO.
2. Kde or gnome guis along with all the available options installed.
3. Follow the install guide on installing OVJ-ddr without network.
4. Reboot after Load.nmr loads all packages.
5. Re-run load.nmr, install Varian Options, add users in vnmrj admin, skip nmrpipe and manuals.
6. Setup console networking on correct nic card and default ip. 
7. During console download, the libutilsxx.so missing error pops up.

**Version of OpenVnmrJ you are using.**
- Latest OVJ release from github (VNMRS version, ddr)

**Environment (please complete the following information):**
 - OS: CentOs 7.7

ctsubasaddr, 2020-02-10:
Hi Dan, Thank you. I was wondering if you could answer a related question. Would it be correct to say that once the console downloads the software for OVJ, vnmrj 2.1b will cease to function? If so, is there a restore feature similar to the one in vnmrj 3.0?

DanIverson, 2020-02-10:
This is an oversight. VnmrJ 2.1C uses an old version of NDDS (version 3) which requires these libraries. A zip file containing these libraries may be downloaded from https://www.dropbox.com/s/ypyowqn8pttviqe/ndds3.zip?dl=0  After unzipping the file, put the two libraries (libutilsip.so and libutilsxx.so) in /vnmr/lib. The repository ovjTools now includes these libraries as well.

DanIverson, 2020-02-10:
There is no simple way to restore the system to function with VnmrJ 2.1b after it is updated to OpenVnmrJ or a newer version of VnmrJ, such as VnmrJ 4.2.  It requires reflashing the console controller boards. It may be possible using the verifyCntlrsFlash.py in /vnmr/adm/acq. See the document Reflashing_Script.pdf for details. If you are concerned about Agilent support, you could first upgrade to VnmrJ 4.2 and then install OpenVnmrJ. You could then use the restore feature.

NeutralKaon, 2020-02-10:
Hi Dan 

I might be being dim, but doesn't a setacq reflash the boards with the currently compiled psg -- did that not exist in 2.1? 

I have found version hopping a _relatively_ painless experience, but only from 3 onwards. 

All the best, 

-- Jack

DanIverson, 2020-02-10:
setacq did reflash the boards as needed. I don't remember exactly when, but either VJ 2.2 or 2.3 changed from using NDDS version 3 to NDDS version 4. We had to add some special reflashing programs to handle this switch. These programs are in /vnmr/acqbin; consoledownload3x, flashia3x, and testconf3x. The problem with OVJ was that some libraries used by these programs were not present. That caused the reflashing to fail.

I just checked the VJ 3 installation guide and there is a mechanism to restore VJ 2.x after upgrading to VJ 3.x.  This has never been tested with OVJ but I am willing to bet that OVJ could be coaxed into downgrading back to VJ 2.x

ctsubasaddr, 2020-02-11:
Would a script like this (https://github.com/timburrow/ovj3/blob/master/src/scripts/restore3x.sh) be useful for the rollback, provided the correct files are provided? (sorry, I'm still new to Linux)

DanIverson, 2020-02-11:
With a small change, it may work. Line 59 needs to be changed from
cp /vnmr/bin/lnvsetacq2 lnvsetacq
to
cp /vnmr/bin/setacq lnvsetacq
Again, this has not been tested.

closed 2021-09-24 by DanIverson