OpenVnmrJ/OpenVnmrJ #361: Installation on Windows 10 WSL

andrazek, 2020-03-17:
Hi! Thanks for the wonderful work. I was wondering whether it's possible to install OpenVnmrj 2 on windows 10 with WSL Ubuntu 18.04? In the past, I was using windows build of vnmrj 3 which worked on Windows XP and Windows 7 (with SUA tools). But as Win 7 won't get any updates it's necessary to move on and switch to Windows 10, which doesn't have SUA, but they've added support for WSL.

I've already tried to install OpenVnmrj and was almost successful. After the installation and setting the users executing vnmrj leads to an error stating "Vnmrbg could not be started". Any suggestions?

DanIverson, 2020-03-17:
I have not tried WSL, however I know OVJ 2 works on Ubuntu 18.04. There are two common causes for Vnmrbg failing to start. Missing libraries or a network issue. To test the first, run from a terminal,
  Vnmrbg -mback -n1 dg
If that shows the "dg" parameters, then the libraries are present. If it fails, the command
   ldd /vnmr/bin/Vnmrbg
will tell you which libraries are missing.

For the network issue, can you 
   ping `hostname`
If not, that is the problem. I am not sure what the fix is in the WSL environment.

andrazek, 2020-03-18:
I cannot run Vnmrbg:
-bash: /vnmr/bin/Vnmrbg: cannot execute binary file: Exec format error

Is this an 32-bit executable? Surely there have to be some workarounds to make it run.

DanIverson, 2020-03-18:
It is a 32-bit executable. It may be that some libraries are missing. What does 
    ldd /vnmr/bin/Vnmrbg
show. It the 32-bit libraries are missing, you might try the following to install them
    sudo dpkg --add-architecture i386
    sudo apt-get update

andrazek, 2020-03-18:
ldd returns a message "not a dynamic executable".
I've tried to add architecture i386, but still the same Exec format error.

tesch1, 2020-03-19:
are you using WSL2?  https://github.com/microsoft/WSL/issues/2468

andrazek, 2020-03-19:
Not yet, as it will only be available in April 2020. But thanks for the info, now I am full of hopes. Will report it here as soon as I try it.

DanIverson, 2020-08-09:
Can you describe how you set up the X11 server in the WSL2 environment?

andrazek, 2020-08-10:
Install vcXsrv then launch XLaunch and disable access control. Save config.xlaunch to VcXsrv folder and place a shortcut with "C:\Program Files\VcXsrv\xlaunch.exe" -run "config.xlaunch" as a target in C:\ProgramData\Microsoft\Windows\Start Menu\Programs\StartUp\ folder.

In Ubuntu bash execute the following:
echo "export DISPLAY=\"\$(awk '/nameserver/ { print \$2 }' < /etc/resolv.conf)\":0"  >> ~/.bashrc

Restart the terminal, it should work now.

closed 2020-08-27 by andrazek

darekb94, 2022-11-15:
> 

> It's working now! WSL2 was required. It will be officially available in April 2020, but everyone can decide to join the Insider preview which offers updates in advance. Thank you!!

Recently, I have tried to install OpenVnmrJ 3.1 on WSL2 on Windows 11 but unfortunately I encountered the problems similar to those described above. I would like to use this program for data analysis only so I did not configure it as a spectrometer. The problems are as follows:
-  after launching OpenVnmrJ, a window opened for a while, but then I returned error "Vnmrbg could not be started" and the program had to be closed down
- I could not invoke Vnmrbg from the terminal (allthoug it exists in the folder vnmr/bin) due to an error "No such file or directory" whereas a command ldd Vnmrbg returns "       not a dynamic executable" (with many spaces at the beginning)
- ovjDiag returns the same error: "       not a dynamic executable" (with many spaces at the beginning)
- the file VnmrjMsgLog ends with the following error:

"makeDBConnection failed on <here my hostname> port 5432,
be sure database exists and that vnmr1 has access to it.
Try having the system administrator execute the command 'createuser vnmr1'.
Also check that if a special port number  is used for postgres,
that this user has the environment variable 'PGPORT' set  correctly.
Is it possible that a specific error can be found in the log file:
/vnmr/pgsql/pgsql.log "
- I tried to invoke 'createuser vnmr1' but I obtained the following error:

"createuser: error: connection to server on socket "/var/run/postgresql/.s.PGSQL.5432" failed: No such file or directory
        Is the server running locally and accepting connections on that socket?"
- the file '/vnmr/pgsql/pgsql.log' has the following content:

"LOG:  database system was shut down at 2022-11-12 20:51:46 CET
LOG:  MultiXact member wraparound protections are now enabled
LOG:  database system is ready to accept connections
LOG:  autovacuum launcher started
LOG:  autovacuum launcher shutting down
LOG:  received smart shutdown request
LOG:  received immediate shutdown request
WARNING:  terminating connection because of crash of another server process
DETAIL:  The postmaster has commanded this server process to roll back the current transaction and exit, because another server process exited abnormally and possibly corrupted shared memory.
HINT:  In a moment you should be able to reconnect to the database and repeat your command.
LOG:  database system is shut down
LOG:  database system was shut down at 2022-11-13 15:10:54 CET
LOG:  MultiXact member wraparound protections are now enabled
LOG:  database system is ready to accept connections
LOG:  autovacuum launcher started
ERROR:  role "walkup" already exists
STATEMENT:  CREATE ROLE walkup NOSUPERUSER CREATEDB NOCREATEROLE INHERIT LOGIN;
ERROR:  role "service" already exists
STATEMENT:  CREATE ROLE service NOSUPERUSER CREATEDB NOCREATEROLE INHERIT LOGIN;"

Can anyone find a solution to these problems?
Is it really possible to run OpenVnmrJ on WSL2?