OpenVnmrJ/OpenVnmrJ #373: Vnmrbg could not be started.

whoa2k, 2020-05-27:
Hi. Thank you for wonderful work to make openvnmj running. 

After download 'OpenVnmrJ_2.1_DDR.zip" and I could  to install 'openvnmrj 2.1A' on CentOS 7.8

But I launch 'vnmrj' on terminal, error appeared as below
________________________________________________________________________
"Vnmrbg could not be started. This could be due to network issues,
missing libraries, or other reasons. Please run /vnmr/bin/ovjDiag
to help track down the cause of the failure. "
________________________________________________________________________

I ran '/vnmr/bin/ovjDiag', but I can't see any diagonostic message. 

Also I tried to install 'OpenVnmrJ-2.1_DDR.zip" on Window10 WSL2 (Ubunt 20.04 LTS) 
but same trouble happens.

# 'vnmrj adm' is working, but status is still 'initializing' after 'update user'
# 32bit library was installed with yum or apt-get

# 1st try 
OS :  CentOS 7.8 (64bit) 
Workstation : HP Z4 Workstation (VGA : Quardro P400) 
version : openvnmrj 2.1A (DDR)

# 2nd try
OS :  Ubuntu 20.04 LTS on Windows 10 WSL2  (+VcXrv X Server) 
Workstation : Intel CPU i5 + 16G RAM (VGA : intel display)
version : openvnmrj 2.1A (DDR)

Please give any suggestion for troubleshooting. 
Thanks in advance.

DanIverson, 2020-05-27:
Could you send me your ~/vnmrsys/VnmrjMsgLog file. Also, what happens when, from a terminal, you run
  Vnmrbg -mback -n1 dg

whoa2k, 2020-05-27:
Thanks for your prompt response. 

I attached VnmrjMsgLog file and Vnmrbg log. 
# in case of win10, Vnmrbg doesn't work so I write error message in file. 

[log_whoa2k_openvnmrj.zip](https://github.com/OpenVnmrJ/OpenVnmrJ/files/4686637/log_whoa2k_openvnmrj.zip)

DanIverson, 2020-05-27:
From the vnmrbg.log.centos7 file, Vnmrbg is running. From the VnmrjMsgLog files, something is going wrong. Is it possible for me to access your system remotely?

whoa2k, 2020-05-27:
Unfortunately, server is blocked by VPN now.  
If you don't mind, let me know you email address to send VPN account and password, server IP address.
( I request temporary VPN account to IT service team now, it will take 1~2 days for approval.)

DanIverson, 2020-05-27:
My email is daniverson2016@gmail.com . I am not 100% certain I can fix it
but I will certainly try.

whoa2k, 2020-05-29:
Problem is solved!!!

As you mention in e-mail, problem is related to host name. 
hostname of localhost at '/etc/hosts' file is not modified properly 
OpenVnmrJ is working, after modifying '/etc/hosts'

Thanks for kind help.

DanIverson, 2020-05-29:
That’s great.  Can you send me your updated /etc/hosts file

whoa2k, 2020-05-29:
I sent a e-mail of my /etc/hosts. 

By the way, in case of Win10/Ubuntu, after it showovjDiag

whoa2k, 2020-05-29:
I close this issues because I find now that this issue is already discussed.  #361 
> 
> 
> I sent a e-mail of my /etc/hosts.
> 
> By the way, in case of Win10/Ubuntu, after it showovjDiag it shows error like this
> 
> not a dynamic excutable
> /vnmr/bin/ovjDiag complete
> 
> Is openvmrj not possible on Winodw10/Ubuntu? or can I try something ?

muralivenkat1, 2021-10-15:
can you please let me know how to add hostname /etc/hosts

DanIverson, 2021-10-15:
From a terminal, enter
  hostname
to determine the name. Then, as root, edit the /etc/hosts file and either append the host name to the line that starts with 127.0.0.1, or if the PC has its own IP address, add that IP address and the host name on a single line in the /etc/hosts file. When you finish, the command
  ping `hostname`
should not give an error.

DanIverson, 2021-10-19:
Turned out to be missing RHEL packages

closed 2021-10-19 by DanIverson