OpenVnmrJ/OpenVnmrJ #400: Vnmrbg could not be started

ruetherm, 2020-08-07:
**Describe the bug**
Installation of OpenVnmrJ on Centos 8.2 worked fine. If main program is started an error message "Vnmrbg could not be started" is shown. Clicking on "OK" will close the program. Other GUI start and work normally i.e OpenVnmrJ Administrtor, Accountant. Only the main program currently give this error message.

**To Reproduce**
Steps to reproduce the behavior:
This error occurs always when starting OpenVnmrJ

**Expected behavior**
To open and run OpenVnmrJ normally

**Version of OpenVnmrJ you are using.**
- Using newest version 2,1
- Downloaded from GitHub 
- Followed installation instruction. Installation was performed with "noPing" as our institution does not support "ping" to the outside world

**Messages and Logs**
## INFO  Aug 7, 2020 10:04:51 **********************************************
## INFO  Aug 7, 2020 10:04:51 ********** Starting OpenVnmrJ as vnmr1 **********
## INFO  Aug 7, 2020 10:04:51 ** OpenVnmrJ VERSION 2.1 REVISION A **
## INFO  Aug 7, 2020 10:04:51 ** January 22, 2020 **
## INFO  Aug 7, 2020 10:04:51 ** vnmrs **
## INFO  Aug 7, 2020 10:04:51 **********************************************
## DEBUG Aug 7, 2020 10:04:57 Calibration  Experiment Selector
## DEBUG Aug 7, 2020 10:06:09 Calibration  Experiment Selector
## INFO  Aug 7, 2020 10:20:03 **********************************************
## INFO  Aug 7, 2020 10:20:03 ********** Starting OpenVnmrJ as vnmr1 **********
## INFO  Aug 7, 2020 10:20:03 ** OpenVnmrJ VERSION 2.1 REVISION A **
## INFO  Aug 7, 2020 10:20:03 ** January 22, 2020 **
## INFO  Aug 7, 2020 10:20:03 ** vnmrs **
## INFO  Aug 7, 2020 10:20:03 **********************************************
## DEBUG Aug 7, 2020 10:20:09 Calibration  Experiment Selector
## DEBUG Aug 7, 2020 10:25:04 ** Start DataBase Contents Report **
## DEBUG Aug 7, 2020 10:25:04 DB version: pg_ctl (PostgreSQL) 7.4.5  DB Host: spinecho-chy-tcd-ie
              Local Host: spinecho-chy-tcd-ie     Network Mode: false
## DEBUG Aug 7, 2020 10:25:04 Number of vnmr_data items in DB = 0
## DEBUG Aug 7, 2020 10:25:04 Number of vnmr_par items in DB = 0
## DEBUG Aug 7, 2020 10:25:04 Number of vnmr_record items in DB = 0
## DEBUG Aug 7, 2020 10:25:04 Number of vnmr_rec_data items in DB = 0
## DEBUG Aug 7, 2020 10:25:04 Number of automation items in DB = 0
## DEBUG Aug 7, 2020 10:25:04 Number of workspace items in DB = 7
## DEBUG Aug 7, 2020 10:25:04 Number of panels_n_components items in DB = 1213
## DEBUG Aug 7, 2020 10:25:04 Number of shims items in DB = 1
## DEBUG Aug 7, 2020 10:25:04 Number of command_n_macro items in DB = 99
## DEBUG Aug 7, 2020 10:25:04 Number of pulse_sequence_macro items in DB = 88
## DEBUG Aug 7, 2020 10:25:04 Number of protocol items in DB = 237
## DEBUG Aug 7, 2020 10:25:04 Number of study items in DB = 0
## DEBUG Aug 7, 2020 10:25:04 Number of lcstudy items in DB = 0
## DEBUG Aug 7, 2020 10:25:04 Number of image_dir items in DB = 0
## DEBUG Aug 7, 2020 10:25:04 Number of trash items in DB = 0
## DEBUG Aug 7, 2020 10:25:04 Total Number of vnmr_data attributes = 91
## DEBUG Aug 7, 2020 10:25:04 Total Number of study attributes = 52
## DEBUG Aug 7, 2020 10:25:04 Total Number of protocol attributes = 90
## DEBUG Aug 7, 2020 10:25:04 Total Number of image_dir attributes = 70
## DEBUG Aug 7, 2020 10:25:04 ** End DataBase Contents Report **
## DEBUG Aug 7, 2020 10:29:45 Calibration  Experiment Selector

**Environment (please complete the following information):**
 - OS: Centos 8.2

**Additional context**
Other GUI start and work normally i.e OpenVnmrJ Administrtor, Accountant. Only the main program currently give this error message. OpenVnmrJ is not connected to a spectrometer.

DanIverson, 2020-08-08:
Did you try running the ovjDiag script? It might give us a clue. The normal cause is network configuration. What do you get from running the following from a terminal:
   hostname
   ping `hostname`

ruetherm, 2020-08-08:
Hi Dan,

yes I tried pinging the host and that is not a problem. I also have internet access.

Running the script ovjDiag does not show anything. It just says "completed". Is there a log file saved somewhere when running the script?

DanIverson, 2020-08-08:
There is no log file for the ovjDiag script. Could you send me the ~/vnmrsys/VnmrjMsgLog file? What do you get when you run the following from a terminal window:
  Vnmrbg -mback -n1 dg

ruetherm, 2020-08-10:
[VnmrjMsgLog.zip](https://github.com/OpenVnmrJ/OpenVnmrJ/files/5049705/VnmrjMsgLog.zip)

I attached the VnmrjMsgLog file

The output for Vnmrbg -mback -n1 dg is:
[vnmr1@spinecho Desktop]$ Vnmrbg -mback -n1 dg
Exp:1   
Index: 1   
           
Seq: PROTON
Exp:1   
Index: 1   
           
   ACQUISITION         TRANSMITTER           SPECIAL            PROCESSING
seqfil       s2pul  tn              H1  temp          25.0  lb            0.30
sw          8012.8  sfrq       499.838  spin             0  sb        not used
at           2.045  tof          499.7  gain            34  gf        not used
np           32768  tpwr            58  hst          0.008  awc       not used
fb            4000  pw           4.900  pw90         9.800  lsfid     not used
bs              32      DECOUPLER       alfa        10.000  fn          262144
ss               2  dn             C13    PRESATURATION           FLAGS
d1           1.000  dof              0  satmode          n  il               n
nt               8  dm             nnn  wet              n  in               n
ct               8  decwave W40_HCN5mm   Details in dgss    dp               y
      SAMPLE        dpwr            40                      hs              nn
date   Mar 30 2014  dmf          32258
solvent      cdcl3
sample Ethylindano~
                ne
SVPAR   
SAVEDATA

ruetherm, 2020-08-10:
Hi Dan,

problem is solved. The hostname was not in the .etc/host file. That caused the problem. The program seems to work now.

Thank you for all your help,
Regards,
Manuel

closed 2020-08-10 by ruetherm