OpenVnmrJ/OpenVnmrJ #403: semsdw timing issue

Guangyu-Dan, 2020-08-10:
In the MRI sequence src/IMAGE/psglib/semsdw.c, from line 155 to 156. 

I think 
del2 = te_delay1 - diff_grad.duration - del1;
del3 = te_delay2 - diff_grad.duration - del4;

Line 159 to 151, I think 
del1 += del3;
del2 -= del3;
del4 -= del3;

Same issue for line 164 to 166.

Could anyone answer my problem?

alanrath, 2020-08-10:
I didn't write this sequence, but I think what this code does is shift the diffusion gradient event in the window to account for pulse lengths, TE, diffusion gradient length.

What exactly is your question?

Guangyu-Dan, 2020-08-10:
I think the original code was wrong. Could you please check the timing issue in this sequence, or help me to contact who wrote this sequence?

alanrath, 2020-08-10:
Since Agilent killed it's MRI and NMR business, there is no one left inside the company who has expertise of this kind. The semsdw sequence evolved over time, and I'm not sure who last modified it. But if you think there is an error, you can draw out the sequence, and check the calculations as easily as anyone else.

But, I do know that the sems.c sequence was modified later to include diffusion, and I think it is a better option if that's what you are trying to do. Use sems, not semsdw.