OpenVnmrJ/OpenVnmrJ #419: setacq fails on Inova after installing OpenVnmrJ

srburt, 2020-09-17:
**Describe the bug**
After the install script runs setacq (or manually running setacq) the console won't bootup after pressing the reset button.

**To Reproduce**
Inova 300 MHz spectrometer has been running VnmrJ 4.2A (latest patch) for years.
Installed OpenVnmrJ 2.1 (Jan 2020 release), told it to configure as a spectrometer.
When the install script reaches setacq (or if manually running setacq) the console starts to reboot, but never finishes (the status light on the magnet regulation board remains red).
Because of this, OVJ claims it's a datastation and won't operate the spectrometer.

Work-around: 
If I run /home/vnmrj_4.2_A/bin/setacq then everything works as expected and the console is able to boot up.
OVJ appears to work fine after using the old setacq.  I don't know if this is dangerous or not recommended.

**Expected behavior**
setacq should work as it normally does.  After being instructed to reboot the spectrometer, the spectrometer should boot up all the way until showstat can successfully show the spectrometer status.

**Version of OpenVnmrJ you are using.**
- Downloaded from github in Aug.  Release date in the package says Jan 2020
- Unzipped, ran load.nmr script

**Environment (please complete the following information):**
 - RedHat 6.10

**Additional context**
Note: It appears to work fine on our DD 500 MHz instrument (same OS, same VnmrJ, but DD version, etc.)

DanIverson, 2020-09-18:
It is not dangerous to use the VJ42 setacq. The console software in VJ42 + 110 patch is the same as that in OVJ.
Can you tell what is different when you use the two setacqs. Do they both select the same NIC card (eth0 vs eth1)? Do they both select the same IP address (10.0.0.x vs 172.16.0.x)? Do they both use the same tftpboot directory (/tftpboot vs /var/lib/tftpboot)? Are the permissions of the files in the tftpboot directory the same? It is /etc/bootptab file the same? Does the OVJ version of setacq ever give the message "NMR Console software installation complete"?

srburt, 2020-09-18:
Dan,

	When running the two different setacqs, the only difference I see in the text output is the following:

original setacq) It gives the option of doing the blink test (which I skip, because it's set correctly).

openVJ setacq) Doesn't do the blink test, but appears to look at the settings of eth0 and eth1 to decide which is correct.  It correctly chooses eth1 on my setup.  

	Yes, they both report using 172.16.0.1 with eth1.

	They both appear to be using /var/lib/tftpboot
original setacq) It looks for the existence of a /var/lib/tftpboot dir, which exists on this instrument.
openVJ setacq) It doesn't appear to look for the existence of /var/lib/tftpboot, but it does look at /etc/xinet.d/tftp and that file has server_args = -s /var/lib/tftpboot
	There does not appear to be a environment variable setting $tftpdir (at least echo $tftpdir doesn't return anything), so it looks like they're both pointing to /var/lib/tftpboot. 

	Everything in /var/lib/tftpboot is owned by root:root, most of it just symbolic links.  The dir vxBoot is 755
	Within vxBoot, vxWorks and vxWorksPPC are owned by vnmr1:nmr, but vxWorks.auto is owned by root:root.  They're all 755
	/etc/bootptab is own by root:root and is 644

	Yes, the openVJ reports "NMR Console software installation complete".  Note, however, with the original setacq, it says this before the data transfer lights start blinking on the NVME board.  With the openVJ setacq, no data transfer lights are seen after this message appears (see below).

	The real difference that I see is watching the status lights on the NVME and sample regulation boar after the reset + enter step.

original setacq) After a couple seconds (i.e. after the "installation complete" message) I can see lots of fast blinking of the lights on the VNME board corresponding to data transfer, after which the magnet regulation board status light goes from red to green followed by the lightshow.

openVJ setacq) There are never any signs of data transfer on the NVME board, even after several minutes. The magnet regulation board status remains at red and no lightshow happens.

	Note: Dennis Edwards suggested doing a full power cycle to ensure there wasn't any corrupt files in the spectrometer messing things up (apparently he's seen that on the Inova consoles before).  However, after power cycling and then running the new setacq I still see the same behavior, whereas the original setacq transfers the files and then things work.

	Again, this is only my Inova 300 MHz doing this.  The DirectDrive 500 MHz worked fine.

-Scott

---
Dr. Scott Burt
Teaching Professor
NMR Facility Manager
Department of Chemistry and Biochemistry
Brigham Young University
C414 BNSN / C008A BNSN
Provo, Utah  84602-5700
Phone:  (801) 422-2404
Fax: (801) 422-0153
chembio.byu.edu/nmr-facility <https://chembio.byu.edu/nmr-facility>
email: srburt@chem.byu.edu

DanIverson, 2020-09-18:
All that the setacqs do is setup Linux system files to handle the console software download, so there must be a difference somewhere. With a functioning system, could you make a copy of /etc/bootptab. Then run setNIC, which will have the option for a blink test. Then run OVJ setacq. Are the bootptab files before and after the OVJ setacq identical? Finally, could you do the following.
   script log
   su
   tcpdump -i eth1

Then reboot the console. After a minute or so, go back the the terminal and cntrl-c to exit the tcpdump command,
cntrl-d to exit root account, and cntrl-d to quit the script command. Then send me the log file.

srburt, 2020-09-21:
Here's the results:

1) Running setNIC correctly blinks eth0 (which is connected to the wall), but when it goes to blink eth1 it throws an error saying it couldn't identify the NIC.

2) Running diff on copies of /etc/bootptab before and after setacq show the files are identical.

3) I ran script log twice:
	-after running the old VnmrJ setacq so the spectrometer is working before logging and pressing reset: log_oldsetacq
	-after running the new VnmrJ setacq so the spectrometer is not working before logging and pressing reset: log_newsetacq

-Scott

---
Dr. Scott Burt
Teaching Professor
NMR Facility Manager
Department of Chemistry and Biochemistry
Brigham Young University
C414 BNSN / C008A BNSN
Provo, Utah  84602-5700
Phone:  (801) 422-2404
Fax: (801) 422-0153
chembio.byu.edu/nmr-facility <https://chembio.byu.edu/nmr-facility>
email: srburt@chem.byu.edu

DanIverson, 2020-09-21:
There are no log files attached. Also, can you compare /etc/sysconfig/network-scripts/ifcfg-eth1 files for a working and non-working system. Also, compare results of running  /sbin/ifconfig -a for working and non-working system.

srburt, 2020-09-21:
That's odd, my email software claims they were attached...  Trying again.  If this doesn't work, I guess I can try adding an extension to the files?

-Scott

DanIverson, 2020-09-21:
Still no attachment. You can send them to me at daniverson2016@gmail.com

closed 2020-09-23 by DanIverson