OpenVnmrJ/OpenVnmrJ #456: lock failure occurs after gradient shimming failed

whoa2k, 2020-10-29:
**Describe the bug**
- lock failure occurs after gradient shimming failed 

**To Reproduce**

1. click 'find z0' 
2. click 'Gradient Shimming ' 
<== Shimming failure happens (maybe nmr sample preparation is not good - bad nmr tube or poor mixing -)
==>  Z3~Z5 values are out-of-range

3, lock frequency is adjusted automatically ex> "Parameter z0 reset to minimum of -32767"

4. lock failure happened
<== lock frequency at 'system config' is changed, z0 value is out-of-range 

5. after restoring lock frequency and shimming parameters, lock problem is fixed. 

**Expected behavior**
- lock frequency is changed improperly after gradient shimming failed

**Version of OpenVnmrJ you are using.**
- Software : openvnmrJ 2.1A
- OS : CentOS 7

**Messages and Logs**
## DEBUG Oct 29, 2020 13:31:11 Login: (AAAAAA)
## DEBUG Oct 29, 2020 13:31:11 ### appDir 0 /vnmr/personalib/daytime
## DEBUG Oct 29, 2020 13:31:11 ### appDir 1 /home/vnmr1/vnmrsys/CPpatch
## DEBUG Oct 29, 2020 13:31:11 ### appDir 2 /home/vnmr1/vnmrsys
## DEBUG Oct 29, 2020 13:31:11 ### appDir 3 /vnmr
## ERROR Oct 29, 2020 13:35:20 shim z5 out-of-range
## DEBUG Oct 29, 2020 13:37:14 Login: (BBBBBBB)
## DEBUG Oct 29, 2020 13:37:14 ### appDir 0 /vnmr/personalib/Owner
## DEBUG Oct 29, 2020 13:37:14 ### appDir 1 /home/vnmr1/vnmrsys/CPpatch
## DEBUG Oct 29, 2020 13:37:14 ### appDir 2 /home/vnmr1/vnmrsys
## DEBUG Oct 29, 2020 13:37:14 ### appDir 3 /vnmr
## DEBUG Oct 29, 2020 13:37:14 ### appDir 4 /vnmr/gxyzshim
## DEBUG Oct 29, 2020 13:37:17 ### appDir 0 /vnmr/personalib/daytime
## DEBUG Oct 29, 2020 13:37:17 ### appDir 1 /home/vnmr1/vnmrsys/CPpatch
## DEBUG Oct 29, 2020 13:37:17 ### appDir 2 /home/vnmr1/vnmrsys
## DEBUG Oct 29, 2020 13:37:17 ### appDir 3 /vnmr
## DEBUG Oct 29, 2020 13:37:17 ### appDir 4 /vnmr/gxyzshim
## ERROR Oct 29, 2020 13:40:32 shim z4 out-of-range
## ERROR Oct 29, 2020 13:41:23 shim z4 out-of-range
## ERROR Oct 29, 2020 13:41:55 shim z5 out-of-range
## ERROR Oct 29, 2020 13:42:03 shim z4 out-of-range
## DEBUG Oct 29, 2020 13:43:23 Login: (BBBBBBB)
## ERROR Oct 29, 2020 13:43:49 Set hardware: value of -41016 for z0 is too small
## ERROR Oct 29, 2020 13:44:06 shim z5 out-of-range
## ERROR Oct 29, 2020 13:44:24 shim z5 out-of-range
## ERROR Oct 29, 2020 13:46:21 Set hardware: value of -42172 for z0 is too small
## ACQ ERROR Oct 29, 2020 13:47:07 exp3:  Acquisition aborted
## ERROR Oct 29, 2020 13:47:28 Set hardware: value of -41867 for z0 is too small

If applicable, attach logs to help explain your problem.  Log files can be found
- copy/paste the error message text here,
- in ~/vnmrsys/VnmrjMsgLog,
- by enabling recording while performing the action that produces an error (from the OpenVnmrJ command line):
```
jFunc(55,'/vnmr/tmp/plbug')
debug('c3')
```
and then do whatever you did to generate the error. After the error appears, enter
`debug('C')` to end the recording, and then attach the /vnmr/tmp/plbug file.

**Environment (please complete the following information):**
 - OS: [e.g. CentOS 5, Ubuntu 14.04, etc..]

**Additional context**
Add any other context about the problem here.

DanIverson, 2020-10-30:
I don't think it is related to this problem but you should turn off the CPpatch appdir. Chempack is integrated with OVJ 2.1 and the things in CPpatch are out of date.
You might try the "z0 calibration and H2 (lk) gradient map" in Veripulse to see if it helps.

whoa2k, 2020-11-01:
Thanks for your comment. 

I have two question. 

1. How can I turn off CPpatch appdir? 
I can't find CPpatch under /home/vnmr1/vnmrsys. 

2. veripulse is not appeared at 'Tool' on menu bar . 
I saw veripulse at 'Tool' on VNMRJ 4.2A but I can't see it on OVJ 2.1
How can I make veripulse appear at 'Tool' ?

DanIverson, 2020-11-01:
From the menu system, select Edit->Applications...  In the popup, on the line with CPpatch, select Remove(d) from the menu at the left. On the line with /vnmr/veripulse, select Enabled from the menu at the left. Then click the OK button.

whoa2k, 2020-11-05:
Thank you very much. 
As your comment, I turned off CPpatch and turned on veripulse. 
I will keep watching to monitor when exactly bug happens.