OpenVnmrJ/OpenVnmrJ #472: tftp package is not available; prevents

matwillans, 2020-12-14:
Attempting to run OpenVnmrJ 2.1 with CentOS 8.2 and an Inova spectrometer.  Installation seemed to go smoothly, except when I get to the /vnmr/bin/setacq step.  Here's the output in the terminal window when that command is issued:

**_Verify NIC setup
Verify NIC enp2s0 with IP address 172.16.0.1
Network for the spectrometer is configured correctly

Please reboot the console.
Press the Enter key after pressing the console reset button:

One moment please...
tftp package is not available.  Exit.**

So the setacq does not finish and no connection between the computer and the spectrometer is established.  Quick check shows that the rpms tftp-5-2-24.el8.x86_64 and tftp-server-5.2.24.el8.x64_64 have been installed.  Not sure how to proceed.  Any help would be appreciated.

DanIverson, 2020-12-15:
I found the problem. When you installed OVJ2.1, did you use the ver81.zip file? If not, you should remove /home/openvnmrj_2.1_A and the /vnmr link and re-install with the ver81.zip file. Instructions are in the Install.md document. After that, unzip the attachment and copy
 setacq to /vnmr/bin/setacq. Make sure the permission mode is set with
  chmod +x /vnmr/bin/setacq
It should then work.

[setacq.zip](https://github.com/OpenVnmrJ/OpenVnmrJ/files/5696601/setacq.zip)

matwillans, 2020-12-15:
Hi Dan,
I did install using the ver81.zip file.  Regardless I removed the /home/openvnmrj_2.1_A directory and the /vnmr link, then re-installed as you described.  Using the modified setacq, the setacq process appeared to complete without error.  However, even with the setacq working correctly, I am unable to establish communication with the console.

DanIverson, 2020-12-15:
What is failing? Did the download of software to the console complete? Are the "procs" running (su acqproc)?

matwillans, 2020-12-15:
Yes, the download of the software to the console completed.  And yes, the procs are running.  But the console status is still "inactive".

DanIverson, 2020-12-15:
What to you get if you run
  ps -ef | grep proc
  ps -ef | grep Info
from a Linux terminal. Also, the command
  hostname
will display the host name of your PC. Is that name, along with its IP address in the /etc/hosts file?
Does
  ping `hostname`
work?

matwillans, 2020-12-15:
Output from the ...proc and ...Info is attached.
The correct IP address and host name are in the /etc/hosts file.  When I ping the hostname, I do get a active response.  

Output from the ...proc command:
root        1522    1521  0 14:55 ?        00:00:00 ./Expproc
chemnmr     1717    1703  0 14:55 ?        00:00:00 postgres: stats buffer process
chemnmr     1718    1717  0 14:55 ?        00:00:00 postgres: stats collector process
vnmr1       6540    6504  0 14:56 ?        00:00:00 /usr/libexec/evolution-calendar-factory-subprocess --factory all --bus-name org.gnome.evolution.dataserver.Subprocess.Backend.Calendarx6504x2 --own-path /org/gnome/evolution/dataserver/Subprocess/Backend/Calendar/6504/2
vnmr1       6660    6593  0 14:56 ?        00:00:00 /usr/libexec/evolution-addressbook-factory-subprocess --factory all --bus-name org.gnome.evolution.dataserver.Subprocess.Backend.AddressBookx6593x2 --own-path /org/gnome/evolution/dataserver/Subprocess/Backend/AddressBook/6593/2
vnmr1       7223    7173  0 14:59 pts/0    00:00:00 grep --color=auto proc

Output from the ...Info command:
vnmr1       7267    7173  0 14:59 pts/0    00:00:00 grep --color=auto Info

DanIverson, 2020-12-16:
Could you do another diagnostic? From a Linux terminal as root, enter the command
  tcpdump -i enp2s0 -w logfile
Then reboot the Inova console, wait about 5 minutes and enter a cntrl-c to kill the tcpdump command. Send me the logfile

DanIverson, 2021-09-24:
Assuming this is resolved. Re-open if it is not resolved.

closed 2021-09-24 by DanIverson