OpenVnmrJ/OpenVnmrJ #486: Installer fails with noPing

clintwb, 2021-02-17:
Running ./load.nmr noPing fails with the following output:

xhost:  must be on local machine to enable or disable access control.
Checking for Linux packages required by OpenVnmrJ
Package update may take up to 40 minutes to complete...........
You can monitor the progress in a separate terminal window with the command
tail -f /tmp/ovj_preinstall/pkgInstall.log_2021-02-17_07_26_56
Installing standard packages (1 of 3)
Installing required packages (2 of 3)
Installing additional packages (3 of 3)
Removed /run/systemd/system/packagekit.service.
CentOS / RedHat package installation complete
 
\nUser "noPing" does not exist,
use Linux admintool to create user and rerun ./load.nmr
\naborting this program.....

To reproduce, run load.nmr noPing

According to the docs, it should install without trying to ping google.

Attempting this with the OpenVnmrJ_2.1_DDR.zip version with the ver81.zip applied.

The system is RHEL8.3 on a private/unroutable subnet using an http/https proxy.  Pings outside won't work, but http/https connections will.  Commenting the if/then block for the noPing check in ovjGetRepo allowed the install to proceed.

DanIverson, 2021-02-17:
Thanks for the information. I have fixed this and will check in the fix in a day or two. The other option is to use
   ./load.nmr skippkgchk

DanIverson, 2021-02-18:
Commit #489 fixes this problem.

closed 2021-02-18 by DanIverson