OpenVnmrJ/OpenVnmrJ #509: big sur has read-only filesystem mounted at /

tesch1, 2021-03-26:
**Describe the bug**

At least from macos Big Sur onward, the root fs is strongly protected - this causes the symbolic link from /vnmr/ -> .OpenVnmrJ..app/.../vnmr to fail during installation.

**To Reproduce**
Steps to reproduce the behavior:
- Install OpenVnmrJ.pkg on osx > 11.2.x
- notice that /vmnr/ link is missing

**Workarounds**

There are some fancy things that are bad for the security of your system, but appear to be a possible work around for this issue - I have not tried myself - https://developer.apple.com/forums/thread/649832

**Better fix**
The installation dir of vnmrj could be made directory-independent, perhaps pulling it from an environment variable.

DanIverson, 2021-03-26:
Version 2.1 of OVJ for Mac has the synthetic.conf fix. see src/macos/postinstall. I see the problem. Apple changed the numbering scheme for Big Sur. Catalina is 10.15.x  Big Sur 11.0.x I check that the second number is >= 15. Need to change to either first number > 10 or second number >= 15.

DanIverson, 2021-03-29:
This should be fixed with PR #510

closed 2021-03-29 by DanIverson