OpenVnmrJ/OpenVnmrJ #514: Temperature Controller in OpenVnmrJ?

groenoord, 2021-04-12:
We have an old Varian L950 temperature controller for our Varian/CMX solid-state NMR probes. We also have a Varian/CMX spinning speed controller.

Is it possible to use OpenVnmrJ to connect to the controllers if they are connected to serial ports on the computer that OpenVnmrJ is running on?

DanIverson, 2021-04-12:
If they worked with VnmrJ 4.2, they should work the same way with OpenVnmrJ. The speed controller connects to a serial port on the host computer. The temperature controller connects to either the acquisition CPU or the MSR board.

groenoord, 2021-04-12:
Thanks, Dan. Our temperature controller was originally also directly connected to the serial port of a Sun workstation running the CMX temperature controller software (we don't actually have a Varian console). Is there a way to replicate this in OpenVnmrJ or is the source of the CMX software available that could access the temperature controller directly?

DanIverson, 2021-04-12:
Joe DiVerdi has the source code for the CMX system. Kevin Geohring (blueskynmr.com) may also be able to help.

NeutralKaon, 2021-04-13:
Hi all -- if you have a Bruker one lying around, I have written an open-source and linux-friendly reimplementation of its control software that you can easily use on a Varian scanner (like I do...) -- c.f. https://github.com/NeutralKaon/BVTserialInterfacer

DanIverson, 2021-12-09:
Assuming this resolved. Reopen if needed.

closed 2021-12-09 by DanIverson