OpenVnmrJ/OpenVnmrJ #553: System is Inactive with new computer and software

prent1s, 2021-07-14:
Our computer for our Varian Unity Inova 500 died and so we decided to upgrade to the latest OpenVnmrJ 2.1A and use Centos 8 for our operating system. My IT set up the Operating system and installed the software (all they really know how to do). When the OpenVNMRj is showing "Inactive" in the status bar and I'm getting "Acquisition system is not active". What should I do next?

DanIverson, 2021-07-14:
Support for CentOS 8 will end at the end of this year. You might want to have your IT folks install CentOS 7 instead. Support for that will continue until June, 2024.

prent1s, 2021-07-22:
Dan,

Thanks for that update, but I guess I didn’t realize that we are using Centos 8 Streaming not just Centos 8. Any idea why the software is giving me “Inactive”?

Scott P.

Scott Prentice
Instrument Technician/NMR
Central Michigan University – Chemistry
Dow Science Building
201 East Ottawa Court
Room 342
Mount Pleasant, MI 48859
989-774-1455

DanIverson, 2021-07-22:
I have never installed OVJ on CentOS 8 Streaming. From what I can tell, it is a perpetual "beta-test" version of RHEL. I have gotten OVJ to work on RHEL 8.3 and CentOS 8.3, but OVJ 2.1 does not work on those versions "out-of-the-box".  I see that you are on SpinSights. If you enter tftp in the search box, you will see the thread "OVJ installation problem: TFTP and console won't boot". That thread discusses the problem and has the new files you will need (setacq, bootpd.service, and installpkgs). You will also need the ver81.zip file.  Another option is to install CentOS 7. OVJ 2.1 will then work "out-of-the-box"

NeutralKaon, 2021-07-22:
HI all -- out of curiosity and with huge apologies for the spam; is it still the case that CentOS 7 is the only supported OS for spectrometers? What about debian based systems – is it the tftpd configuration that is the origin of the problem? I know CentOS 7 is supported for a good while yet but as RHEL has killed it I'd quite like to try to migrate to something a bit longer lived than 2024 when we do something similar… 

Dan – also – thank you *so* much for your hard work on OpenVnmrJ. It's delightful to see. When we (Oxford) upgrade to it from Vnmrj 4.2A we have a set of private agilent-supplied patches to do things like increase the number of rtvars on imaging consoles and I'd like to try to help upstream them (when I get some time…) if you don't have them already. These were given only to Paul Kinchesch, as far as I know. 

All the best, 

-- Jack

DanIverson, 2021-07-23:
Hi Jack,

Nice to hear from you and thanks for the kind words. I think I still have everything I sent to Paul.

The current OVJ 2.1 only supports CentOS. However, I have gotten Inova, Mercury, and DDR systems to work with Ubuntu 20 and RHEL 8.3. These will be in the next release of OVJ. I intend to start a beta test of the new release very soon.

DanIverson, 2021-09-24:
Assuming this is resolved. Re-open if needed.

closed 2021-09-24 by DanIverson