OpenVnmrJ/OpenVnmrJ #619: setacq is not reflashing controllers

ronkerkhof, 2021-09-14:
I have installed CentOS 7 with openvnmrj 2.1 on VNMRS500
The system works fine with the software.
When I install an old controller on the console I am not able to have the new firmware written. Although there is no error message running /vnmr/setacq.
I also tried /vnmr/adm/acq/verifyControllers and got the following error message:
Failed to copy '/vnmr/acq/download/filename' to Controller
I have checked the files and they do exist and owned by vnmr1
Also rsh is enable and hosts.equiv is filled with controller names. Also tftp is enabled

DanIverson, 2021-09-14:
I see that you are on SpinSights. If you search for boot.ini, you will find a document titled "Reflashing script - a tool to manually set firmware on DD consoles".  There is a "Controller_Reflashing.doc" that may help.

ronkerkhof, 2021-09-14:
Hello Dan,This is also not working. It looks like the tftp service is not working as it works on centos6.I have followed this document. I can delete files but i cannot copy the files. I think itis not an openvnmrj problem but more related to centos 7. Is setacq using tftpboot or rsh?There is no error messages running setsacq. When i have an old controller in the system i can see setacq is not updating the files. When i rlogin the controller and running setacq i see error messages like the files are not uploading.Running from centos6 with vnmrj 4  it is working wellRegards,Ron KerkhofVerzonden vanaf Samsung-tablet.

"target": "https://github.com/OpenVnmrJ/OpenVnmrJ/issues/619#issuecomment-919191677",
"url": "https://github.com/OpenVnmrJ/OpenVnmrJ/issues/619#issuecomment-919191677",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"name": "GitHub",
"url": "https://github.com"
}
}
]

DanIverson, 2021-09-14:
Try running, as root,
  systemctl enable tftp.service
  systemctl start tftp.service

DanIverson, 2021-11-19:
Assuming this is fixed. Reopen if still a problem

closed 2021-11-19 by DanIverson