OpenVnmrJ/OpenVnmrJ #649: Error in version checking in some macros

srburt, 2021-11-09:
**Describe the bug**
several macros check the version of vnmrj using substr(ver,2):$ver but this fails with open VnmrJ because the version string contains an extra word before the version number so this check produces an error as substr tries to put a string into a variable initialized as real.

I ran across this with an older macro of mine that used piv and pivn, but it appears to affect the following macros:
dpiv, dpivn, piv, pivn, svfj, svllj, svlsj, svsj, svxyj

**To Reproduce**
open a Proton spectrum and type dpiv

**Expected behavior**
It should display the integral values vertically.  It does this, but also throws an error "expected REAL" 

Note: in the specific case of piv and dpiv, the integral values themselves are bad (they don't match the scaling value), but this seems to be unrelated to this specific error.

**Version of OpenVnmrJ you are using.**
OpenVnmrJ VERSION 2.1 REVISION A
January 22, 2020
propulse

Downloaded from GitHub and installed on RedHat 
I also see this in the version on the OpenVnmrJ CentOS image that I run in VMWare on my Mac.

**Environment (please complete the following information):**
 - OS: CentOS 5 and RedHat 7.8

**Additional context**
Note: I re-wrote my script to use dpir and integralOrient instead of dpiv

DanIverson, 2021-11-09:
These are both fixed in OVJ 3.1. For dpiv, replace lines 167 - 179
 
  substr(rev,1):$rev1
  format($rev1,'lower'):$rev1
  if $rev1='vnmrj' then
    $rev=6.1
  else
    $rev=0
    substr(rev,2):$rev
  endif
  if $rev<5.3 then      " old integration mode, based on is / ins "
    $oldint=1
  else
    $oldint=0
  endif

with the line

  $oldint=0

This will also fix the bad integral values.
For svfj, replace lines 258 - 266

if $token1='VNMRJ' then
  $sw='VnmrJ'
  substr(rev,3):$rev
  substr(rev,5):$subrev
else
  $sw='VNMR'
  substr(rev,2):$rev
  substr(rev,4):$subrev
endif

with the lines

  $sw='OpenVnmrJ'
  substr(rev,3):$rev
  substr(rev,5):$subrev

DanIverson, 2021-11-17:
Assuming this resolved the problem. Reopen if it is still a problem.

closed 2021-11-17 by DanIverson