OpenVnmrJ/OpenVnmrJ #655: OVJ 3.1 Installation Error on RHEL 7.9

srburt, 2021-12-02:
**Describe the bug**
I was attempting to install OVJ 3.1 on my data processing computers.  When I run load.nmr, I get an error saying that java couldn't find LoadNmr

**To Reproduce**
Steps to reproduce the behavior:
1. as vnmr1, unzip the DDR file downloaded from github
2. run ./load.nmr
3. After 3 packages were updated I get the error when it tries to use java to run the installation step (line 494 of load.nmr)

**Expected behavior**
The installation menu should open at this point.

**Version of OpenVnmrJ you are using.**
OpenVnmrJ 3.1 downloaded from GitHub on Nov 29th, 2021.

**Messages and Logs**
Starting the OpenVnmrJ installation program...
Exception in thread "main" java.lang.UnsupportedClassVersionError: LoadNmr : Unsupported major.minor version 52.0
	at java.lang.ClassLoader.defineClass1(Native Method)
	at java.lang.ClassLoader.defineClassCond(ClassLoader.java:631)
	at java.lang.ClassLoader.defineClass(ClassLoader.java:615)
	at java.security.SecureClassLoader.defineClass(SecureClassLoader.java:141)
	at java.net.URLClassLoader.defineClass(URLClassLoader.java:283)
	at java.net.URLClassLoader.access$000(URLClassLoader.java:58)
	at java.net.URLClassLoader$1.run(URLClassLoader.java:197)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.net.URLClassLoader.findClass(URLClassLoader.java:190)
	at java.lang.ClassLoader.loadClass(ClassLoader.java:306)
	at sun.misc.Launcher$AppClassLoader.loadClass(Launcher.java:301)
	at java.lang.ClassLoader.loadClass(ClassLoader.java:247)
Could not find the main class: LoadNmr.  Program will exit.

**Environment (please complete the following information):**
 - OS: RHEL 7.9
hostnamectl gives me this info in case it's useful:
CPE OS Name: cpe:/o:redhat:enterprise_linux:7.9:GA:server
Kernel: Linux 3.10.0-1160.42.2.el7.x86_64
Architecture: x86-64

DanIverson, 2021-12-02:
I think this is caused by using an older version of java. OVJ 3 requires java 1.8 or newer. Can you run
  java -version
and confirm it is 1.8 or newer

srburt, 2021-12-02:
"which java" shows that I'm using /vnmr/jre/bin/java
"java -version" says that's 1.6.0_39

	Comparing the older version of load.nmr in OVJ 2.1 against the newer version in OVJ 3.1, the old version would force the use of java included in the dvdimageOVJ while the newer load.nmr is using whatever java is in the current path variable (but, the newer dvdimageOVJ doesn't include a standalone java like the 2.1 version did).

	Can I just use yum to update the java version?

-Scott

---
Dr. Scott Burt
Teaching Professor
NMR Facility Manager
Department of Chemistry and Biochemistry
Brigham Young University
C414 BNSN / C008A BNSN
Provo, Utah  84602-5700
Phone:  (801) 422-2404
Fax: (801) 422-0153
chembio.byu.edu/nmr-facility <https://chembio.byu.edu/nmr-facility>
email: ***@***.***

DanIverson, 2021-12-02:
The load.nmr in OVJ3.1 should ignore the jre from the previous installation of OVJ2.1. (See lines 479 - 482 in load.nmr)  I am not sure why removing /vnmr/jre/bin from the PATH variable is not working.  CentOS 7.9 already has java 1.8 installed so we just need load.nmr to not use /vnmr/jre from OVJ2. One option would be to do the following
  cd /vnmr
  mv jre jre_
and then run load.nmr

srburt, 2021-12-02:
That worked (as did changing the line in load.nmr to /bin/java or /usr/bin/java).  

	Note: I got an "error" that I've seen before.  After installation, it says it's loading files for the users and stalls at 95%.  There's no update to the screen after 30+ minutes (i.e., no extra dots are added), so I closed the window and then continued on with the installation (questions about admin, nmrpipe, spectrometer, etc.).  So no specific error is given, but there's definitely something causing a problem with the locator db (or whatever is being processed during that final 5% of installation).

-Scott

---
Dr. Scott Burt
Teaching Professor
NMR Facility Manager
Department of Chemistry and Biochemistry
Brigham Young University
C414 BNSN / C008A BNSN
Provo, Utah  84602-5700
Phone:  (801) 422-2404
Fax: (801) 422-0153
chembio.byu.edu/nmr-facility <https://chembio.byu.edu/nmr-facility>
email: ***@***.***

DanIverson, 2021-12-02:
The last 5% is doing the locator db. If you run 
  dbsetup
does it finish. A lot of people just turn the locator off. That can be done from "vnmrj admin" in the Configure->Locator Config... popup. Also, if you do
  echo $PATH
what does it show?

srburt, 2021-12-02:
Related to my previous email on the IST thread, the data processing computer that's running RHEL 7.7 successfully completed the last bit of installation and that's also the one that is running NMRpip correctly for IST.

	The two computers running RHEL 7.9 that both failed to finish the dbset are the ones that can't finish dbsetup (even when run from the command line).

The path on the computer that _is_ running dbsetup correctly is
/vnmr/pgsql/bin:/opt/topspin4.1.3/prog/bin/scripts:/usr/local/bin:/usr/bin:/usr/local/sbin:/usr/sbin:/home/vnmr1/.local/bin:/home/vnmr1/bin:/vnmr/bin

The path on the computers that are _not_ running dbsetup correctly is
/vnmr/pgsql/bin:/opt/topspin4.1.3/prog/bin/scripts:/usr/local/bin:/usr/bin:/usr/local/sbin:/usr/sbin:/vnmr/bin:/vnmr/pgsql/bin:/home/vnmr1/.local/bin:/home/vnmr1/bin:/vnmr/bin

The only difference appears to be the order of /vnmr/bin vs /home/vnmr1/.local/bin and /home/vnmr1/bin

-Scott

DanIverson, 2021-12-02:
On the ones that don't work, what does
  postgres --version
show. If it is version 10 or higher, that is the problem.

srburt, 2021-12-02:
I'm seeing a postgres version of 9.2.24

-Scott