OpenVnmrJ/OpenVnmrJ #656: OVJ3.1 for Mac - can't use study queue to open data

srburt, 2021-12-02:
**Describe the bug**
After opening a study, you can't open individual spectra from the study queue.  For example, double-clicking the spectrum name or dragging it to a viewport or right-clicking and selecting "open experiment" – all these simply do nothing.  You have to use the File-Open...  menu option to open spectra in each viewport.

**To Reproduce**
Steps to reproduce the behavior:
1. Open OVJ 3.1
2. Use File-Open... to open a study (you can see the individual spectra populate the study queue if 'sample' is enabled)
3. Try to double click a spectrum or drag and drop it into a viewport or right-click on the spectrum name and choose "open experiment"
4. Nothing happens

**Expected behavior**
Normally, these actions would open the spectrum in the active viewport (or, if drag-and-drop, in the viewport that it was dragged to)

**Version of OpenVnmrJ you are using.**
OpenVnmrJ 3.1
downloaded from github on Nov 29th, 2021

**Messages and Logs**
No errors, it simply doesn't do anything

**Environment (please complete the following information):**
 - OS: OSX 10.15.7

**Additional context**
General processing appears to work fine aside from this.

DanIverson, 2021-12-02:
I cannot duplicate this behaviour. Can you send the ~/vnmrsys/VnmrjMsgLog file? It may contain some clues.

srburt, 2021-12-02:
Here you go.

-Scott

---
Dr. Scott Burt
Teaching Professor
NMR Facility Manager
Department of Chemistry and Biochemistry
Brigham Young University
C414 BNSN / C008A BNSN
Provo, Utah  84602-5700
Phone:  (801) 422-2404
Fax: (801) 422-0153
chembio.byu.edu/nmr-facility <https://chembio.byu.edu/nmr-facility>
email: ***@***.***

DanIverson, 2021-12-02:
No file was attached.

srburt, 2021-12-02:
Hmmm... I could see a file, but maybe it didn't like that it had no extension.  I'll make it a tgz to see if that helps.

-Scott

---
Dr. Scott Burt
Teaching Professor
NMR Facility Manager
Department of Chemistry and Biochemistry
Brigham Young University
C414 BNSN / C008A BNSN
Provo, Utah  84602-5700
Phone:  (801) 422-2404
Fax: (801) 422-0153
chembio.byu.edu/nmr-facility <https://chembio.byu.edu/nmr-facility>
email: ***@***.***

srburt, 2021-12-03:
Dan,

	I tried debug tracing for this OSX study queue issue.  After debug('c3') I opened a data set and then tried double clicking several spectra followed by right-clicking and choosing open followed by dragging a protocol to the viewport.

-Scott

---
Dr. Scott Burt
Teaching Professor
NMR Facility Manager
Department of Chemistry and Biochemistry
Brigham Young University
C414 BNSN / C008A BNSN
Provo, Utah  84602-5700
Phone:  (801) 422-2404
Fax: (801) 422-0153
chembio.byu.edu/nmr-facility <https://chembio.byu.edu/nmr-facility>
email: ***@***.***

DanIverson, 2021-12-06:
Can you send me the debug file at daniverson2016@gmail.com