OpenVnmrJ/OpenVnmrJ #657: Can't Process NUS Data with IST

srburt, 2021-12-02:
**Describe the bug**
Data can't be processed using the IST method.

**To Reproduce**
Steps to reproduce the behavior:
1. Open a data set collected with NUS enabled
2. Under Process-Default click "restore raw data"
3. Under "use advanced options" change the method to "IST" and click "process"
4. See error: pread: cannot access /Users/srburt/vnmrsys/exp1/pipe/spec_ist.ft2

**Expected behavior**
The data should be processed using IST 
(e.g., data with low SNR or big differences in peak height are better handled by IST than CLEAN - I have some ADEQUATE spectra I collected with NUS and CLEAN ends up losing most of the data and setting the vs to ~10^9 so it's impossible to access any peaks weaker than the methyls)

**Version of OpenVnmrJ you are using.**
OpenVnmrJ 3.1 for Mac, downloaded from GitHub on Nov 29th, 2021

**Messages and Logs**
The debug tracing didn't generate any log data, but it did give a message on the command line that wasn't there before:

Error macro: NUSprocIST, line 216

**Environment (please complete the following information):**
 - OS: OSX 10.15.7
 - I am also seeing this error in OVJ 2.1 in RHEL 6.10 and 7.9 (I was hoping upgrading to OVJ 3.1 would solve the IST issue)

**Additional context**
I'm testing things related to NUS so I can (finally) get this setup as the default approach in my facility.  Clean seems to work fine for HSQC and COSY, but the quality is quite very poor for HMBC and ADEQUATE, so I'd like to try using IST as (based on what I've read) it's supposed to better handle low SNR data and data with bigger ranges of peak heights.

DanIverson, 2021-12-02:
Can you send a dataset that fails?

srburt, 2021-12-02:
Note: here's some additional info I've collected in the past 15 minutes or so:

	I've now installed OVJ 3.1 on three of my data processing computers over the past hour or so.  One of them, when I replied 'y' to installing NMRpipe showed that it was downloading 7 files whereas the other two didn't show any downloading or installation despite me answering 'y' to the same question.  The computer that downloaded and installed the NMRpipe files _is_ correctly running IST now whereas the other two are not.  So, this looks like a problem with how NMRpipe is installed on the machine.  I don't know if this correlation implies causation, but the computer that _is_ working is running RHEL 7.7 and the two that are _not_ working are running RHEL 7.9.

	For those two that didn't download and install NMRpipe, I manually ran /vnmr/bin/ovjGetpipe and I could see it downloading the 7 files and installing them (presumably successfully, there weren't any errors that I could see).  However, I still get the pread error when attempting to do IST and I do see (briefly) an error about vj2pipe.tcl (I think?) in the nmrpipe dialogue that pops up, but it disappears rather quickly.  If there's a log file somewhere that I can send you about this OVJ-NMRPipe communication, I'd be happy to send it to you.

	Attached is a link to a tgz of a studies with NUS data that I've been testing:
https://drive.google.com/file/d/1ydFpHS8iLeeAQt2MSE3wgytX9ERu58hb/view?usp=sharing <https://drive.google.com/file/d/1ydFpHS8iLeeAQt2MSE3wgytX9ERu58hb/view?usp=sharing> 
(The 2D spectra are collected in sets of 3 with the first being a normal spectrum and the next two being NUS of varying amounts.  For example, I've been using gHSQCAD_03 and gHMBCAD_03 as test cases with NUS of 25% and 50% respectively)

-Scott

---
Dr. Scott Burt
Teaching Professor
NMR Facility Manager
Department of Chemistry and Biochemistry
Brigham Young University
C414 BNSN / C008A BNSN
Provo, Utah  84602-5700
Phone:  (801) 422-2404
Fax: (801) 422-0153
chembio.byu.edu/nmr-facility <https://chembio.byu.edu/nmr-facility>
email: ***@***.***

DanIverson, 2021-12-03:
I installed OVJ3 and NMRPipe on CentOS 7.9 and your data sets processed correctly. Can I access your system with AnyDesk? Maybe tomorrow (Friday)

srburt, 2021-12-03:
Yeah, it's definitely looking like a configuration problem.  For another data point, I'm seeing the same behavior on my Mac – I was able to run /vnmr/bin/ovjGetpipe successfully, but IST processing gives the same pread error.

	My NMR computers are behind a firewall; will AnyDesk be able to work in that situation?  I can work on getting that installed tonight or tomorrow morning.  I'm available 9-10 and 12-3 mountain time tomorrow.

-Scott

DanIverson, 2021-12-06:
From a terminal, can you run the following. This command assumes vnmr1 is running and the data is in exp1

/vnmr/bin/vnmrpipe vj2pipe.com    -nograph  -clean -cd /home/vnmr1/vnmrsys/exp1 -vjfid /home/vnmr1/vnmrsys/exp1/acqfil/fid     -xVJSB -0.0750 -xVJSBS 0.0000 -istTMult 0.7 -istCMult 0.3 -istIter 1024 -istMaxRes 1  -pipeDir /home/vnmr1/vnmrsys/exp1/pipe  -procpar /home/vnmr1/vnmrsys/exp1/curpar -xZFARG size=2048   -yZFARG size=102

srburt, 2021-12-06:
Looks like a TCL/TK issue.

	On the data station that _is_ able to run IST, that command worked fine from the command line.

	On one of the data stations that is _not_ able to run IST, I get the following output:

vj2pipe Changing Working Directory:
 /home/vnmr1/vnmrsys/exp1

vj2pipe Data and Working Directory Status:
 /home/vnmr1/vnmrsys/exp1

vj2pipe Creating Directory:
 /home/vnmr1/vnmrsys/exp1/pipe

vj2pipe Changing Working Subdirectory:
 /home/vnmr1/vnmrsys/exp1/pipe

vj2pipe Working Directory Status:
 /home/vnmr1/vnmrsys/exp1/pipe
vj2pipe Setting procpar File Name:
 /home/vnmr1/vnmrsys/exp1/curpar

vj2pipe Directory Status:
 /home/vnmr1/vnmrsys/exp1/pipe

Skipping Directory . ... No timestamp.vj2pipe file found.
nmrWish: error while loading shared libraries: libtk8.4.so: wrong ELF class: ELFCLASS32
vj2pipe Error while executing vj2pipe.tcl ... Exiting.

Note: There is no libtk8.4.so in /lib, /usr/lib, or /usr/local/lib; nor is it in /lib64, /usr/lib64, or /usr/local/lib64 (but there is a libtk8.5.so in /lib64)

I can only find this library in /vnmr/lib and /vnmr is pointing to /vnmr/home/openvnmrj_3.1_A

-Scott

---
Dr. Scott Burt
Teaching Professor
NMR Facility Manager
Department of Chemistry and Biochemistry
Brigham Young University
C414 BNSN / C008A BNSN
Provo, Utah  84602-5700
Phone:  (801) 422-2404
Fax: (801) 422-0153
chembio.byu.edu/nmr-facility <https://chembio.byu.edu/nmr-facility>
email: ***@***.***

DanIverson, 2021-12-06:
pipe installs its own version of the libraries. They are in /vnmr/nmrpipe/nmrbin.linux212_64/lib
Is it possible you have an old version of pipe installed somewhere?

srburt, 2021-12-07:
I'm not aware of any previous NMRpipe installations.  I looked around and couldn't find any obvious evidence of an NMRpipe installation.  There's nothing in the environment variables related to NMRpipe.

	I compared a few things on a machine that isn't working vs. the one that is working:
		The error was the elf class of /vnmr/libtk8.4.so but "readelf -h /vnmr/libtk8.4.so" shows both to be ELF32 
		Both computers are running TCL/TK 8.5.13 (determined by running tclsh and then "info patchlevel")
		The environment variables are the same as far as I can tell.

-Scott

---
Dr. Scott Burt
Teaching Professor
NMR Facility Manager
Department of Chemistry and Biochemistry
Brigham Young University
C414 BNSN / C008A BNSN
Provo, Utah  84602-5700
Phone:  (801) 422-2404
Fax: (801) 422-0153
chembio.byu.edu/nmr-facility <https://chembio.byu.edu/nmr-facility>
email: ***@***.***

DanIverson, 2021-12-07:
On the Mac, there should not be any  /vnmr/lib/libtk8.4 libraries. What happens if you run the above command on the Mac? 
Also on the Mac, from the OVJ command line, what does 
  pwd
give. If the path has a space in it, could you cd to a directory without a space (eg, cd('/vnmr') ) and then see if IST works.

srburt, 2021-12-07:
Excellent question.  I meant to test that command you sent on the Mac and got distracted by some of the other things I was looking at.

1) Within OVJ on the Mac, "pwd" returns "/Users/srburt"

2) Modifying the command you sent so it's pointing to /Users/srburt/vnmrsy/exp1 I have

/vnmr/bin/vnmrpipe vj2pipe.com    -nograph  -clean -cd /Users/srburt/vnmrsys/exp1 -vjfid /Users/srburt/vnmrsys/exp1/acqfil/fid     -xVJSB -0.0750 -xVJSBS 0.0000 -istTMult 0.7 -istCMult 0.3 -istIter 1024 -istMaxRes 1  -pipeDir /Users/srburt/vnmrsys/exp1/pipe  -procpar /Users/srburt/vnmrsys/exp1/curpar -xZFARG size=2048   -yZFARG size=102

	When run from a terminal, it works just fine!  I even get the IST convergence progress bar that pops up on my screen.  So that looks hopeful.

	Note, however, if I "restore raw data" and then "use advanced options" to choose "IST" I just get the usual pread error

(VnmrjMsgLog)
 35 ## INFO  Dec 7, 2021 14:13:57 ********** Starting OpenVnmrJ as srburt **********
 36 ## INFO  Dec 7, 2021 14:13:57 ** OpenVnmrJ VERSION 3.1 REVISION A **
 37 ## INFO  Dec 7, 2021 14:13:57 ** October 06, 2021 **
 38 ## INFO  Dec 7, 2021 14:13:57 ** vnmrs **
 39 ## INFO  Dec 7, 2021 14:13:57 **********************************************
 40 ## DEBUG Dec 7, 2021 14:14:04 ### appDir 0 /Users/srburt/vnmrsys
 41 ## DEBUG Dec 7, 2021 14:14:04 ### appDir 1 /Users/srburt/vnmrsys/procpack
 42 ## DEBUG Dec 7, 2021 14:14:04 ### appDir 2 /Users/srburt/vnmrsys/byupack
 43 ## DEBUG Dec 7, 2021 14:14:04 ### appDir 3 /vnmr
 44 ## DEBUG Dec 7, 2021 14:14:04 ### appDir 4 /vnmr/openvnmrj-utils
 45 ## ERROR Dec 7, 2021 14:16:21 pread: cannot access /Users/srburt/vnmrsys/exp1/pipe/spec_ist.ft2

	Note: before, when trying to run IST from OVJ, there wasn't a pipe subdir within exp1 at all.  Now, after running that vj2pipe from the command line, I can see that ~/vnmrsys/pipe/spec_ist.ft2 exists, but the pread error still complains that it can't access that file.  There doesn't appear to be any weird permissions issues:

-rw-r--r--  1 srburt  staff   837632 Dec  7 14:19 spec_ist.ft2

	Any other useful logs I can pass along?

-Scott

srburt, 2021-12-15:
I finally have AnyDesk installed on my three data processing computers.  Two of which have trouble with IST and one of which works with IST.

	I've not used AnyDesk before, but I've used VNC a lot.  Let me know what I need to do to get a session open that allows you to connect and what information I need to pass on to you so you can find the sessions I've created.

-Scott

---
Dr. Scott Burt
Teaching Professor
NMR Facility Manager
Department of Chemistry and Biochemistry
Brigham Young University
C414 BNSN / C008A BNSN
Provo, Utah  84602-5700
Phone:  (801) 422-2404
Fax: (801) 422-0153
chembio.byu.edu/nmr-facility <https://chembio.byu.edu/nmr-facility>
email: ***@***.***

srburt, 2021-12-17:
Dan,

	Update on the IST and .vnmrenv stuff from yesterday:

	When I arrived at work today, a student was complaining that some samples he submitted on the NMRS 500 did run last night.  I tried the usual things (restarting the automation run, restarting OVJ, resetting the spectrometer, etc.) without success.  Any time a sample is submitted, it shows up as queued, but nothing ever starts.

	Looking through the log files, I found the first failed sample was shortly after the time I deleted .vnmrenv and tested IST processing.  I checked our Inova 300 MHz and found the same thing (samples will queue, but not start).

	As soon as I restored .vnmrenv (either by running vnmrj adm, update users; or by simply copying ~vnmr1/.vnmrenv over) the automation starts up (without even needing to restart OVJ).

	I then tested IST processing and it fails again now that .vnmrenv is restored.

	Summary:

1) Whatever is bad in .vnmrenv that prevents IST processing from working is in the template file that vnmrj adm -> update users copies into the user dir.

2) Removing .vnmrenv allows IST processing to work, but kills automation.

3) Restoring .vnmrenv gets automation working again, but kills IST processing.

-Scott

---
Dr. Scott Burt
Teaching Professor
NMR Facility Manager
Department of Chemistry and Biochemistry
Brigham Young University
C414 BNSN / C008A BNSN
Provo, Utah  84602-5700
Phone:  (801) 422-2404
Fax: (801) 422-0153
chembio.byu.edu/nmr-facility <https://chembio.byu.edu/nmr-facility>
email: ***@***.***

DanIverson, 2021-12-17:
I believe this is now fixed.

closed 2021-12-17 by DanIverson

srburt, 2021-12-17:
Yes, thank you!

-Scott

---
Dr. Scott Burt
Teaching Professor
NMR Facility Manager
Department of Chemistry and Biochemistry
Brigham Young University
C414 BNSN / C008A BNSN
Provo, Utah  84602-5700
Phone:  (801) 422-2404
Fax: (801) 422-0153
chembio.byu.edu/nmr-facility <https://chembio.byu.edu/nmr-facility>
email: ***@***.***