OpenVnmrJ/OpenVnmrJ #658: NUS Processing Ignores foldflg

srburt, 2021-12-02:
**Describe the bug**
When processing homonuclear spectra, NUS reconstruction ignores the value of foldflg and forces data symmetrization.  This produces false crosspeaks anywhere there is significant t1 noise ridges (e.g. methyls and other tall peaks).  I have all my homonuclear protocols customized to set foldflg='n' but NUS processing ignore this. 

**To Reproduce**
Steps to reproduce the behavior:
1. Setup a COSY and set foldflg='n' before running it (with NUS enabled)
2. Process the data and the result is symmetrized

**Expected behavior**
The NUS processing should obey foldflg so that those of us who detest symmetrization can ensure that symmetrization is not applied.

**Version of OpenVnmrJ you are using.**
OpenVnmrJ 2.1 and 3.1 (both downloaded from GitHub)
Both the DDR and OSX versions

**Messages and Logs**
No errors are generated, just undesired behavior.

**Environment (please complete the following information):**
RHEL 6.10, 7.9 and OSX 10.15.7

**Additional context**
I've been testing out NUS on TopSpin and homonuclear spectra are _not_ symmetrized in NUS reconstruction (e.g., you can see the t1 noise ridges and the obnoxious false crosspeaks aren't generated)

DanIverson, 2021-12-02:
Can you try the NUSproc macro in the attached zip file.  Copy it to ~/vnmrsys/maclib or /vnmr/maclib

[sb.zip](https://github.com/OpenVnmrJ/OpenVnmrJ/files/7645444/sb.zip)

srburt, 2021-12-02:
I have now tested this (in ~/vnmrsys/maclib) on both my Mac and on one of the data processing computers that is failing with IST (so OSC 10.15.7 and RHEL 7.9) and I get the same pread error (with a brief window that says something about an error with vj2pipe.tcl)

-Scott

---
Dr. Scott Burt
Teaching Professor
NMR Facility Manager
Department of Chemistry and Biochemistry
Brigham Young University
C414 BNSN / C008A BNSN
Provo, Utah  84602-5700
Phone:  (801) 422-2404
Fax: (801) 422-0153
chembio.byu.edu/nmr-facility <https://chembio.byu.edu/nmr-facility>
email: ***@***.***

DanIverson, 2021-12-02:
This macro won't fix the IST issue but should take care of the foldflg issue.

srburt, 2021-12-02:
Oh, got it.  I'll test that.

-Scott

srburt, 2021-12-03:
Yes, the NUSproc that you sent fixes the foldflg issue.  The COSY data look much better now.

-Scott

---
Dr. Scott Burt
Teaching Professor
NMR Facility Manager
Department of Chemistry and Biochemistry
Brigham Young University
C414 BNSN / C008A BNSN
Provo, Utah  84602-5700
Phone:  (801) 422-2404
Fax: (801) 422-0153
chembio.byu.edu/nmr-facility <https://chembio.byu.edu/nmr-facility>
email: ***@***.***

closed 2021-12-03 by DanIverson