OpenVnmrJ/OpenVnmrJ #659: Spectrometer inactive after installing OVJ 3.2 on Inova

srburt, 2021-12-03:
**Describe the bug**
I installed OVJ 3.2 on our Inova 300 MHz.  The installation went well (dbsetup worked fine) and "acqcomm status" claims the acquisition is running, but OVJ shows "inactive".  If I manually run /vnmr/bin/setacq I never see the file transfer blinking lights on the card after pressing the reset button and hitting enter.

**To Reproduce**
Steps to reproduce the behavior:
1. Run setacq, press the reset button and press enter on the keyboard
2. No lights blink on the card, the red light never switches to green
3. acqcomm status claims things are running
4. vnmrj shows "inactive" in the status window

**Expected behavior**
After running setacq, the lights should blink rapidly as files are transferred and then the red light should switch to green after which vnmrj will show active instead of inactive.

**Version of OpenVnmrJ you are using.**
OVJ 3.1 downloaded from github

**Messages and Logs**

**Environment (please complete the following information):**
 - OS: RHEL 6.10

**Additional context**
Note: I recognize that the OVJ website says RHEL 6 isn't officially supported for OVJ 3.1 so I understand if it's not worth investing time in this.  The instrument computer is 13 years old and will hopefully be replaced early next year (still waiting to hear back from our IT people...)

I don't think this is related, but when I first ran vnmrj I got an error from vnmrjbg that libstdc++.so.6 couldn't find some glibcxxx_3.4.something.  I was able to fix this by pointing the libstdc++.so.6 symlink to libstdc++.so.6.0.19 copied from my RHEL 7 boxes (the version on the RHEL 6 box was 6.0.13).

DanIverson, 2021-12-03:
Yes, the downloaded OVJ 3.1 does not support RHEL 6. However, if you build it yourself on the RHEL 6 PC, it should work. OVJ 3.1 includes the script /vnmr/bin/ovjBuild which will build the dvdimageOVJMI directory locally.

srburt, 2021-12-03:
I tried the OVJ3.1 ovjBuild and I see the following in the makeovjlog file:

1) Lots of warning when compiling C files (maybe those are OK, things like implicit type declaration or return with no value in a non-void function, etc.)

2) Several scons errors
	a handful of Error 1 (/lib64/libc.so.6 missing)
	and one Error 127 (gfortran not found for src/DOSY/contin.o)
	and a final error saying it ended due to errors

Note: the RHEL 6.10 boxes have gcc 4.4.7-23

-Scott

---
Dr. Scott Burt
Teaching Professor
NMR Facility Manager
Department of Chemistry and Biochemistry
Brigham Young University
C414 BNSN / C008A BNSN
Provo, Utah  84602-5700
Phone:  (801) 422-2404
Fax: (801) 422-0153
chembio.byu.edu/nmr-facility <https://chembio.byu.edu/nmr-facility>
email: ***@***.***

DanIverson, 2021-12-04:
I was able to build OVJ3.1 on CentOS 6.9. You can get the dvdimageOVJMI directory from the sb.zip file on dropbox. https://www.dropbox.com/s/ac70sbx2blqnage/sb.zip?dl=0

srburt, 2021-12-06:
That worked beautifully!  Thank you so much.

	Notes
		There were no setacq problems.  
		The updatedb portion of the installation worked fine.
		I just finished running some test samples and everything looks good.
		The nmrpipe portion of the installation went fine (7 files downloaded, etc.)  However, I get the same pread error when trying to use IST reconstruction.  Must be the same configuration error that I have on the two data processing computers.

	Would it be possible to get a similar dvdimage for my VNMRS/DDR console that's on RHEL 6.10?

	(I'm still waiting to hear back from my IT people regarding how to get you remote access through our firewall - I'll bug them again today).

-Scott

closed 2021-12-06 by DanIverson