OpenVnmrJ/OpenVnmrJ #660: Various errors after installing OVJ 3.1 on DD 500 MHz

srburt, 2021-12-03:
**Describe the bug**
OVJ 3.1 installed fine (no dbsetup error).  The automation initially appeared to work fine.  However, several problems were encountered that prevented normal usage:
1) adding HSQCAD to a queue and then trying to customize it created several errors and completely locked up vnmrj so I had to kill it from the command line. The error was something about tmp_ad_shape.rf not found and then an error about longQ not defined (even though my schedule doesn't use shortq or longq, it uses the dayQ/nightQ schedule with per-sample limits in the dayQ)
2) trying to process NUS data fails (the usual IST pread error, but now even CLEAN fails with an error about data reduction failed)
There's probably more, but those are the main ones that I encountered.
Falling back to OVJ2.1 (running setacq, etc.) resolved all these problems.

**To Reproduce**
Steps to reproduce the behavior:
1. Point /vnmr to /home/openvnmrj_3.1_A and run setacq and then config within vnmrj
2. Start a new automation file/automation run
3. Create a new study, add a proton and hsqcad (from the common tab)
4. Double click on HSQCAD to customize 
5. Several errors show up in the error dialogue (the one at the end, that you can see, is about longQ not being define; but opening the error message window shows that the root error is the temporary shaped pulse isn't found)
6. At this point, you can no longer use the "cancel" button to get of submit mode, you can't logout, you can close vnmrj, etc.

**Expected behavior**
You should be able to customize the protocol (e.g., ns and ni) without errors

**Version of OpenVnmrJ you are using.**
OVJ 3.2 from github

**Messages and Logs**
## ERROR Dec 2, 2021 21:29:45 pboxget: input file /home/nmruser/vnmrsys/shapelib/tmp_ad_shape.RF not found
## ERROR Dec 2, 2021 21:29:45 Pbox: power not calibrated!
## ERROR Dec 2, 2021 21:29:45 Variable "$pwr" is undefined!
## ERROR Dec 2, 2021 21:29:45 Error macro:  pbox_pwr,  line 23
## ERROR Dec 2, 2021 21:29:45 exec: 'Dch_adiabatic_module('acquisition','','')' aborted
## ERROR Dec 2, 2021 21:29:45 exec: 'hetero2D('acquisition','std1D')' aborted
## ERROR Dec 2, 2021 21:29:45 exec: 'HSQCAD' aborted
## ERROR Dec 2, 2021 21:29:54 Variable "longQ" is undefined!
## ERROR Dec 2, 2021 21:29:54 Error macro:  walkupQ_clear,  line 15

**Environment (please complete the following information):**
 - OS: RHEL 6.10 
Note: like my previous bug report, I realize that RHEL 6 isn't officially supported, so I understand if it's not worth the time to resolve this error. The computer is ~13 years old, so it should be replaced soon (but I've been saying that for 3 years now...)

**Additional context**
Falling back to OVJ 2.1 resolves all these issues.

Note: I don't know if it's related, but there was an error when I first tried to run vnmrj (vnmrjbg complained about libstdc++.so.6 missing a newer version of glibc; I copied libstdc++.so.6.0.19 from a RHEL 7 box and pointed the libstdc++.so.6 symlink to that instead of the 6.0.13 version that's on the RHEL 6 box.)

DanIverson, 2021-12-06:
Here is a link for DDR on RHEL 6.9 or newer. I gave the zip file a bit more descriptive name

https://www.dropbox.com/s/8entvtghuazfcui/ovj31_DDR_centos69.zip?dl=0

srburt, 2021-12-06:
That worked beautifully.  I've run a few test samples and it seems to be working fine.

-Scott

---
Dr. Scott Burt
Teaching Professor
NMR Facility Manager
Department of Chemistry and Biochemistry
Brigham Young University
C414 BNSN / C008A BNSN
Provo, Utah  84602-5700
Phone:  (801) 422-2404
Fax: (801) 422-0153
chembio.byu.edu/nmr-facility <https://chembio.byu.edu/nmr-facility>
email: ***@***.***

closed 2021-12-06 by DanIverson