OpenVnmrJ/OpenVnmrJ #676: Varian 400 MR connection crash upon any setup

thezfisher, 2022-02-16:
As of this week, every time I attempt to begin an experiment, whether doing the su command, or attempting a lock protocol, the communication stops, and then the acquisition system locks. The only way to re-establish communication is an acquisition system reset. I'm trying to figure out why it won't start any experiment, and what's causing this stuck acquisition state. I am not currently an NMR expert and i'm not the best at navigating this program, but nobody at my school has the experience to troubleshoot it effectively. Has anyone seen an issue like this, and if so where should I start in troubleshooting? 
Here are some of the recent error codes from the VNMRJ messages log:
## INFO  Feb 15, 2022 14:58:52 ********** Starting OpenVnmrJ Admin as vnmr1 **********
## INFO  Feb 15, 2022 14:58:52 ** OpenVnmrJ VERSION 2.1 REVISION A **
## INFO  Feb 15, 2022 14:58:52 ** January 22, 2020 **
## INFO  Feb 15, 2022 14:58:52 ** mr400 **
## INFO  Feb 15, 2022 14:58:52 **********************************************
## INFO  Feb 15, 2022 15:04:11 **********************************************
## INFO  Feb 15, 2022 15:04:11 ********** Starting OpenVnmrJ Admin as vnmr1 **********
## INFO  Feb 15, 2022 15:04:11 ** OpenVnmrJ VERSION 2.1 REVISION A **
## INFO  Feb 15, 2022 15:04:11 ** January 22, 2020 **
## INFO  Feb 15, 2022 15:04:11 ** mr400 **
## INFO  Feb 15, 2022 15:04:11 **********************************************
## INFO  Feb 16, 2022 10:55:21 **********************************************
## INFO  Feb 16, 2022 10:55:21 ********** Starting OpenVnmrJ as vnmr1 **********
## INFO  Feb 16, 2022 10:55:21 ** OpenVnmrJ VERSION 2.1 REVISION A **
## INFO  Feb 16, 2022 10:55:21 ** January 22, 2020 **
## INFO  Feb 16, 2022 10:55:21 ** mr400 **
## INFO  Feb 16, 2022 10:55:21 **********************************************
## DEBUG Feb 16, 2022 10:55:29 PROTON  Common
## DEBUG Feb 16, 2022 10:55:29 ### appDir 0 /home/vnmr1/vnmrsys
## DEBUG Feb 16, 2022 10:55:29 ### appDir 1 /vnmr
## DEBUG Feb 16, 2022 10:55:29 ### appDir 2 /vnmr/openvnmrj-utils
## INFO  Feb 16, 2022 11:01:10 **********************************************
## INFO  Feb 16, 2022 11:01:10 ********** Starting OpenVnmrJ Admin as vnmr1 **********
## INFO  Feb 16, 2022 11:01:10 ** OpenVnmrJ VERSION 2.1 REVISION A **
## INFO  Feb 16, 2022 11:01:10 ** January 22, 2020 **
## INFO  Feb 16, 2022 11:01:10 ** mr400 **
## INFO  Feb 16, 2022 11:01:10 **********************************************
## INFO  Feb 16, 2022 11:01:54 **********************************************
## INFO  Feb 16, 2022 11:01:54 ********** Starting OpenVnmrJ as vnmr1 **********
## INFO  Feb 16, 2022 11:01:54 ** OpenVnmrJ VERSION 2.1 REVISION A **
## INFO  Feb 16, 2022 11:01:54 ** January 22, 2020 **
## INFO  Feb 16, 2022 11:01:54 ** mr400 **
## INFO  Feb 16, 2022 11:01:54 **********************************************
## DEBUG Feb 16, 2022 11:02:02 PROTON  Common
## DEBUG Feb 16, 2022 11:02:02 ### appDir 0 /home/vnmr1/vnmrsys
## DEBUG Feb 16, 2022 11:02:02 ### appDir 1 /vnmr
## DEBUG Feb 16, 2022 11:02:02 ### appDir 2 /vnmr/openvnmrj-utils
## ERROR Feb 16, 2022 11:02:13 experiment /home/vnmr1/vnmrsys/exp2 is not accessible
## INFO  Feb 16, 2022 12:56:21 **********************************************
## INFO  Feb 16, 2022 12:56:21 ********** Starting OpenVnmrJ Admin as vnmr1 **********
## INFO  Feb 16, 2022 12:56:21 ** OpenVnmrJ VERSION 2.1 REVISION A **
## INFO  Feb 16, 2022 12:56:21 ** January 22, 2020 **
## INFO  Feb 16, 2022 12:56:21 ** mr400 **
## INFO  Feb 16, 2022 12:56:21 **********************************************
## DEBUG Feb 16, 2022 12:56:47 Running script: vnmr_open https://spinsights.chem.agilent.com
## DEBUG Feb 16, 2022 12:57:11 Running script: vnmr_open /vnmr/help/WebHelp/index.htm
## INFO  Feb 16, 2022 12:57:27 **********************************************
## INFO  Feb 16, 2022 12:57:27 ********** Starting OpenVnmrJ as vnmr1 **********
## INFO  Feb 16, 2022 12:57:27 ** OpenVnmrJ VERSION 2.1 REVISION A **
## INFO  Feb 16, 2022 12:57:27 ** January 22, 2020 **
## INFO  Feb 16, 2022 12:57:27 ** mr400 **
## INFO  Feb 16, 2022 12:57:27 **********************************************
## DEBUG Feb 16, 2022 12:57:36 PROTON  Common
## DEBUG Feb 16, 2022 12:57:37 ### appDir 0 /home/vnmr1/vnmrsys
## DEBUG Feb 16, 2022 12:57:37 ### appDir 1 /vnmr
## DEBUG Feb 16, 2022 12:57:37 ### appDir 2 /vnmr/openvnmrj-utils
## DEBUG Feb 16, 2022 13:02:32 ** Start DataBase Contents Report **
## DEBUG Feb 16, 2022 13:02:32 DB version: pg_ctl (PostgreSQL) 9.2.24  DB Host: localhost.localdomain
              Local Host: localhost.localdomain     Network Mode: false
## DEBUG Feb 16, 2022 13:02:32 Number of vnmr_data items in DB = 330
## DEBUG Feb 16, 2022 13:02:32 Number of vnmr_par items in DB = 313
## DEBUG Feb 16, 2022 13:02:32 Number of vnmr_record items in DB = 0
## DEBUG Feb 16, 2022 13:02:32 Number of vnmr_rec_data items in DB = 0
## DEBUG Feb 16, 2022 13:02:32 Number of automation items in DB = 21
## DEBUG Feb 16, 2022 13:02:32 Number of workspace items in DB = 15
## DEBUG Feb 16, 2022 13:02:32 Number of panels_n_components items in DB = 1214
## DEBUG Feb 16, 2022 13:02:32 Number of shims items in DB = 13
## DEBUG Feb 16, 2022 13:02:32 Number of command_n_macro items in DB = 99
## DEBUG Feb 16, 2022 13:02:32 Number of pulse_sequence_macro items in DB = 88
## DEBUG Feb 16, 2022 13:02:32 Number of protocol items in DB = 237
## DEBUG Feb 16, 2022 13:02:32 Number of study items in DB = 12
## DEBUG Feb 16, 2022 13:02:32 Number of lcstudy items in DB = 0
## DEBUG Feb 16, 2022 13:02:32 Number of image_dir items in DB = 0
## DEBUG Feb 16, 2022 13:02:32 Number of trash items in DB = 0
## DEBUG Feb 16, 2022 13:02:32 Total Number of vnmr_data attributes = 91
## DEBUG Feb 16, 2022 13:02:32 Total Number of study attributes = 52
## DEBUG Feb 16, 2022 13:02:32 Total Number of protocol attributes = 90
## DEBUG Feb 16, 2022 13:02:32 Total Number of image_dir attributes = 70
## DEBUG Feb 16, 2022 13:02:32 ** End DataBase Contents Report **
## INFO  Feb 16, 2022 13:05:00 **********************************************
## INFO  Feb 16, 2022 13:05:00 ********** Starting OpenVnmrJ as vnmr1 **********
## INFO  Feb 16, 2022 13:05:00 ** OpenVnmrJ VERSION 2.1 REVISION A **
## INFO  Feb 16, 2022 13:05:00 ** January 22, 2020 **
## INFO  Feb 16, 2022 13:05:00 ** mr400 **
## INFO  Feb 16, 2022 13:05:00 **********************************************
## DEBUG Feb 16, 2022 13:05:08 PROTON  Common
## DEBUG Feb 16, 2022 13:05:08 ### appDir 0 /home/vnmr1/vnmrsys
## DEBUG Feb 16, 2022 13:05:08 ### appDir 1 /vnmr
## DEBUG Feb 16, 2022 13:05:08 ### appDir 2 /vnmr/openvnmrj-utils
## ERROR Feb 16, 2022 13:07:14 No Experiment Active.
## DEBUG Feb 16, 2022 13:10:04 ** Start DataBase Contents Report **
## DEBUG Feb 16, 2022 13:10:04 DB version: pg_ctl (PostgreSQL) 9.2.24  DB Host: localhost.localdomain
              Local Host: localhost.localdomain     Network Mode: false
## DEBUG Feb 16, 2022 13:10:04 Number of vnmr_data items in DB = 330
## DEBUG Feb 16, 2022 13:10:04 Number of vnmr_par items in DB = 313
## DEBUG Feb 16, 2022 13:10:04 Number of vnmr_record items in DB = 0
## DEBUG Feb 16, 2022 13:10:04 Number of vnmr_rec_data items in DB = 0
## DEBUG Feb 16, 2022 13:10:04 Number of automation items in DB = 21
## DEBUG Feb 16, 2022 13:10:04 Number of workspace items in DB = 15
## DEBUG Feb 16, 2022 13:10:04 Number of panels_n_components items in DB = 1214
## DEBUG Feb 16, 2022 13:10:04 Number of shims items in DB = 13
## DEBUG Feb 16, 2022 13:10:04 Number of command_n_macro items in DB = 99
## DEBUG Feb 16, 2022 13:10:04 Number of pulse_sequence_macro items in DB = 88
## DEBUG Feb 16, 2022 13:10:04 Number of protocol items in DB = 237
## DEBUG Feb 16, 2022 13:10:04 Number of study items in DB = 12
## DEBUG Feb 16, 2022 13:10:04 Number of lcstudy items in DB = 0
## DEBUG Feb 16, 2022 13:10:04 Number of image_dir items in DB = 0
## DEBUG Feb 16, 2022 13:10:04 Number of trash items in DB = 0
## DEBUG Feb 16, 2022 13:10:04 Total Number of vnmr_data attributes = 91
## DEBUG Feb 16, 2022 13:10:04 Total Number of study attributes = 52
## DEBUG Feb 16, 2022 13:10:04 Total Number of protocol attributes = 90
## DEBUG Feb 16, 2022 13:10:04 Total Number of image_dir attributes = 70
## DEBUG Feb 16, 2022 13:10:04 ** End DataBase Contents Report **
## ERROR Feb 16, 2022 13:11:08 No Experiment Active.
## ERROR Feb 16, 2022 13:11:31 No Experiment Active.

DanIverson, 2022-02-17:
The log file does not indicate the problem. I assume this is OVJ 3.1. You said the acquisition system locks yet the status returns Idle. What do you mean by "acquisition system locks". Following any console reboot, restart the communications programs by running acqcomm twice from a terminal. The first to stop the current programs and the second to re-start them.

thezfisher, 2022-02-17:
> The log file does not indicate the problem. I assume this is OVJ 3.1. You said the acquisition system locks yet the status returns Idle. What do you mean by "acquisition system locks". Following any [...]

I have re-established communications, but if I attempt a setup for an experiment i.e. 'su'
the console returns: 'experiment started'
but never returns 'setup complete' 
After this happens, the instrument status says "Idle" but doesn't respond to any commands.

DanIverson, 2022-02-17:
Yes, an error message would help. Also, after re-establishing communications, from a terminal enter the following

script log
rlogin master1
DebugLevel=1

Then from OVJ, do the su. After it goes idle, in the terminal run

logout
cntrl-d

Send the log file.

thezfisher, 2022-02-17:
This was the total activity the day the issue began.

## INFO  Feb 13, 2022 13:35:54 **********************************************
## INFO  Feb 13, 2022 13:35:54 ********** Starting OpenVnmrJ as Hussaini **********
## INFO  Feb 13, 2022 13:35:54 ** OpenVnmrJ VERSION 2.1 REVISION A **
## INFO  Feb 13, 2022 13:35:54 ** January 22, 2020 **
## INFO  Feb 13, 2022 13:35:54 ** mr400 **
## INFO  Feb 13, 2022 13:35:54 **********************************************
## DEBUG Feb 13, 2022 13:36:01 Calibration  Experiment Selector
## DEBUG Feb 13, 2022 13:36:01 ### appDir 0 /home/Hussaini/vnmrsys
## DEBUG Feb 13, 2022 13:36:01 ### appDir 1 /vnmr
## DEBUG Feb 13, 2022 13:36:01 ### appDir 2 /vnmr/openvnmrj-utils
## INFO  Feb 13, 2022 13:36:36 **********************************************
## INFO  Feb 13, 2022 13:36:36 ********** Starting OpenVnmrJ as Hussaini **********
## INFO  Feb 13, 2022 13:36:36 ** OpenVnmrJ VERSION 2.1 REVISION A **
## INFO  Feb 13, 2022 13:36:36 ** January 22, 2020 **
## INFO  Feb 13, 2022 13:36:36 ** mr400 **
## INFO  Feb 13, 2022 13:36:36 **********************************************
## DEBUG Feb 13, 2022 13:36:43 Calibration  Experiment Selector
## DEBUG Feb 13, 2022 13:36:44 ### appDir 0 /home/Hussaini/vnmrsys
## DEBUG Feb 13, 2022 13:36:44 ### appDir 1 /vnmr
## DEBUG Feb 13, 2022 13:36:44 ### appDir 2 /vnmr/openvnmrj-utils
## INFO  Feb 13, 2022 14:10:00 **********************************************
## INFO  Feb 13, 2022 14:10:00 ********** Starting OpenVnmrJ as Hussaini **********
## INFO  Feb 13, 2022 14:10:00 ** OpenVnmrJ VERSION 2.1 REVISION A **
## INFO  Feb 13, 2022 14:10:00 ** January 22, 2020 **
## INFO  Feb 13, 2022 14:10:00 ** mr400 **
## INFO  Feb 13, 2022 14:10:00 **********************************************
## DEBUG Feb 13, 2022 14:10:08 Calibration  Experiment Selector
## DEBUG Feb 13, 2022 14:10:09 ### appDir 0 /home/Hussaini/vnmrsys
## DEBUG Feb 13, 2022 14:10:09 ### appDir 1 /vnmr
## DEBUG Feb 13, 2022 14:10:09 ### appDir 2 /vnmr/openvnmrj-utils
## INFO  Feb 13, 2022 14:11:14 **********************************************
## INFO  Feb 13, 2022 14:11:14 ********** Starting OpenVnmrJ as Hussaini **********
## INFO  Feb 13, 2022 14:11:14 ** OpenVnmrJ VERSION 2.1 REVISION A **
## INFO  Feb 13, 2022 14:11:14 ** January 22, 2020 **
## INFO  Feb 13, 2022 14:11:14 ** mr400 **
## INFO  Feb 13, 2022 14:11:14 **********************************************
## DEBUG Feb 13, 2022 14:11:21 Calibration  Experiment Selector
## DEBUG Feb 13, 2022 14:11:22 ### appDir 0 /home/Hussaini/vnmrsys
## DEBUG Feb 13, 2022 14:11:22 ### appDir 1 /vnmr
## DEBUG Feb 13, 2022 14:11:22 ### appDir 2 /vnmr/openvnmrj-utils
## ERROR Feb 13, 2022 14:12:07 current active acquisition is not owned by Hussaini
## INFO  Feb 13, 2022 14:14:05 **********************************************
## INFO  Feb 13, 2022 14:14:05 ********** Starting OpenVnmrJ as Hussaini **********
## INFO  Feb 13, 2022 14:14:05 ** OpenVnmrJ VERSION 2.1 REVISION A **
## INFO  Feb 13, 2022 14:14:05 ** January 22, 2020 **
## INFO  Feb 13, 2022 14:14:05 ** mr400 **
## INFO  Feb 13, 2022 14:14:05 **********************************************
## DEBUG Feb 13, 2022 14:14:12 Calibration  Experiment Selector
## DEBUG Feb 13, 2022 14:14:12 ### appDir 0 /home/Hussaini/vnmrsys
## DEBUG Feb 13, 2022 14:14:12 ### appDir 1 /vnmr
## DEBUG Feb 13, 2022 14:14:12 ### appDir 2 /vnmr/openvnmrj-utils
## INFO  Feb 13, 2022 14:15:49 **********************************************
## INFO  Feb 13, 2022 14:15:49 ********** Starting OpenVnmrJ as Hussaini **********
## INFO  Feb 13, 2022 14:15:49 ** OpenVnmrJ VERSION 2.1 REVISION A **
## INFO  Feb 13, 2022 14:15:49 ** January 22, 2020 **
## INFO  Feb 13, 2022 14:15:49 ** mr400 **
## INFO  Feb 13, 2022 14:15:49 **********************************************
## DEBUG Feb 13, 2022 14:15:57 Calibration  Experiment Selector
## DEBUG Feb 13, 2022 14:15:57 ### appDir 0 /home/Hussaini/vnmrsys
## DEBUG Feb 13, 2022 14:15:57 ### appDir 1 /vnmr
## DEBUG Feb 13, 2022 14:15:57 ### appDir 2 /vnmr/openvnmrj-utils
## ACQ ERROR Feb 13, 2022 14:17:32 exp1:  Controller(s) are unresponsive in initial synchronization step
## ACQ ERROR Feb 13, 2022 14:20:03 exp1:  Controller(s) are unresponsive in initial synchronization step
## ACQ ERROR Feb 13, 2022 14:23:09 exp1:  Controller(s) are unresponsive in initial synchronization step
## ACQ ERROR Feb 13, 2022 14:23:57 exp1:  Controller(s) are unresponsive in initial synchronization step
## ACQ ERROR Feb 13, 2022 14:24:49 exp1:  Controller(s) are unresponsive in initial synchronization step
## ACQ ERROR Feb 13, 2022 14:27:07 exp1:  Controller(s) are unresponsive in initial synchronization step
## ACQ ERROR Feb 13, 2022 14:30:03 exp1:  Controller(s) are unresponsive in initial synchronization step
## ACQ ERROR Feb 13, 2022 14:30:40 exp1:  Controller(s) are unresponsive in initial synchronization step
## ACQ ERROR Feb 13, 2022 14:31:39 exp1:  Controller(s) are unresponsive in initial synchronization step
## ACQ ERROR Feb 13, 2022 14:33:22 exp1:  Controller(s) are unresponsive in initial synchronization ste
[log.txt](https://github.com/OpenVnmrJ/OpenVnmrJ/files/8091467/log.txt)
p

DanIverson, 2022-02-17:
The log.txt indicates an error with the VT airflow. Are there any red lights on the magnet leg for the pneumatics?  The sychronization error is usually caused by a bad system reference clock.

thezfisher, 2022-02-17:
Going through the log file, there's an error 1902, which is consistent with an undervoltage error. Is this possibly CMOS batteries dying? This instrument has been here for almost 20 years so I'm thinking it may just be dropping CMOS, which would through off the reference clock, right?

DanIverson, 2022-02-17:
You may be right. I do not know. I am just a software guy. BTW, who are you and where are you? I may be able to put you in touch with a more hardware-oriented person.

I think the orange blinking light is a warning. You can reduce the VT airflow from the Spin/Temp panel in OVJ.

thezfisher, 2022-02-17:
> You may be right. I do not know. I am just a software guy. BTW, who are you and where are you? I may be able to put you in touch with a more hardware-oriented person.
I'm an undergraduate at the University of Tulsa. Ideally if I can verify the problem I can work with some of our electrical engineers to get the cards working properly.

> I think the orange blinking light is a warning. You can reduce the VT airflow from the Spin/Temp panel in OVJ.
Roger that. I'll do some work in getting the appropriate flow to fix that. Thank you so much for your help. I can get the superficial work done usually but as soon as there's deeper software issues I end up out of my depth.... I feel like I need a computer science degree just to keep up with instruments.

DanIverson, 2022-02-17:
For what it is worth, the outputs from the Clock Distribution Unit should be 80 MHz @ average 0.85V p-p @ 50Ω.

thezfisher, 2022-02-17:
I'll look into it. If you do know a hardware person that would be willing to answer a question or two that would be great, but no worries either way. You've been incredibly helpful across the board.... as a lot of this information is really difficult to find. Out of curiosity, is there anything worthwhile to read/study to learn more about the communications systems of these spectrometers, and get better at troubleshooting?

DanIverson, 2022-02-17:
You might want to join the IVAN SpinSights at https://join.slack.com/t/ivan-spinsights/shared_invite/zt-fbfzblja-vET5Gp5VsQdoCFbeBrNAfg
This is where NMR people can help each other. There is a nmr_services channel that lists some service providers. There is also a z_manuals section with all kinds of documentation. One particular one you may find useful is the HW training manual. If you enter "HW training" in the "Search IVAN SpinSights" field, Slack will find it for you.

thezfisher, 2022-02-17:
Thank you so much! I'll do some reading and see what I can find.

closed 2022-02-17 by thezfisher